WillCodeForCats/solaredge-modbus-multi #2: Ethernet is also for Solaredge comms - no wireless or cell options

stuarts0011, 2021-11-16:
Hi, 
I can't get this working.

Logger: pymodbus.client.sync
Source: /usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py:201
First occurred: 15 November 2021, 12:54:21 (4656 occurrences)
Last logged: 14:46:11

Connection to (192.168.178.13, 502) failed: [Errno 111] Connection refused
Connection to (192.168.178.13, 502) failed: timed out
Connection to (192.168.178.13, 502) failed: [Errno 113] Host is unreachable

Also

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:170
Integration: SolarEdge Modbus Multi-Inverter (documentation)
First occurred: 15 November 2021, 12:54:41 (4655 occurrences)
Last logged: 14:46:11

No open Modbus/TCP connection for solaredge

I can ping the inverter. 

Is it something to do with your instruction that "Ethernet is also for Solaredge comms - no wireless or cell options" because my inverter is connected over WIFI not Ethernet? If so, is there any simply fix for this?

WillCodeForCats, 2021-11-16:
Do you have Modbus/TCP enabled in the inverter? The default is disabled.

The default port is 1502 in the inverter config instead of 502, unless it was changed.

Connection refused usually means Modbus/TCP is disabled or the port number isn't set correctly.

I've heard that Modbus/TCP should work with wifi, but since I don't have wifi on my inverter I can't try it personally.

stuarts0011, 2021-11-16:
Yes, all of that is set correct. 
In fact I had it  working fine with one of the other multi inverter forks. But then I discovered that the fork I was using the cause of some  very serious lag on HA performance.  
So I tried yours.

WillCodeForCats, 2021-11-16:
Is the other integration still running? The solaredge inverter will only accept one TCP connection at a time, any other connection attempts will get connection refused from the inverter.

My saying "no wireless or cell options" has nothing to do with whether it will work or not, I'm just describing my test and development environment. Anything that will lead to a Modbus/TCP connection will work.

stuarts0011, 2021-11-16:
Thanks. 

No, the other integration is definitely uninstalled.

stuarts0011, 2021-11-16:
To me, it seems to be something very flaky about solaredge modbus.  As you say, something is wrong with the port on the inverter. I've reinstalled the other integration and that's not working either now. I've done everything I can think of to try and clear the inverter port block to no avail. Very annoying.

closed 2021-11-28 by WillCodeForCats