WillCodeForCats/solaredge-modbus-multi #7: Improved meter coniguration

WillCodeForCats, 2021-12-24:
* Make a separate config flow for meters
* Inverter device id configurable for meters

WillCodeForCats, 2022-03-21:
Working on ideas for auto-detection instead of as a config option.

purcell-lab, 2022-04-04:
> Working on ideas for auto-detection instead of as a config option.

Have a look at:
https://github.com/CJNE/ha-sunspec

It works with my SolarEdge inverter and auto detects my meters.

WillCodeForCats, 2022-04-06:
Yeah I only recently became aware of that integration. When I stared looking for a solution for my situation it never came up, probably because I only searched for "solarege" related integrations but also it appears to be pretty new.

The code for this integration has been hacked on to where I've kind of reached a limit of what I can do with it vs. what I wanted to do, and so I'm planning to rewrite it after I get another integration i'm working on released (my tekmar-482 integration). The rewrite will end up looking more like my tekmar integration than the current code is.

WillCodeForCats, 2022-05-12:
Auto detection for meters and batteries will be implemented in the v2.0.0 rewrite; it's currently working on my dev system.

purcell-lab, 2022-05-12:
> Auto detection for meters and batteries will be implemented in the v2.0.0 rewrite; it's currently working on my dev system.

Great, you might want to do a beta release so you can get a broader testing base before an actual release.

WillCodeForCats, 2022-05-12:
The rewrite is still a way off from anything resembling a release.

WillCodeForCats, 2022-05-13:
Anyone who wants to try auto-detection can manually install the `rewrite-for-2.0` branch. The only thing it does right now is create devices for each inverter, meter, or battery found (up to 3 meters and 2 batteries per inverter) and populate the device registries. That's it: it's not usable for anything yet. But even this is a huge improvement over the old code.

The battery part is best guess since I don't actually have batteries and there's no official documentation from SolarEdge or sunspec model support. That's for the old LG battery solution. The new SolarEdge Home Battery solution says up to 3 per Energy Hub, which doesn't match the existing scheme. So we'll just have to see what happens.

closed 2022-06-10 by WillCodeForCats