WillCodeForCats/solaredge-modbus-multi #12: Fronius modbus

dhover, 2022-01-25:
I have a SolarEdge inverter and also a Fronius Galvo inverter. I've installed this component for both inverters. The sensors of the Solaredge inverter looks very good. Better then other modbus components I've tested so far. Good work! 

[image]

[image]

The sensors of the Fronius inverter looks good. Some small improvement could be made.
As you can see the main sensor values are ok. The Fronius status codes are different and also the power factor is different.
So with no modifications I can use this custom component already for both inverters!

I think with some minor changes this component could also be interesting for other inverter brands e.g. Fronius.
@WillCodeForCats are you interested? If so, I could help you with debugging/testing the Fronius inverter for sure.

WillCodeForCats, 2022-01-26:
Since I've tried to use Sunspec as much as possible (except vendor specific stuff), it makes sense it should work with any inverter that follows the Sunspec modbus model. I don't have a problem adding other inverters to it, just need to find and read over their docs.

Power factor range is 1 to -1, so a negative value is valid because it indicates if it's leading or lagging.

One thing I see just quickly doing some first looks is that the Fronius inverter might have more configuration settings for modbus. It looks like there is a "sunspec model type" setting, and it has "float" or "int + SF" settings. Solaredge only has "int + SF" model, so make sure the Fronius is also set to "int + SF" if it has that option.

dhover, 2022-01-26:
> One thing I see just quickly doing some first looks is that the Fronius inverter might have more configuration settings for modbus. It looks like there is a "sunspec model type" setting, and it has [...]

You're right. My setting was indeed "int+SF". I was curious and tried the other settings but got an error saying a value too high or so. The int+SF setting for the Fronius together with this component is very useful. The status codes are different but that is not so important. I can change them myself with a template if necessary.

[42,0410,2049.pdf](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/7942926/42.0410.2049.pdf)

Attached the documentation of the Fronius datamanager modbus TCP. On page 32 and 33 the status codes are explained.

WillCodeForCats, 2022-01-26:
Thanks for the file, I will read it and see what the differences are.

WillCodeForCats, 2022-01-26:
As far as the "int+SF" model, what that does is make floating point values out of integer registers using a "scaling factor" that you apply to the integer value. Since the registers would be completely different between the two there's no way it would work in "float" mode as is.

dhover, 2022-01-26:
> As far as the "int+SF" model, what that does is make floating point values out of integer registers using a "scaling factor" that you apply to the integer value. Since the registers would be completely different between the two there's no way it would work in "float" mode as is.

I understand. Thanks for your reply.

dhover, 2022-01-26:
One thing which could be changed is the Manufacturer field.

[image]

It always shows SolarEdge. An improvement would be to fill this field with the text from the manufacturer sensor.

[image]

WillCodeForCats, 2022-01-26:
That is on the to-do list but requires a somewhat major rewrite of how sensors are added. I've tried a few hackish ways to get it to set that value with appropriate device groupings, but they didn't work. I have 3 devices and the original code this is based on just groups everything into one "device", but I don't like that personally and am already working on it.

What I intend to do is create classes for inverters and meters, then sensors as their own classes associated to the parent device class. This would also make it easier to add other inverter types since I can easily make a Solaredge class and a Fronius class, then base which ones to call on the Sunspec common model registers.

WillCodeForCats, 2022-01-26:
> It always shows SolarEdge. An improvement would be to fill this field with the text from the manufacturer sensor.

This is part of the general-cleanup-1 branch.

WillCodeForCats, 2022-01-26:
So the bad news on the Fronius temperature sensors is that they are documented as "not supported".
[image]

dhover, 2022-01-26:
Ok. Best would be to create no temperature sensor but a zero value is also a good solution for me.

WillCodeForCats, 2022-01-26:
Created PR #15 (Add support for Fronius inverters)

WillCodeForCats, 2022-02-02:
I've tested it as much as I can on my side, so it should be ready to merge in and we can do a release with the minor Fronius changes.

dhover, 2022-03-20:
Sorry for the delay. If I can test anything let me know.

dhover, 2022-03-20:
Oh now I see you've released v1.1.4.
I installed it. Name and status seems ok. Temperature is unavailable. So that's also ok.

dhover, 2022-03-21:
Yes I know that sunspec integration. I tried that one in the past but it is not so good.
So keep up the good work!

WillCodeForCats, 2022-03-21:
OK I didn't know anything about it. I ended up going my own way with this integration because of things I needed (multiple inverters was the major one) then changing it more to add stuff I wanted without waiting for someone else to do it.

closed 2022-03-30 by WillCodeForCats