WillCodeForCats/solaredge-modbus-multi #13: Feature list

maartenhendrix, 2022-01-26:
Hi,
Could you please add a feature list?
I’m on the binsentsu version now and not sure if I should change to this one. 

Thank you for the great work

WillCodeForCats, 2022-01-26:
The biggest major feature is the multiple inverter support. I needed that personally, and there's no support I could find for it elsewhere probably since most people only have one inverter. I saw someone had forked the binsentsu integration and did work for multiple inverters, but their work wasn't getting merged. So that's what started me doing this.

Since I gave up on my fork I started making changes to this one as its own integration without any concern for compatibility to others, so it's not a drop in replacement. All of the entity names in this integration will be different since they contain an inverter designator.

Smaller features I added:
- Inverter information sensors
- Meter information sensors
- "Description" attributes on status sensors
- Configuration (config flow) rewrite so it's user friendly
- Configuration option for inverter modbus address (Device ID)

Other things I've changed:
- Requires Home Assistant 2021.12.0 and newer, removed backwards compatibility
- Change things to Home Assistant recommended practices without breaking existing data
- Set measurement state class where applicable
- Use Home Assistant core constants for sensor units
- Upgrade pymodbus to 2.5.3
- Open/close the modbus/tcp connection every poll cycle instead of holding it open.
- Meter event descriptions: list of one or more active meter events as an attribute, or an empty list if no events are active.
- Add entity category support (PR #8)
- Catch cycle errors better - don't pass exceptions up to Home Assistant (PR #9)
- Add validation for sensors based on Sunspec model. (PR #11)
- Sensors that are unsupported or invalid will now be treated as "unavailable" in Home Assistant. (PR #11)
- Treat sensors with units VA, var, VAh, and varh as advanced sensor types. Make them disabled by default to avoid sensor clutter, but advanced users can enable them if desired. (PR #10)
- Breaking change: Changed VAR to var and VARh to varh to match SI units for volt amp reactive. However, unless someone was using an alternative meter the SolarEdge branded meter used in testing did not use these values (they were always 0)
- Device classes were added to more sensors, and depreciated constants were removed per Home Assistant release 2021.12.0

WillCodeForCats, 2022-01-26:
I've put this information on the [wiki](/WillCodeForCats/solaredge-modbus-multi/wiki) section.

closed 2022-01-26 by maartenhendrix