WillCodeForCats/solaredge-modbus-multi #16: Not working with 2 inverters

FranciscoFMayoralas, 2022-02-06:
Hello, thanks for the work as it is useful for installations with more than 1 inverter

I first tried https://github.com/binsentsu/home-assistant-solaredge-modbus but it was only detecting one inverter and only importing solar production reading from the master inverter then I realised you started this great add-on so I downloaded through HACS.

The Solaredge installation
- 2 inverters
- Master ID: 1
- Meter 1 ID: 2
- Master ID connected with ethernet

Installation of addon
- First I deleted the initial home-assistant-solaredge-modbus addon, restarted HA and then install solaredge-modbus-multi
- After restarting again HA, I introduced the data: local IP, port number, inverter ID (1), inverters amount (2), Meter 1 and polling 10 seconds

The result is that no data is imported to HA, Unavailable

WillCodeForCats, 2022-02-06:
Have you verified the second inverter has the unit id set correctly? If the master inverter has unit ID 1, the second one should be ID 2. When my system was installed the installer left both inverters set to unit ID 1 and I had to change it later. It would still report to solaredge, but not work with modbus.

Check the following settings in your second inverter and make sure it has Device ID set to 2 (when the master is Device ID 1):

[image]

[image]

[image]

WillCodeForCats, 2022-02-06:
Otherwise check the logs in Home Assistant and see if it logged any errors.

FranciscoBigWings, 2022-02-06:
Thanks. And the id for the meter? Should be in this case 3?

Francisco F. Mayoralas

FranciscoBigWings, 2022-02-06:
I just checked and slave inverter had ID also as number 1. As soon as I changed it to number 2 the system started working. You should make some kind of notice in documentation to advice people checking their inverter ID because I think that all inverters come with default ID number 1

The add-on doesn't read battery values right? Or is it something I misconfigured?

Thanks

WillCodeForCats, 2022-02-06:
I've added it to the wiki page: https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

No, it doesn't read batteries. That might be a future thing.

WillCodeForCats, 2022-02-06:
> Thanks. And the id for the meter? Should be in this case 3? Francisco F. Mayoralas

Meter ID can overlap because they should be on a different RS485 bus.

closed 2022-02-23 by WillCodeForCats