WillCodeForCats/solaredge-modbus-multi #17: Set up procedure with optional meter installed in inverter

Ekkib, 2022-02-23:
The component works when set up with "standard" entries IP address and port no. Then the actual inverter values are shown in Home Assistant. 
I was not successfull to readout the power meter, which is installed additionally in the inverter to measure the total ingoing/outgoing current to my house. The SolarEdge App shows these additional values.
I tried to turn on the 3 checkboxes in the configuration dialog in many different ways. Only then no values at all were reported, not even the values transmitted before.
There are two lines after you have to enter the TCP port (in my case 502). These default both to "1", without detailing, what the "1"s are used for. Other entries needed ?
I use this integration, because it works. I only have 1 inverter.
The additional meter may have ID 2 ?
Please help, maybe with adding some text to the wiki.
Thank you.

WillCodeForCats, 2022-02-23:
If you only have one meter then you should only select "Use Meter 1".

Otherwise if there are errors can you copy and paste errors from the logs?

WillCodeForCats, 2022-02-23:
The configuration fields are described here:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration#inverter-configuration

Ekkib, 2022-02-23:
> you should only select "Use Meter 1"

Magic. Now it seems to work perfect ! 

SolarEdge model SE7K, Firmware 3.2537, connected per WiFi

Many Thanks !

closed 2022-03-05 by WillCodeForCats