WillCodeForCats/solaredge-modbus-multi #18: two integrations with the same file location - how to safely remove one?

pejotigrek, 2022-03-06:
it's probably not a bug per se, but - maybe you can help me ;)
I was using the [Solaredge Modbus integration by bitsentsu](https://github.com/binsentsu/home-assistant-solaredge-modbus) - lets call it "A", and because I wanted to give a try to your integration (as being updated "more ecently", and I like to have my software "up-to-date") without changing anything I went full Leeroy Jenkins and additionaly installed your **Solaredge Modbus Multi-Inverter integration** - lets call it "B". both integrations installed via HACS.

integration B started to work without anything additional, as it probably read all configurations from the A, with one exception: as the integration B is planned for multi-inverter setup, an "I1" was added to entity names so now I had twice as many entities. I've noticed, that entities from integration A went unknown/unavailable, but all from your integration showed all the readings correctly. so - as I wanted to keep all my readings/history/recorder/influxdb archive to be consistent, I went to the dashboard and entity by entity started to delete entity from A, and to rename entity from B (by deleting "I1" - I'm not planning multi-inverter setup anyway). everything went smooth, and is working without problems.

BUT - after that I've noticed, that I still have the A integration listed in HACS... I've checked custom_components folder and if I'm correct, A & B use the same folder, just the files were overwritten ;)

so now goes my question: is it safe to uninstall the A integration via HACS, or it would remove the B integration along with data? or maybe there's some other way to delete the A integration by hand, so it won't be recognized by HACS as also installed in the system?

WillCodeForCats, 2022-03-07:
You bring up a good point, the folder name is the same. I didn't really consider that conflict because I'm still new to developing for home assistant. I've used home assistant for a long time but only recently started to try writing integrations for it, so I guess this is part of the learning process.

It's because the history is that I started with a fork, tried to get it merged upstream but I didn't do it right so I abandoned the fork and copied all my files to a new repo to start fresh. I've made a lot of changes since then to meet my own needs but it still has a base in common were it came from. I gave up on the other integration because it didn't support multiple inverters (mandatory for me since I have 2 inverters) and for some reason any pull requests to add multiple inverter support weren't getting merged in even though I'd tested one and reported it worked.

I don't know what will happen if you delete the other one in HACS. I would think HACS will delete the folder since they share the same name. So I'll have to look in to seeing if it's possible to change the folder name without breaking things.

pejotigrek, 2022-03-07:
thank you for such a detailed answer!

> I don't know what will happen if you delete the other one in HACS. I would think HACS will delete the folder since they share the same name. So I'll have to look in to seeing if it's possible to change the folder name without breaking things.

thank you! right now I'll keep the files/integratiouns untouched and will wait for you with any movements :) I guess the folder renaming can be attached to [issue #14](https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/14) if you're considering it.

WillCodeForCats, 2022-05-07:
I've created branch [solaredge-modbus-multi/tree/rename](https://github.com/WillCodeForCats/solaredge-modbus-multi/tree/rename) to get everything changed from solaredge_modbus to solaredge_modbus_multi both in code and custom component directory name. This will be version v1.2.0.

Unfortunately this is going to be a sort of breaking change, because some manual steps will need to happen such as: deleting the old integration (because it references the old directory name), re-adding the integration (to reference the new directory name), and deleting the old directory from config/custom_components. I don't think there's a way around those steps and unfortunately that was my beginner's mistake by not renaming it before releasing it into HACS.

The good news though is that entity names won't change.

I will be creating a wiki "how to" page on how to manually upgrade from v1.1.4 to v1.2.0 and hopefully it won't be too painful.

WillCodeForCats, 2022-05-07:
Created wiki page: [How To Upgrade from v1.1.x to v1.2.x](https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/How-To-Upgrade-from-v1.1.x-to-v1.2.x)

The procedure were based on a test VM and the WillCodeForCats/hacs-rename-test custom HACS repo.

Ekkib, 2022-05-07:
Hi Seth,

Removing the driectory can be done easier. Just go to HACS, click the
three dots within the integration box, remove/delete (last entry in the
menu, I see the German translation). As far as I remember this also deletes
the directory. So no manual activity is needed. Please try yourself, as I
am also not that advanced.

Once again thanks for your phantastic software !

pejotigrek, 2022-05-08:
> Created wiki page: [How To Upgrade from v1.1.x to v1.2.x](https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/How-To-Upgrade-from-v1.1.x-to-v1.2.x)
> 
> The procedure were based on a test VM and the WillCodeForCats/hacs-rename-test custom HACS repo.

thanks for the guide!
do you know what happens to the data recorded with integration?
I mean - you wrote earlier _"The good news though is that entity names won't change"_ and that's cool [I won't need to change my dashboard/automations etc. based on those names] but I wonder if deleting the integration [and getting HomeAssisstant to know that - by rebooting it] will cause some database work like "integration is gone, so flush those entities"? probably not, but I had to ask so I won't lose my solar statistics ;)

WillCodeForCats, 2022-05-08:
Entity names and the unique IDs Home Assistant uses to reference them internally shouldn't change, so there should not be any history data loss.

However, there will eventually be an unavoidable breaking change that will cause history loss coming because the integration this was based on did not follow Home Assistant development guidelines. Discussion about that here: https://github.com/WillCodeForCats/solaredge-modbus-multi/discussions/21

WillCodeForCats, 2022-05-13:
As soon as home-assistant/brands/pull/3391 is approved and checks pass, PR #29 will be merged.

WillCodeForCats, 2022-05-14:
Had to resubmit a new PR because I messed up something causing checks to fail.

New one is: home-assistant/brands/pull/3415

WillCodeForCats, 2022-05-15:
Ready to merge the name change.

This will be release v1.2.0

WillCodeForCats, 2022-05-17:
Doing some final testing on my test VM to make sure PR #29 doesn't affect existing entity names or history.

WillCodeForCats, 2022-05-19:
I did the rename onmy test VM after collecting data for a day and entity history wasn't lost for me, but I've added a note to the wiki page that it's possible to lose entity history in the instructions.

And with that I'm going to merge #29 for a v1.2.0 release.