WillCodeForCats/solaredge-modbus-multi #22: Battery integration

sragas, 2022-04-05:
Are there any plans to integrate connected batteries?

WillCodeForCats, 2022-04-06:
Yeah eventually. Right now I'm focusing on getting another integration released (my tekmar-482 integration) and once that's out I'll revisit this one. I'm planning to rewrite this integration since I'm finding how it was written is limiting to things I want to do with it, and he structure I used for the tekmar-482 integration is more flexible for me.

skl283, 2022-04-20:
Will also appreciate the battery integration to use your Integration! :-)

WillCodeForCats, 2022-05-31:
The [rewrite-for-2.0](https://github.com/WillCodeForCats/solaredge-modbus-multi/tree/rewrite-for-2.0) branch is at a point where it should detect batteries and create device entries for them, but there aren't any sensor entities yet.

WillCodeForCats, 2022-06-02:
Anyone who has batteries please try the [rewrite-for-2.0](https://github.com/WillCodeForCats/solaredge-modbus-multi/tree/rewrite-for-2.0) branch and report if it detects your batteries or not. It will have some basic diagnostic sensors and show a device for each battery (up to 2 per inverter for up to 32 inverters).

Because battery registers are not technically officially supported by SolarEdge, auto-detection for batteries is disabled by default. To enabled battery detection, go to the device options and enable it.

[image]

[image]

Doonga, 2022-06-02:
Hi, I gave this a shot, I got devices for the inverter and the 2 meters, but no batteries. I have the newer SolarEdge Battery Banks BAT-10K1P, there are two of them.

I rolled back to the normal version for now since I wasn't sure if all of the device names changed. I'll need to go edit a bunch of stuff to keep things working.

WillCodeForCats, 2022-06-02:
Inverter entity names will be the same, meter entities will be slightly different since auto detect will scan every inverter for all 3 meters, which means it's possible to have more than 3 meters, so it names the meter entities with the inverter ID and the meter ID instead of fixed numbers.

If you set debug logging it will log something for batteries. Basically the way auto detect works is it tries to read some basic info registers and if the read is OK then it adds the device. If the read returns any kind of error it ignores that location. The battery registers have been taken from an engineering pdf for the LG batteries where only 2 batteries per inverter were supported.

To enable debug logging:
```
logger:
    default: warning
    logs:
        custom_components.solaredge_modbus_multi: debug
```

And then in my logs the important lines are:

```
DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 battery 1: Exception Response(131, 3, IllegalAddress)
DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 battery 2: Exception Response(131, 3, IllegalAddress)
```

Which means there aren't any batteries at the battery register, in my case because I don't have any so the address doesn't exist. If you see those exceptions and you're not using the old LG batteries, then that means it's likely SolarEdge changed the battery registers or isn't using them.

WillCodeForCats, 2022-06-02:
I just pushed some updates to logging attributions so you'll need to re-download.

Doonga, 2022-06-02:
Hmm yep looks like the new ones aren't using the registers... That's very unfortunate...

`2022-06-02 19:18:17 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 battery 1: Exception Response(131, 3, IllegalAddress)
2022-06-02 19:18:17 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 battery 2: Exception Response(131, 3, IllegalAddress)
`

WillCodeForCats, 2022-06-02:
Well, I'll keep an eye out for any information about registers for the new Energy Bank system. Otherwise it's anyone's guess if they moved to different addresses or what.

Doonga, 2022-06-03:
Yep, no worries. Hopefully somewhere down the road. Thanks!

WillCodeForCats, 2022-06-10:
Pre-release v2.0.0-pre.1 is available now.

muenchris, 2022-06-15:
I have two Inverter and each inverter has one battery attached. But I do not think SolarEdge is exposing the batteries via Modbus TCP at all. Using modbus test tools, there are no values exported where they should be according to the SunSpec. I guess the inverters are collecting the battery info via RS485 but do not copy the values into the Modbus TCP range. Luckily the LG batteries have a mini WebServer on board that can easily be used to read all BMS values

WillCodeForCats, 2022-06-15:
LG batteries should be detected. My code may have been off by one count on the battery info read block, I will release a pre.7 with what I think might be the correct value.

muenchris, 2022-06-15:
Nice, I can test right away if you like

WillCodeForCats, 2022-06-15:
[Release v2.0.0-pre.7](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.0.0-pre.7)

WillCodeForCats, 2022-06-15:
If you're using HACS you will have to click "update information" on the integration card in HACS.

muenchris, 2022-06-15:
I have auto-detect batteries on and restarted HA. After the restart - will it try to auto-detect the batteries automatically or do I have to do something? I updated the HACS integration to Pre7 first then restarted. The Integration still shows 5 devices (2 inverters + 3 Meters) - no batteries.

Also, do the batteries require specific DeviceIDs?

WillCodeForCats, 2022-06-15:
You have to enable auto battery detect in config options:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration#integration-configuration-options

It's disabled by default.

muenchris, 2022-06-15:
yes I had it on before the restart

WillCodeForCats, 2022-06-15:
Hmm, well, I don't know then. I don't have batteries so anything I try for them is random guessing based on other information I can find.

muenchris, 2022-06-15:
As I wrote before, I used a Modbus tool to see if the SolarEdge inverter somehow exposes the batteries, but there is no data returned where the SunSpec is referencing it. I guess that SE is not exposing it via Modbus TCP.
My next step is to see if I can somehow use the same path the SE SetApp is going. That seams to have much more information including the optimizers of each panel. The only tricky thing with that is that the server is only on during the "Commissioning" (with a Phone/PC connected with Wifi). Also they are using Protobuf  and that is hard to decode without the .proto file.

WillCodeForCats, 2022-06-15:
They're supposed to start at 57600 for batteries, they do not use sunspec storage model.

muenchris, 2022-06-15:
I see, makes sense. I will try to see what I get at that address

WillCodeForCats, 2022-06-15:
Thanks for trying. In a perfect world, SolarEdge would expand the modbus interface to support the sunspec storage model.

muenchris, 2022-06-15:
yes, I hope they do someday

muenchris, 2022-06-16:
I just had my Span panels commissioned with my two SolarEdge inverters and they told me that SolarEdge has to "Enable the battery" for the Modbus interface. Looks like SolarEdge is limiting local access to the system more and more. 

So if anyone does not see the battery, its because SolarEdge has not enabled the modbus TCP interface for the batteries.

WillCodeForCats, 2022-06-16:
Is that something they will do if you request it?

muenchris, 2022-06-16:
I am not sure. Span is doing this for me but I am not sure anyone can request it. I will try to find out

mbillings6, 2022-06-22:
I just downloaded pre7 to test it, I have a single inverter with the LG batteries. I have been searching for information on this an I have seen some that say they are using references up in the 62786 range and above for batteries lately.

WillCodeForCats, 2022-06-23:
FYI pre.7 is the same as release v2.0.0, just without the pre-release flag.

If the registers changed I wonder if it's related to firmware updates to support new their new batteries.

mbillings6, 2022-06-25:
I am not sure how helpful this will be but I found this document related to SunSpec on SolarEdge power control protocols. I am still pretty new at working with Modbus TCP or Modbus period so I have yet to make sense of it.

https://www.photovoltaikforum.com/core/attachment/88445-power-control-open-protocol-for-solaredge-inverters-pdf/

WillCodeForCats, 2022-06-25:
That PDF is what gives the 57600 starting register for batteries, but it's also 5 years old.

What I will have to try myself (when I have some free time) is simulate a modbus slave with data to make sure the integration at least works with a simulated battery since I don't have a real one.

mbillings6, 2022-06-25:
If you want, we try it on my setup. I have the SE3800H-USS3BBC14 inverter and the LG RESU10H Prime for the battery. I have tried accessing registers above like 41000 and it sends back and illegal address message.

Dave-Casey, 2022-07-01:
I'm also using the LG RESU10H Prime battery attached to my LG RESU10H Prime.  I have enabled `Auto-detect batteries` but do not see it.  Let me know how I can help.

WillCodeForCats, 2022-07-01:
See these instructions for debug logging and what to look for:

https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/22#issuecomment-1145413524

muenchris, 2022-07-02:
My Span.io panels finally work with my two inverters and the two batteries. There was some "Magic" going on between the SolarEdge Support and the Span.IO developers leading up to SolarEdge "enabling the registers" for my inverter. After a restart of my inverter (something they did without telling me and occurred right in the middle of a Zoom Call), the Span App now shows the correct battery status. Since they are accessing my inverter via Modbus TCP, I assume that "Magic" is something SolarEdge needs to enable for the batteries to show up. 

Unfortunately Span is keeping the Modbus TCP connection open to the inverter and since the SolarEdge inverter is a Highlander (there can be only one...connection), I cannot access it with the this HA integration as well. 

One way of testing would be to disconnect the Span panel temporarily and then try to use this integration. My fear is that after that, my span app might not work anymore. 

My best advise to you all: Contact SolarEdge Support and ask them to enable the "full" Modbus TCP registers.

WillCodeForCats, 2022-07-02:
There's a modbus proxy add-on thing to deal with the there can only be one issue. I don't remember the name offhand though. Maybe this one? https://github.com/Akulatraxas/ha-modbusproxy/ 

Since this integration closes the connection after reading technically it can coexist with other stuff, but only as long as those other things also close the connection between their polling cycles.

muenchris, 2022-07-02:
yes, my software and your integration does that...but Span does not. And since they discover the invertersomehow, I am not sure if it would work with their implementation. 
All these integrations are still very uncoordinated. I hope things are getting better over time

mbillings6, 2022-07-08:
I was on the phone with Solar Edge Support earlier, and I was able to confirm the Nov 2017 document is the latest on the Power Control Open Protocol. However, they were not really able to help me outside of that. Based on what @muenchris mentioned previously about Solar Edge doing something to unlock the addition registers and my conversation and research. I would say the answer is probably in the Power Control section of the inverter. I may eventually test this theory.

WillCodeForCats, 2022-07-10:
I suppose if I had a battery, the first thing I would try is checking the value of `0xF142` and try setting `0xF142 = 1` (if it was `0`) then commit it with `0xF100 = 1`.

rzulian, 2022-07-12:
I do have a LG battery connected to the inverter 1, that is not being recognised by this custom component, but it works with others. I'll check what's wrong

TheGardenMonkey, 2022-07-12:
Which other approach has worked for you? I have an LG as well, but using the solaredge_modbus python module directly I am not able to get the batteries (also don't seem to get accurate values, but no battery). 

> I do have a LG battery connected to the inverter 1, that is not being recognised by this custom component, but it works with others. I'll check what's wrong

Can you point me to some instructions on how to try this out?
> I suppose if I had a battery, the first thing I would try is checking the value of `0xF142` and try setting `0xF142 = 1` (if it was `0`) then commit it with `0xF100 = 1`.

rzulian, 2022-07-12:
> 

I've been working on this fork https://github.com/rzulian/home-assistant-solaredge-modbus/tree/mpredfearn/multiple_inverters
I've enabled multiple inverters. This is working just fine, but it is not maintained so I was trying this component instead.

rzulian, 2022-07-12:
I'm using v2.0.3, accordingly to the manifest.json
Thisn is nthe line that I've added after the try.
```
        if battery_info.isError():
            _LOGGER.debug(f"Inverter {self.inverter_unit_id} battery {self.battery_id}: {battery_info}")
            raise RuntimeError(battery_info)
        _LOGGER.debug(f"Inverter {self.inverter_unit_id} battery {self.battery_id} found: {battery_info}")
```

WillCodeForCats, 2022-07-12:
Oh, so that was a line you added. It didn't make sense to me since I didn't see it in my copy.

It's probably just an issue with the decoded_common block then, possibly I have a type wrong or the register count is off and it's trying to decode off the end.

WillCodeForCats, 2022-07-12:
I see it now, it's a simple mistake.

On line 711 `decoded_common` should be `self.decoded_common`

It's probably bailing out there.

WillCodeForCats, 2022-07-12:
I pushed the correction to the `main` branch.

TheGardenMonkey, 2022-07-13:
Still no battery detected for me after using the correct code. For reference these are the devices I have:

SolarEdge SE7600H-USS3BBC14 
SolarEdge SE-RGMTR-1D-240C-B 
SolarEdge SE-MTR-3Y-240V-A 
LG RESU10H

It feels like something else must be blocked since I am not able to see the battery through the python clients directly.

rzulian, 2022-07-13:
I've a working version which is loading the battery. I cannot create a pull request, because my formatting is adding a lot o changes. @WillCodeForCats you can check my changes in my fork, and if you want to share your formatting settings, I can create a pull request.

WillCodeForCats, 2022-07-13:
I don't have any formatting settings, I use BBEdit on MacOS to do it all manually.

Can't you just change what the issue without making formatting changes? If it's the variable I already pushed just try that.

WillCodeForCats, 2022-07-13:
I used a modbus simulator to find and fix other errors related to the battery device class. I will publish a pre-release shortly.

WillCodeForCats, 2022-07-13:
[Release v2.0.3-battery.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.0.3-battery.1)

rzulian, 2022-07-13:
I've tested it and it works. Are you going to create the sensors for the batteries?

WillCodeForCats, 2022-07-13:
Yes, that's the plan. There should be one sensor already: the device sensor.

Can you provide all the attributes for each device sensor it created?

TheGardenMonkey, 2022-07-13:
I now see the following with debug enabled and the new beta:

```
2022-07-13 16:40:13 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 battery 1: Exception Response(131, 3, IllegalAddress)
2022-07-13 16:40:13 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Inverter 1 battery 2: Exception Response(131, 3, IllegalAddress)
```

Are there some suggestions for how i can try to figure out what are the proper registers for my configuration?

rzulian, 2022-07-13:
> Yes, that's the plan. There should be one sensor already: the device sensor.
> 
> Can you provide all the attributes for each device sensor it created?

> Yes, that's the plan. There should be one sensor already: the device sensor.
> 
> Can you provide all the attributes for each device sensor it created?

these?

[image]

WillCodeForCats, 2022-07-13:
Yeah, but they don't look quite right. It says you have a 2 watt battery.

Can you try changing line 704 to `battery_info.registers, byteorder=Endian.Big, wordorder=Endian.Little` and see how that changes the reported values?

WillCodeForCats, 2022-07-13:
Change these lines: c2da772e079c6f2059dde5b288e3405e9dd3d2d0

Or download the branch: https://github.com/WillCodeForCats/solaredge-modbus-multi/archive/refs/heads/22-battery-integration.zip

WillCodeForCats, 2022-07-13:
[Release v2.0.3-battery.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.0.3-battery.2)

WillCodeForCats, 2022-07-14:
> 
> Are there some suggestions for how i can try to figure out what are the proper registers for my configuration?

All inverters should use the same registers. These messages mean your inverter says they are invalid. It's been suggested that some inverters (newer systems?) have them disabled and requireSolarEdge to enable them, but since I don't have batteries personally I don't have any experience with that so maybe someone else will have a suggestion.

[58 more replies]