WillCodeForCats/solaredge-modbus-multi #23: Bug / feature request

X3Cutor, 2022-04-11:
I'm building some automation around energy optimization in my house. For my use case, it would be fantastic if a failure to read the modbus-sensors would make them unavailable so the automation can go to safe defaults.

At present, the sensors will keep displaying the last value they had if the network connection between homeassistant and solaredge inverter is interrupted. I am willing to help and/or test, I had a look at the code but didn't see any obvious place to plug in what I'm looking for but then again I'm far from an expert in Python and even less so in home-assistant integration specifics.

WillCodeForCats, 2022-04-11:
Yes it's possible, will do it in a future version.

WillCodeForCats, 2022-05-07:
To do after release v1.2.0

closed 2022-06-10 by WillCodeForCats