WillCodeForCats/solaredge-modbus-multi #24: Error: filedescriptor out of range in select()

gieljnssns, 2022-04-22:
Before reporting a bug, make sure your configuration is correct and your LAN is working.
https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

**Describe the bug**
A clear and concise description of what the bug is.
I'm using your integration for about a half year now.
But now the integration stops working after a while.
When I restart HA it works again for some time

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect it keeps working

**Screenshots**
If applicable, add screenshots of the problem and/or configuration.
[image]

**Logs**
Copy any log messages/errors from Home Assistant from "custom_components.solaredge_modbus" and "pymodbus.client.sync".
```
Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:491
Integration: SolarEdge Modbus Multi-Inverter (documentation, issues)
First occurred: 10:49:17 (7 occurrences)
Last logged: 10:55:17

Error reading inverter at id 1: filedescriptor out of range in select()
```
**Home Assistant (please complete the following information):**
 - Version: 2022.4.6

**Additional context**
Add any other context about the problem here.

gieljnssns, 2022-04-26:
Debug information
```
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.79.54', 50909)
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Current transaction state - TRANSACTION_COMPLETE
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Running transaction 23
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] SEND: 0x0 0x17 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x44 0x0 0x6c
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] RECV: 0x0 0x17 0x0 0x0 0x0 0xdb 0x1 0x3 0xd8 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x33 0x2e 0x32 0x35 0x33 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x37 0x46 0x31 0x35 0x30 0x35 0x35 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x65 0x0 0x32 0x46 0xcd 0x46 0xcd 0xff 0xff 0xff 0xff 0xff 0xfc 0x9 0x1a 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x10 0x54 0xff 0xff 0xc3 0x6a 0xff 0xfd 0x10 0x76 0xff 0xff 0x15 0x18 0xff 0xfe 0xd9 0x44 0xff 0xfe 0x1 0x22 0x74 0x72 0x0 0x0 0x2f 0x27 0xff 0xfc 0xd 0xbb 0xff 0xff 0x10 0x93 0xff 0xff 0x80 0x0 0x7 0x13 0x80 0x0 0x80 0x0 0xff 0xfe 0x0 0x4 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xff
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x17 0x0 0x0 0x0 0xdb 0x1 0x3 0xd8 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x33 0x2e 0x32 0x35 0x33 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x37 0x46 0x31 0x35 0x30 0x35 0x35 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x65 0x0 0x32 0x46 0xcd 0x46 0xcd 0xff 0xff 0xff 0xff 0xff 0xfc 0x9 0x1a 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x10 0x54 0xff 0xff 0xc3 0x6a 0xff 0xfd 0x10 0x76 0xff 0xff 0x15 0x18 0xff 0xfe 0xd9 0x44 0xff 0xfe 0x1 0x22 0x74 0x72 0x0 0x0 0x2f 0x27 0xff 0xfc 0xd 0xbb 0xff 0xff 0x10 0x93 0xff 0xff 0x80 0x0 0x7 0x13 0x80 0x0 0x80 0x0 0xff 0xfe 0x0 0x4 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xff
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 23
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 23
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.payload] [21359, 27745, 29253, 25703, 25888, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21317, 13104, 12336, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12336, 12339, 11826, 13619, 14080, 0, 0, 0, 14150, 12597, 12341, 13622, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 101, 50, 18125, 18125, 65535, 65535, 65532, 2330, 65535, 65535, 65535, 65535, 65535, 65535, 4180, 65535, 50026, 65533, 4214, 65535, 5400, 65534, 55620, 65534, 290, 29810, 0, 12071, 65532, 3515, 65535, 4243, 65535, 32768, 1811, 32768, 32768, 65534, 4, 0, 65535, 65535, 65535]
2022-04-26 09:06:53 DEBUG (MainThread) [pymodbus.payload] [b'\x01"', b'tr']
2022-04-26 09:07:06 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0x0 0x69 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2022-04-26 09:07:06 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 105
2022-04-26 09:07:06 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0x0 0x69 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x1 0xf2 0xf 0xae 0xf 0xb1 0xf 0xb8 0x9 0x13 0x9 0xa 0x9 0x13 0x0 0x0 0x2 0xd0 0x0 0x0 0x2 0xc9 0x0 0x0 0x2 0xe3 0x0 0x0 0x1 0xe1 0x0 0x0 0x1 0xe0 0x0 0x0 0x0 0x0 0x3 0xe8 0x13 0x8b 0x25 0xa2 0x1 0x7c 0xb 0xb8 0x2 0x0 0x0 0x0 0x62 0x67 0x98 0x48 0xff 0xff 0xff 0xff 0x0 0x0 0x1 0xe9 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x9 0x52 0x68 0x0 0x9 0xdc 0x42 0x62 0x67 0xb6 0x35 0x0 0x0 0x0 0x5 0x0 0x0 0x0 0x25
2022-04-26 09:07:06 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x69 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x1 0xf2 0xf 0xae 0xf 0xb1 0xf 0xb8 0x9 0x13 0x9 0xa 0x9 0x13 0x0 0x0 0x2 0xd0 0x0 0x0 0x2 0xc9 0x0 0x0 0x2 0xe3 0x0 0x0 0x1 0xe1 0x0 0x0 0x1 0xe0 0x0 0x0 0x0 0x0 0x3 0xe8 0x13 0x8b 0x25 0xa2 0x1 0x7c 0xb 0xb8 0x2 0x0 0x0 0x0 0x62 0x67 0x98 0x48 0xff 0xff 0xff 0xff 0x0 0x0 0x1 0xe9 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x9 0x52 0x68 0x0 0x9 0xdc 0x42 0x62 0x67 0xb6 0x35 0x0 0x0 0x0 0x5 0x0 0x0 0x0 0x25
[... cut ...]
2022-04-26 09:08:53 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-04-26 09:08:53 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading inverter at id 1: filedescriptor out of range in select()
```
09:07:53 is the first time I get this error

Maybe something to know. I'm also using another modbus integration https://github.com/wlcrs/huawei_solar/blob/main/manifest.json
So I don't know of which integration the pymodbus debug is.
[image]

WillCodeForCats, 2022-04-27:
These could be related:

https://github.com/riptideio/pymodbus/issues/721
https://github.com/riptideio/pymodbus/pull/722

This integration is pinned to pymodbus 2.5.3, but if it's a network communications issue causing this that would be another thing to consider.

gieljnssns, 2022-04-28:
How can I check if it is something on my network?

gieljnssns, 2022-04-28:
[image]
[image]
[image]
[image]
[image]

gieljnssns, 2022-04-28:
Is it possible to try the dev version?
https://github.com/riptideio/pymodbus/releases/tag/v3.0.0dev.2

gieljnssns, 2022-04-30:
Just a small update, when I downgrade to v1.1.1-meter-id-fix, I still don't have any error after 1h.

gieljnssns, 2022-05-02:
When using v1.1.2 with manifest file adapted to `"requirements": ["pymodbus==3.0.0.dev3"],`
I'm seeing this error
```
Deze fout is ontstaan door een aangepaste integratie.

Logger: homeassistant
Source: custom_components/solaredge_modbus/__init__.py:187
Integration: SolarEdge Modbus Multi-Inverter (documentation, issues)
First occurred: 12:51:25 (193 occurrences)
Last logged: 16:03:26

Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/config/custom_components/solaredge_modbus/__init__.py", line 156, in async_refresh_modbus_data
    update_result = self.read_modbus_data()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 197, in read_modbus_data
    self.read_modbus_data_inverters()
  File "/config/custom_components/solaredge_modbus/__init__.py", line 207, in read_modbus_data_inverters
    inverter_data = self.read_holding_registers(
  File "/config/custom_components/solaredge_modbus/__init__.py", line 187, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 109, in execute
    raise ConnectionException("Failed to connect[%s]" % (self.__str__()))
  File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 174, in execute
    response, last_exception = self._transact(
  File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 294, in _transact
    result = self._recv(response_length, full)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 326, in _recv
    read_min = self.client.framer.recvPacket(min_size)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/framer/__init__.py", line 49, in recvPacket
    return self.client.recv(size)
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 89, in recv
    def recv(self, size):
  File "/usr/local/lib/python3.9/site-packages/pymodbus/client/sync.py", line 293, in _recv
    while recv_size > 0:
ValueError: filedescriptor out of range in select()

```

WillCodeForCats, 2022-05-05:
PR #9 starting with release v1.1.3 should address this and allow it to continue to the polling next cycle instead of waiting to restart HA.

gieljnssns, 2022-05-08:
When I update back to `v1.1.4` and use the `pymodbus 3.0.0.dev3` version
I have this errors
```
Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.solaredge_modbus
Source: custom_components/solaredge_modbus/__init__.py:491
Integration: SolarEdge Modbus Multi-Inverter (documentation, issues)
First occurred: 08:27:26 (15 occurrences)
Last logged: 08:38:41

Error reading inverter at id 1: Modbus Error: [Connection] ModbusTcpClient(192.168.79.56:502): Connection unexpectedly closed 0.000035 seconds into read of 8 bytes without response from unit before it closed connection
Error reading inverter at id 1: Modbus Error: [Connection] ModbusTcpClient(192.168.79.56:502): Connection unexpectedly closed 0.000023 seconds into read of 8 bytes without response from unit before it closed connection
Error reading inverter at id 1: Modbus Error: [Connection] ModbusTcpClient(192.168.79.56:502): Connection unexpectedly closed 0.000020 seconds into read of 8 bytes without response from unit before it closed connection
Error reading inverter at id 1: Modbus Error: [Connection] ModbusTcpClient(192.168.79.56:502): Connection unexpectedly closed 0.000019 seconds into read of 8 bytes without response from unit before it closed connection
Error reading inverter at id 1: Modbus Error: [Connection] ModbusTcpClient(192.168.79.56:502): Connection unexpectedly closed 0.000021 seconds into read of 8 bytes without response from unit before it closed connection
```
The errors Always start about 25min after restarting HA

WillCodeForCats, 2022-05-08:
It's saying the unit is dropping the connection. Maybe try a longer polling interval if it's set to something fast (maybe too fast). I don't have any other ideas at the moment.

gieljnssns, 2022-05-08:
I use 45s as a polling interval
I have disabled my other modbus integration so this debug logs are only from the SolarEdge integration
```
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.79.54', 55027)
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Current transaction state - IDLE
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Running transaction 1
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x44 0x0 0x6c
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] RECV: 0x0 0x1 0x0 0x0 0x0 0xdb 0x1 0x3 0xd8 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x33 0x2e 0x32 0x35 0x33 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x37 0x46 0x31 0x35 0x30 0x35 0x35 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x65 0x0 0x32 0x27 0xf2 0x27 0xf2 0xff 0xff 0xff 0xff 0xff 0xfd 0x9 0x71 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x60 0x40 0xff 0xff 0xc3 0x51 0xff 0xfd 0x60 0x6d 0xff 0xff 0x3a 0x34 0xff 0xfe 0xd9 0x3 0xff 0xfe 0x1 0x25 0x63 0xfc 0x0 0x0 0x1a 0xd6 0xff 0xfd 0xe 0x39 0xff 0xff 0x61 0xb7 0xff 0xff 0x80 0x0 0x9 0x62 0x80 0x0 0x80 0x0 0xff 0xfe 0x0 0x4 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xff
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0xdb 0x1 0x3 0xd8 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x33 0x2e 0x32 0x35 0x33 0x37 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x37 0x46 0x31 0x35 0x30 0x35 0x35 0x36 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1 0x0 0x65 0x0 0x32 0x27 0xf2 0x27 0xf2 0xff 0xff 0xff 0xff 0xff 0xfd 0x9 0x71 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0xff 0x60 0x40 0xff 0xff 0xc3 0x51 0xff 0xfd 0x60 0x6d 0xff 0xff 0x3a 0x34 0xff 0xfe 0xd9 0x3 0xff 0xfe 0x1 0x25 0x63 0xfc 0x0 0x0 0x1a 0xd6 0xff 0xfd 0xe 0x39 0xff 0xff 0x61 0xb7 0xff 0xff 0x80 0x0 0x9 0x62 0x80 0x0 0x80 0x0 0xff 0xfe 0x0 0x4 0x0 0x0 0xff 0xff 0xff 0xff 0xff 0xff
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 1
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 1
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.payload] [21359, 27745, 29253, 25703, 25888, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21317, 13104, 12336, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12336, 12339, 11826, 13619, 14080, 0, 0, 0, 14150, 12597, 12341, 13622, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 101, 50, 10226, 10226, 65535, 65535, 65533, 2417, 65535, 65535, 65535, 65535, 65535, 65535, 24640, 65535, 50001, 65533, 24685, 65535, 14900, 65534, 55555, 65534, 293, 25596, 0, 6870, 65533, 3641, 65535, 25015, 65535, 32768, 2402, 32768, 32768, 65534, 4, 0, 65535, 65535, 65535]
2022-05-08 10:19:19 DEBUG (MainThread) [pymodbus.payload] [b'\x01%', b'c\xfc']
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.79.54', 52049)
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.transaction] Current transaction state - TRANSACTION_COMPLETE
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.transaction] Running transaction 2
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.transaction] SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x44 0x0 0x6c
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-05-08 10:20:04 DEBUG (MainThread) [pymodbus.transaction] RECV: 0x0 0x2 0x0 0x0 0x0 0xdb 0x1 0x3 0xd8 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x33 0x30 0x30 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x33 0x2e 0x32 0x35 0x33 0x37 0x0 0x0 0x0 0x0 0x0 0x0  [... cut ...]
2022-05-08 10:47:49 DEBUG (MainThread) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-05-08 10:47:49 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-05-08 10:47:49 ERROR (MainThread) [custom_components.solaredge_modbus] Error reading inverter at id 1: Modbus Error: [Connection] ModbusTcpClient(192.168.79.56:502): Connection unexpectedly closed 0.000026 seconds into read of 8 bytes without response from unit before it closed connection
```

WillCodeForCats, 2022-05-08:
Please try manually installing the integration from the branch at https://github.com/WillCodeForCats/solaredge-modbus-multi/tree/connect-once

Manually install it and see if that makes any difference. This one is modified to hold the connection open instead of closing and opening a new one every polling cycle, in case the problem is the inverter software has a problem with too many open/close cycles.

gieljnssns, 2022-05-09:
I've installed the `connect-once` version.
After 45min it still works!
Thanks

WillCodeForCats, 2022-05-09:
Great, keep using it for now and I will test it more.

gieljnssns, 2022-05-11:
The `connect-once` version works one day.
Some time after the inverter went to sleep I'm getting an error (2 days in a row).
Yesterday morning I tried to replace te `pymodbus 2.5.3` with `pymodbus 3.0.0.dev3` (I forgot to copy the errors).
This morning I saw there is still an error
```
2022-05-10 22:52:38 DEBUG (MainThread) [pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-05-10 22:52:38 DEBUG (MainThread) [pymodbus.payload] [21359, 27745, 29253, 25703, 25888, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21317, 13104, 12336, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12336, 12339, 11826, 13619, 14080, 0, 0, 0, 14150, 12597, 12341, 13622, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1, 101, 50, 0, 0, 65535, 65535, 0, 2340, 65535, 65535, 65535, 65535, 65535, 65535, 0, 0, 50006, 65533, 0, 0, 0, 0, 10000, 65534, 294, 13856, 0, 0, 0, 14, 65535, 0, 0, 32768, 0, 32768, 32768, 65534, 2, 0, 65535, 65535, 65535]
2022-05-10 22:52:38 DEBUG (MainThread) [pymodbus.payload] [b'\x01&', b'6 ']
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0xe 0x7 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 3591
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0xe 0x7 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x11 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xb 0xb8 0xa0 0x0 0x0 0x0 0x62 0x7a 0x4 0x69 0x62 0x7a 0xd2 0x9 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x9 0xd9 0xb 0x0 0xa 0x6a 0x25 0x62 0x7a 0xec 0xb8 0x0 0x0 0x0 0x0 0x0 0x0 0x6 0xb5
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0xe 0x7 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x11 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xb 0xb8 0xa0 0x0 0x0 0x0 0x62 0x7a 0x4 0x69 0x62 0x7a 0xd2 0x9 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x9 0xd9 0xb 0x0 0xa 0x6a 0x25 0x62 0x7a 0xec 0xb8 0x0 0x0 0x0 0x0 0x0 0x0 0x6 0xb5
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 3591
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 4400, 0, 0, 0, 0, 0, 0, 0, 0, 3000, 40960, 0, 25210, 1129, 25210, 53769, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 9, 55563, 10, 27173, 25210, 60600, 0, 0, 0, 1717]
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\x00']
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\x00']
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\x00']
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\x00']
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x110']
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\x00']
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x00\x00']
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [b'bz', b'\x04i']
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [b'bz', b'\xd2\t']
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [b'\x00\t', b'\xd9\x0b']
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [b'\x00\x00', b'\x06\xb5']
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0xe 0x8 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x10 0x0 0x4
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 3592
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0xe 0x8 0x0 0x0 0x0 0xb 0x0 0x3 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.framer.socket_framer] Processing: 0xe 0x8 0x0 0x0 0x0 0xb 0x0 0x3 0x8 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.transaction] Getting transaction 3592
2022-05-10 22:52:46 DEBUG (MainThread) [pymodbus.payload] [0, 0, 0, 0]
2022-05-10 22:53:16 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] send: 0xe 0x9 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x40 0x0 0x34
2022-05-10 22:53:16 DEBUG (MainThread) [pymodbus.transaction] Adding transaction 3593
2022-05-10 22:53:16 DEBUG (MainThread) [pymodbus.client.asynchronous.async_io] recv: 0xe 0x9 0x0 0x0 0x0 0x6b 0x0 0x3 0x68 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x11 0x30 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xb 0xb8 0xa0 0x0 0x0 0x0 0x62 0x7a 0x4 0x69 0x62 0x7a 0xd2 0x9 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x9 0xd9 0xb 0x0 0xa 0x6a 0x25 0x62 0x7a 0xec 0xcf 0x0 0x0 0x0 0x0 0x0 0x0 0x6 0xb5
[... cut ...]
Error reading inverter at id 1: Modbus Error: [Connection] ModbusTcpClient(192.168.79.56:502): Connection unexpectedly closed 0.000027 seconds into unbounded read without response from unit before it closed connection
Error reading inverter at id 1: Modbus Error: [Connection] ModbusTcpClient(192.168.79.56:502): Connection unexpectedly closed 0.000012 seconds into unbounded read without response from unit before it closed connection
```

Please keep in mind I have 2 modbus integrations so the modbus debug is from 2...

WillCodeForCats, 2022-05-11:
This seems more like a hardware or software bug with the inverter, not the integration of home assistant.

Are multiple things trying to connect to modbus to the same inverter? If so I would suggest only one modbus connection to solaredge inverters.

In the future I plan to optimize some modbus stuff, but I don't really think it would help anything here. The logs clearly show the inverter is dropping the connection. It's not impossible a software fix could do something, but I don't know what that could be and I don't have access to an older inverter to try modifying things like read order, read sizes, etc. to see if the inverter bug appears over time with different types of requests.

For now I'll just have to mark this as "help wanted" and hope someone else has any other ideas.

gieljnssns, 2022-05-24:
I don't think it is a HW or SW bug with the inverter, because I tried this integration, https://github.com/CJNE/ha-sunspec.
And with that one I don't see any errors.

WillCodeForCats, 2022-05-24:
Probably use that one then; I don't have a fix. Sorry.

ChristophCaina, 2022-05-26:
Hi,

SolarEdge Inverters only allow one client to be connected to the Modbus Interface.
This is a "Common" issue, when you have multiple devices that requires data from the Inverter, such as a Wallbox and additionally your HomeAssistant.

A solution for this might be an addon which I just found yesterday - while searching for a workaround to get both, my wallbox and HomeAssistant working with the inverter...
It is called modbus-proxy - a small python tool - and someone has converted this into a HomeAssistant Addon:
[https://github.com/Akulatraxas/ha-modbusproxy](https://github.com/Akulatraxas/ha-modbusproxy)

It might be an idea to take this into consideration... ?!

Ekkib, 2022-05-26:
> SolarEdge Inverters only allow one client to be connected to the Modbus Interface.

I also read this info. Maybe this was really the case in the past with
a specific solaredge firmware ?

This is not true for at least the SE7k firmware
- 3.2251
- 3.2537

I recommend to try out, if this rumor is still true for your specific firmware.

Home-assistant and the firmware of the cfos Wallbox are reading out
the SolarEdge inverter in parallel.

Home-Assistant is not complaining at all and delivers correct values.

cfos has some issues from time to time. Maybe these issues should
simply be filtered away.

Best regards
Ekkehard

ChristophCaina, 2022-05-26:
Hi @Ekkib, thanks for that information.
I just had a chat with the SolarEdge support recently.
They confirmed, that max one connection is allowed - and that they can't provide any information IF or WHEN this might change.

I think, it might be related to the inverter itself... what generation it is, and so on...

Where can I get the Formware of the Inverter?

I only see the following Information:
```
DSP 1 Version 	1.13.1938
DSP 2 Version 	2.19.1511
CPU Version 	4.15.119
```

Ekkib, 2022-05-26:
> Where can I get the Firmware of the Inverter?

I can see this information on the "old" SE7k on the 4 line Display.

Most recent :

TD:7E143DCA 99
D5P1/2:1.0013/2.0019
CPU: 3.2537
Land: GER

Previous :

ID: 7E143DCA 99
D5P1: 1.0013.1031
DSP2: 2.0019.0922
CPU: 3.2251.0000

ChristophCaina, 2022-05-26:
Thanks :)
So my inverter does not have a display... but it seems to be the CPU Version - and based on your information, mine seems to be newer than yours.
Anyway - the SolarEdge support confirmed, that only one client can connect to the inverter.

MAYBE:
There's another case:

I think, the inverter does have two interfaces.
One is being used for my meter - and the other one will be used by the wallbox OR homeassistant - depending, which system connects first.

WillCodeForCats, 2022-06-12:
[Pre-Release v2.0.0-pre.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.0.0-pre.4) has all new read methods with smaller register request blocks.

**v2.0.0 Pre-Release Notes**: [Release v2.0.0-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.0.0-pre.1)

WillCodeForCats, 2022-07-03:
PR #54 will add an option to keep the modbus connection open after each polling cycle.

WillCodeForCats, 2022-08-01:
[Release v2.1.0-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.1.0-pre.1)

This pre-release changes the polling architecture to use the Home Assistant Data Update Coordinator. It's a general improvement not targeted at any specific problem, but I would recommend trying it in case somehow it helps.

WillCodeForCats, 2022-08-21:
I've created a pre-release for the latest pymodbus dev version: [v2.1.2-pymodbusdev.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.1.2-pymodbusdev.1)

WillCodeForCats, 2022-08-24:
I'm linking this issue to PR #81 and will close this when it is merged since there has not been any test feedback on this issue with the pre-releases, and the issues appear similar.

closed 2022-08-24 by WillCodeForCats

WillCodeForCats, 2022-09-29:
Also see issue #80

WillCodeForCats, 2022-10-12:
pymodbus has released v3.0.0

I have updated the pymodbus-dev branch to support the latest release version. You can download the branch to try it manually: https://github.com/WillCodeForCats/solaredge-modbus-multi/archive/refs/heads/pymodbus-dev.zip

WillCodeForCats, 2022-10-24:
pymodbus v3.0.0 is now in [Release v2.2.6-pre.3](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.6-pre.3)