WillCodeForCats/solaredge-modbus-multi #26: Enhancement request

Benny-Git, 2022-04-27:
**Describe the ~bug~ enhancement request**
I set up the integration and like it. However I wanted to now lower the scan interval, and could not do so via the UI, though the setup worked fine via the UI.

**Expected behavior**
The UI config should allow me to make changes to the existing integration config.

**Screenshots**
n/a

**Logs**
n/a

**Home Assistant (please complete the following information):**
 - Version: 2022.4.7

**Additional context**
I could change the scan_interval by editing my .storage/core.config_entries, but this is not as comfortable as using a config UI.

WillCodeForCats, 2022-04-27:
Home Assistant unfortunately doesn't support editing or updating configuration in the UI, but I could move poll interval to the "options" side which shows up when you click the "configure" button for the integration.

Here's the bug in HA: https://github.com/home-assistant/architecture/issues/377 (Ability to edit/update config entries)

Benny-Git, 2022-04-27:
Interesting. I know the "configure" button on some of my integrations, but didn’t know it was very distinct from the initial config.
So moving this to the options means a whole lot of rework?

WillCodeForCats, 2022-04-27:
It's not a lot of work, it can be done.

WillCodeForCats, 2022-05-07:
To do after release v1.2.0

WillCodeForCats, 2022-05-19:
This feature is complete and will be released as v1.2.1 shortly.

closed 2022-05-19 by WillCodeForCats