WillCodeForCats/solaredge-modbus-multi #28: After install, I cannot find it in integration

blowk, 2022-05-05:
Before reporting a bug, make sure your configuration is correct and your LAN is working.
https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

**Describe the bug**
After installing it through HACS and reboot, I cannot find the integration. The onlye integration I find is the one Solaredge API.

**Expected behavior**
Find the SolarEdge (local)

**Screenshots**
*

**Logs**
*

**Home Assistant (please complete the following information):**
 - Version: 2022.5.0 (Fresh install without any modules installed)

WillCodeForCats, 2022-05-05:
No problem on my test system.

blowk, 2022-05-05:
Ok, thanks.
I will try to reinstall it.
Can you send me a print screen of what I need to see in the integration when searching for it?

Is it possible to set the configuration manual in yaml?

Flisan2010, 2022-05-05:
I have the same problem.
Downloaded via HACS, tried to reinstall a several times but it never shows under Integrations (only Solaredge API).
Have version 2022.5.0 samt HACS 1.24.5.

WillCodeForCats, 2022-05-06:
> Is it possible to set the configuration manual in yaml?

No.

closed 2022-05-06 by WillCodeForCats