WillCodeForCats/solaredge-modbus-multi #34: 1.2.1 upgrade, less entities

mvillarejo, 2022-05-29:
Hello,

thank you for this integration is really great. 

I just followed the upgrade process and it worked really well, no duplicated entities. However, I had to rollback to the previous setup mainly because the new integration only detected 30 entities rather than the 100 that previous one detected.

not sure where the issue can be but would be able to point me to what am I missing in the configuration of it? I did it a few months back but just remember to use the default (I have only one inverter and no battery).

Thank you!

WillCodeForCats, 2022-05-29:
There are no code changes in the rename, and 1.2.1 only adds a config feature to change the polling interval. The maximum number of entities per inverter is 31, with 2 of those treated as "advanced" types and disabled by default.

I don't know where 100 entities would come from unless you have multiple inverters and a meter.

mvillarejo, 2022-05-29:
There’s a inverter and a meter. I’ll try again but couldn’t see any disabled entities, I can try to collect some logs/debug files.

WillCodeForCats, 2022-05-29:
If there's a meter you need to check meter 1 in the setup screen when you add the integration.

WillCodeForCats, 2022-05-30:
[image]

mvillarejo, 2022-05-30:
That seemed to be the case, migration worked great. 
Resolving

closed 2022-05-30 by mvillarejo