WillCodeForCats/solaredge-modbus-multi #39: Language Translations

WillCodeForCats, 2022-06-10:
Looking for better language translations. The current translations are just the English version run through Google Translate.

Please help by reviewing and submitting updated translations, or new ones for languages we don't have yet.

tuxedo309, 2022-06-10:
Hi! I can translate to spanish

JefvdA, 2022-06-11:
I am dutch, so made a few adjustments to the existing translations.

Opened a PR: #40

WillCodeForCats, 2022-06-11:
> Hi! I can translate to spanish

Sure, just translate the file `custom_components/solaredge_modbus_multi/strings.json` and either submit a pull request or just put it here and I'll create a translation file for it.

bdkacz, 2022-07-02:
Polish translation:

```
{
  "config": {
    "step": {
      "user": {
        "title": "Konfiguracja SolarEdge Modbus",
        "data": {
          "name": "Prefix sensora",
          "host": "Adres IP inwertera ",
          "port": "Modbus/TCP Port",
          "device_id": "Adres Modbus Inwertera (Device ID)",
          "number_of_inverters": "Ilość inwerterów",
          "read_meter_1": "Użyj  licznika 1",
          "read_meter_2": "Użyj licznika 2",
          "read_meter_3": "Użyj licznika 3"
        }
      }
    },
    "error": {
      "already_configured": "Urządzenie jest już skonfigurowane!",
      "max_device_id": "Device ID musi być pomiędzy  1 i 247.",
      "min_device_id": "Device ID musi być pomiędzy  1 i 247.",
      "max_inverters": "Dopuszczalna liczba inwerterów to od  1 do 32.",
      "min_inverters": "Dopuszczalna liczba inwerterów to od  1 do 32.",
      "too_many_inverters": "Liczbainwerterów za duża dla Device ID.",
      "invalid_host": "Błędny adres IP.",
      "invalid_tcp_port": "Dozwolony zakres portów to od  1 do 65535."
    },
    "abort": {
      "already_configured": "Urządzenie jest już skonfigurowane!"
    }
  },
  "options": {
    "step": {
      "init": {
        "title": "Opcje SolarEdge Modbus",
        "data": {
          "scan_interval": "Częstotliwość odczytu (sekundy)",
          "single_device_entity": "Użyj jednej encji dla urżadzenia",
          "detect_meters": "Automatycznie wykryj liczniki",
          "detect_batteries": "Automatycznie wykryj baterie"
        }
      }
    },
    "error": {
      "invalid_scan_interval": "Próbkowanie musi być w zakresie od 10 do 86400 sekund."
    }
  }
}
```

WillCodeForCats, 2022-07-02:
Thank you @bdkacz, created PR #52

WillCodeForCats, 2022-07-03:
@bdkacz Can you translate `"Keep Modbus Connection Open"` to Polish?

bdkacz, 2022-07-03:
> @bdkacz Can you translate `"Keep Modbus Connection Open"` to Polish?

"Pozostaw połączenie Modbus otwarte"

WillCodeForCats, 2022-07-10:
PR #55 for German translation updates.

Any updates to existing translations is welcome.

gferen, 2022-08-23:
I could help with Slovenian and Bulgarian language.

WillCodeForCats, 2022-08-31:
@gferen You can either submit a pull request by creating a file in the `translations` directory, or I can add it if you translate this:

```{
  "config": {
    "step": {
      "user": {
        "title": "SolarEdge Modbus Configuration",
        "data": {
          "name": "Sensor Prefix",
          "host": "Inverter IP Address",
          "port": "Modbus/TCP Port",
          "device_id": "Inverter Modbus Address (Device ID)",
          "number_of_inverters": "Number of Inverters"
        }
      }
    },
    "error": {
      "already_configured": "Device is already configured!",
      "max_device_id": "Device ID must be between 1 to 247.",
      "min_device_id": "Device ID must be between 1 to 247.",
      "max_inverters": "Must be between 1 to 32 inverters.",
      "min_inverters": "Must be between 1 to 32 inverters.",
      "too_many_inverters": "Number of inverters too high for Device ID.",
      "invalid_host": "Invalid IP address.",
      "invalid_tcp_port": "Valid port range is 1 to 65535."
    },
    "abort": {
      "already_configured": "Device is already configured!"
    }
  },
  "options": {
    "step": {
      "init": {
        "title": "SolarEdge Modbus Options",
        "data": {
          "scan_interval": "Polling Frequency (seconds)",
          "single_device_entity": "Single Device Entity",
          "keep_modbus_open": "Keep Modbus Connection Open",
          "detect_meters": "Auto-Detect Meters",
          "detect_batteries": "Auto-Detect Batteries"
        }
      }
    },
    "error": {
      "invalid_scan_interval": "Valid interval is 1 to 86400 seconds."
    }
  }
}
```

WillCodeForCats, 2022-10-25:
New options were added in PR #147 which means translations had to be updated. Anyone who can check them for quality please do so and submit any corrections.

closed 2022-12-15 by WillCodeForCats