WillCodeForCats/solaredge-modbus-multi #42: Help with reading the sensor values

muenchris, 2022-06-15:
This is not a bug report but rather a request for better understanding.

Background:
I have two inverters with one battery connected to each inverter.
What I want is to know at any given time:
How many power is being produced by my solar cell per inverter
How much power is being exported to the grid (per inverter or in total)
How much power is being used by my home (per inverter or in total)
How much power is being imported from the grid (per inverter or in total)
How much power is being used to charge my batteries (per inverter)
How much power is used from my batteries (per inverter).

I do want these values in "power" (Watts) to visualize the actual flow of power.

At the end of a day I am also interested in the energy (kwh) of
Energy Exported to Grid (Net-Metering)
Energy Used to feed the home
Energy used to charge the batteries
Energy Imported from Grid
Energy produced by Solar

I am sure all the values are there, but its hard to figure out exactly what values need to be calculated in what way to get to these numbers. 
All my values collected are exported via Prometheus to Grafana and visualization happens there. HASS is just the "collector".

Any help is very appreciated

WillCodeForCats, 2022-06-15:
You will need to add at least one meter to get import/export or self consumption data. If you have a consumption meter you can calculate import/export, and if you have an import/export meter you can calculate self consumption.

muenchris, 2022-06-15:
I do have 3 meters on my main inverter and one on my second inverter
Your integration finds two meters on the main inverter (M1 and M2) and the one on my second inverter (M1):

This is the current live output on HASS Prometheus:
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_i1_ac_power",friendly_name="Solaredge I1 AC Power"} 13.475
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_i1_dc_power",friendly_name="Solaredge I1 DC Power"} 13.681
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_i2_ac_power",friendly_name="Solaredge I2 AC Power"} -11.206
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_i2_dc_power",friendly_name="Solaredge I2 DC Power"} -11.376
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_m1_ac_power",friendly_name="Solaredge M1 AC Power"} -8.0
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_m1_ac_power_a",friendly_name="Solaredge M1 AC Power A"} 0.0
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_m1_ac_power_b",friendly_name="Solaredge M1 AC Power B"} 0.0
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_m1_ac_power_c",friendly_name="Solaredge M1 AC Power C"} -8.0
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_m2_ac_power",friendly_name="Solaredge M2 AC Power"} -2337.0
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_m2_ac_power_a",friendly_name="Solaredge M2 AC Power A"} -723.0
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_m2_ac_power_b",friendly_name="Solaredge M2 AC Power B"} -1613.0
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_m2_ac_power_c",friendly_name="Solaredge M2 AC Power C"} 0.0
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_m1_ac_power_2",friendly_name="Solaredge M1 AC Power"} -50.0
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_m1_ac_power_a_2",friendly_name="Solaredge M1 AC Power A"} 0.0
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_m1_ac_power_b_2",friendly_name="Solaredge M1 AC Power B"} 0.0
hass_sensor_power_w{domain="sensor",entity="sensor.solaredge_m1_ac_power_c_2",friendly_name="Solaredge M1 AC Power C"} -49.0

WillCodeForCats, 2022-06-15:
If you look under the device (or in the device entity sensor attributes) it will tell you what kind of meter they're configured as so you know how to use the data.

Example:
[image]

As far as actually calculating all the info, I don't have any example of that myself but I would be happy to add examples to the wiki if someone has some.

muenchris, 2022-06-15:
yes I see this..both my M1s are "Production" and the M2 says "Import/Export" (Firmware 75):

Device info
SE-MTR-3Y-240V-A
by SolarEdge
Firmware: 75
Hardware: Export+Import