WillCodeForCats/solaredge-modbus-multi #45: Integration Missing from Devices & Services

mbillings6, 2022-06-22:
I installed the repo via HACS and then rebooted HA. after it came back online I went to Devices & Services to add the integration. The only integrations for SolarEdge available where the normal SolarEdge and SolarEdge Modbus. The SolarEdge Modbus Multi wasn't listed. Also, this is the a clean install of SolarEdge Modbus Multi, never been installed. I am running Hassio on a RPI4B with version 2022.6.6 and HACS version 1.25.5. Let me know if you need anything to further investigate the issue.

[image]

[image]

WillCodeForCats, 2022-06-22:
What version did you install?

WillCodeForCats, 2022-06-22:
I tried it on my HACS test VM and I had to reload my browser before it would show up in Home Assistant's integration setup menu.

mbillings6, 2022-06-22:
I installed v2.0.0, i will give the reload a try. I rebooted hassio several times after the HACS download to no gain.

mbillings6, 2022-06-22:
I ended up closing my browser out, that did the trick. must be some cache issue with hassio itself.

WillCodeForCats, 2022-06-22:
I also see when I remove it I have to reload the browser to get it to disappear. My guess is the integration list is cached by the browser and HA isn't dynamically updating it.

WillCodeForCats, 2022-06-22:
Will add note to reload/restart browser in docs.

closed 2022-06-22 by WillCodeForCats