WillCodeForCats/solaredge-modbus-multi #48: Inverter device ID 1 not found.

Robby-Swartenbroekx, 2022-06-23:
It's a completely new installation.
If I install 2.0.0 and manually add the IP of the inverter, I get every time the error "Inverter device ID 1 not found."

I removed the devices, downgraded to 1.2.2 and it recognises 238 entities (don't know if they all work, I'm not at that part at the moment).

I'm on the latest versions, I think:
Home Assistant Core 2022.6.7
HA Supervisor: 2022.05.3
HA OS 8.2
HACS 1.25.5

WillCodeForCats, 2022-06-23:
Please use the bug report template, info is missing that it asks for.

Robby-Swartenbroekx, 2022-06-29:
Below the template:

Before reporting a bug, make sure your configuration is correct and your LAN is working.
https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

**Describe the bug**
A clear and concise description of what the bug is.
I installed it with the fixed ip I gave my inverter and I put in ID 1 as this is the default provided ID.
I geth the error message "Inverter device ID 1 not found."
Doesn't work with other ID's either.

**Expected behavior**
A clear and concise description of what you expected to happen.
I expect it to go through with the installation and detecting of all parameters of the invertor

**Screenshots**
If applicable, add screenshots of the problem and/or configuration.
No screenshot at this time available. Can create one later.

**Logs**
Copy any log messages/errors from Home Assistant from "custom_components.solaredge_modbus" and "pymodbus.client.sync".
I'll check when I'm back home. This is probably the crucial info you were referring to that was missing.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2022.6.7
 - solaredge-modbus-multi Version: 2.0.0

**Additional context**
Add any other context about the problem here.

Robby-Swartenbroekx, 2022-06-29:
I already found out that the default ID 1 isn't good.
I've set it now to 2, and I do get data back. Also the message is there again, now about ID 2, but after a few minutes it dissapears.
So it probably is just me that didn't had enough knownledge about what I was doing at the moment.

Also I'm still missing the info of my current power usage. There was an extra meter coupled at the inverter to get this info. But this is clearly not a bug (in my case) and I'm just going to look for more docs where I can get all my info.