WillCodeForCats/solaredge-modbus-multi #50: Platform solaredge_modbus_multi does not generate unique IDs.

purcell-lab, 2022-07-02:
Everything appears to be working correctly. I have two inverters and one meter which are all correct.

**Describe the bug**
I get an error message in the logs about unique IDs, which I suspect is related to 2022.7 beta release.

**Expected behavior**
Integration should generate unique IDs and not report an error message.

**Screenshots**
[image]
[image]
[image]
[image]

**Logs**
Copy any log messages/errors from Home Assistant from "custom_components.solaredge_modbus" and "pymodbus.client.sync".

Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:620
Integration: Sensor (documentation, issues)
First occurred: 11:54:24 AM (28 occurrences)
Last logged: 11:54:24 AM

Platform solaredge_modbus_multi does not generate unique IDs. ID SE-MTR-3Y-400V-A_606773955_exported_c_kwh already exists - ignoring sensor.solaredge_m1_exported_c_kwh
Platform solaredge_modbus_multi does not generate unique IDs. ID SE-MTR-3Y-400V-A_606773955_imported_kwh already exists - ignoring sensor.solaredge_m1_imported_kwh
Platform solaredge_modbus_multi does not generate unique IDs. ID SE-MTR-3Y-400V-A_606773955_imported_a_kwh already exists - ignoring sensor.solaredge_m1_imported_a_kwh
Platform solaredge_modbus_multi does not generate unique IDs. ID SE-MTR-3Y-400V-A_606773955_imported_b_kwh already exists - ignoring sensor.solaredge_m1_imported_b_kwh
Platform solaredge_modbus_multi does not generate unique IDs. ID SE-MTR-3Y-400V-A_606773955_imported_c_kwh already exists - ignoring sensor.solaredge_m1_imported_c_kwh

**Home Assistant (please complete the following information):**
- Home Assistant 2022.7.0b2
- Supervisor 2022.06.2
- Frontend 20220630.0 - latest
 - solaredge-modbus-multi Version: 2.0.0

**Additional context**
I deleted the old integrations before installing 2.0.0, all the history came across correctly.

WillCodeForCats, 2022-07-02:
I tried installing core-2022.7.0b2 on my dev virtual machine, but was not able to duplicate the issue.

WillCodeForCats, 2022-07-02:
Did you manually rename the device to "Solaredge M2" or did it create it like that? It should be M1.

WillCodeForCats, 2022-07-02:
Can you also provide all of the attributes from entity `sensor.solaredge_m1_device`

WillCodeForCats, 2022-07-02:
I added a check to ignore duplicate meter serial numbers, in case the same meter is being reported multiple times. The unique ID relies on the device model and serial number, where the serial number must be unique to each device model.

Please try this branch: https://github.com/WillCodeForCats/solaredge-modbus-multi/archive/refs/heads/50-platform-solaredge_modbus_multi-does-not-generate-unique-ids.zip

purcell-lab, 2022-07-02:
> Did you manually rename the device to "Solaredge M2" or did it create it like that? It should be M1.

No I didn't rename anything, this is how it auto configured.

purcell-lab, 2022-07-02:
> Can you also provide all of the attributes from entity `sensor.solaredge_m1_device`

Here are the attributes.

[image]

WillCodeForCats, 2022-07-02:
The only thing I can think of is that somehow the same meter is being reported more than once, which the branch should fix by ignoring meters where a device was already created with the same serial number.

closed 2022-07-02 by WillCodeForCats

purcell-lab, 2022-07-02:
2.0.1 Has opened up an new error message for me. 

Functionality does't appear impacted.

```
This error originated from a custom integration.

Logger: custom_components.solaredge_modbus_multi.hub
Source: custom_components/solaredge_modbus_multi/hub.py:105
Integration: SolarEdge Modbus Multi-Inverter (documentation, issues)
First occurred: 9:06:50 AM (1 occurrences)
Last logged: 9:06:50 AM

Duplicate serial 606773955. Ignoring meter 2 on inverter ID 1
```
[image]

WillCodeForCats, 2022-07-02:
That is just saying there was a duplicate found and it was ignored. Everything looks correct to me.

The issue of why the same meter is being reported more than once would be an issue to look into on the hardware side (is the meter somehow connected to both inverters?) but otherwise not an error with the integration.

WillCodeForCats, 2022-07-04:
The duplicate detected message will now display at `warning` level instead of `error` level in release v2.0.2.