WillCodeForCats/solaredge-modbus-multi #56: Support for inverters with no consecutive IDs

rzulian, 2022-07-12:
I my system I've two inverters, the first one has ID 1, the other one ID 3.
I'm currently testing this component with an hard coded 
```

        for inverter_index in [1,3]:
            inverter_unit_id = inverter_index # + self.start_device_id
 ```
in line 89 ```hub.py```
I'm happy to contribute on this project and add support for this scenario. What about a list for IDs? Or multiple ID entries?

WillCodeForCats, 2022-07-12:
Why can't you use ID 2?

rzulian, 2022-07-12:
ID 1 is the AC couplig inverter
ID 2 is the meter
ID 3 is the 3 Phase inverter
ID 15 is the battery

Inverter 1 is master. Inverter 3 slave
Inverter 1 is multi device, having the meter and the battery connected to 1

WillCodeForCats, 2022-07-12:
Meter IDs do not conflict with inverter IDs because they would be on different RS485 busses in a SolarEdge system. This is explained in the documentation with examples:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

WillCodeForCats, 2022-07-12:
The documentation examples are from my personal system where I have inverters ID 1 (master) and 2 (slave) and meter ID 2  is connected to inverter 1.

Up to 32 inverters can be supported, so I don't think it's practical to add 32 inverter address options in the Home Assistant setup screen (or how it would even handle formatting to fit that many on screen).

thargy, 2023-07-19:
My inverter ID is also 3, this was setup during the install of my new system because, according to Solaredge (on support call from installer), 1 & 2 are reserved for masters and slave IDs had to start from 3.  When it was set to 2 it did not work.