WillCodeForCats/solaredge-modbus-multi #57: Connection time out

rzulian, 2022-07-12:
I'm getting a lot of connection time out using a polling interval below 10 secs.  I was getting the same problems with other component for modbus-solaredge.
I think that this is related to the new HA version 7.2, and I was hoping that this component wasn't affected using a much more recent ```pymodbus``` library.

[image]

Do you have the same problems?

rzulian, 2022-07-12:
I need to investigate what's wrong on my side then ...
Thanks

rzulian, 2022-07-13:
I was suspecting that I had a performance problem in my HA in a VM. So I've tested  a dev container with only this component, but I'm still unable to set the polling to less than 10 secs. @WillCodeForCats  Any idea on what I can check?

WillCodeForCats, 2022-07-13:
How long did you wait after changing the time? Let it wait for 2 minutes and see if it's just an issue with the inverter thinking the old connection is still open.

WillCodeForCats, 2022-07-16:
PR #59 might help with the immediate connection error after reconfigure.

rzulian, 2022-07-21:
Are you going to merge also the ``data_update_coordinator``. I would test the new main with also the battery support

WillCodeForCats, 2022-07-21:
> Are you going to merge also the `data_update_coordinator`. I would test the new main with also the battery support

No, I'm not.

rzulian, 2022-07-21:
BTW, digging into a very slow trigger issue, I was discussing about the possibility that the solaredge_modbus can be athe problem. In the discord channel they suggested me to "It looks like that package does the same thing. It should be an easy fix, you can just change line 187 there to use an executor job like "update_result = await self._hass.async_add_executor_job(self._read_modbus_data)"
But the better way to do it is to use the asyncio methods which the pymodbus library seems to have".

WillCodeForCats, 2022-07-21:
I have been looking at converting it to use the async client of pymodbus. The first step is to use the data update coordinator then probably async pymodbus later. The branch is there but it doesn't work yet. Once I get it to work with sync then I will work on async.

Because it currently uses sync if you have a fast polling interval it can halt all home assistant execution while reading or waiting for a modbus timeout, so fast polling is not advised.

WillCodeForCats, 2022-07-29:
The `data-update-coordinator` branch is working now, you could try it out if you want. It has not been tested, I've only done enough work so that it loads and runs.

WillCodeForCats, 2022-07-29:
I should have all of the sync modbus calls covered with async_add_executor_job as of commit d8e32540b0eaa363e96d03c0cfd5c2674d460964

rzulian, 2022-07-29:
I can test it early next week

WillCodeForCats, 2022-07-31:
I've created a pre-release v2.1.0-pre.1

WillCodeForCats, 2022-08-21:
I've created a pre-release for the latest pymodbus dev version: [v2.1.2-pymodbusdev.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.1.2-pymodbusdev.1)

WillCodeForCats, 2022-10-12:
pymodbus has released v3.0.0

I have updated the pymodbus-dev branch to support the latest release version. You can download the branch to try it manually: https://github.com/WillCodeForCats/solaredge-modbus-multi/archive/refs/heads/pymodbus-dev.zip

closed 2022-10-24 by WillCodeForCats