WillCodeForCats/solaredge-modbus-multi #61: Find another Energymeter

thewizard-de, 2022-07-24:
Hi,

first the SE-RGMTR-1D-240C-A was found and i got the sensor values. A day later no values were available and another sensor pop up
SE-MTR-3Y-400V-A (and this one is really installed) 

[image]

WillCodeForCats, 2022-07-29:
The integration can only report what the inverter tells it. Maybe something rebooted or got a firmware update. I would probably not worry about this.

skl283, 2022-07-31:
Had the same issue... only a remove of the integration and reconfigure helped to get rid of the 2nd meter.
[image]
[image]
[image]

skl283, 2022-08-06:
@WillCodeForCats do you have any hints to stop getting another meter in the config? This lets my Integration to not work anymore :-(

WillCodeForCats, 2022-08-06:
Can you post all of your device attributes?

Go to: Developer Tools -> States

Search for: `solaredge*device`

Remko76, 2022-08-07:
Had the same issue yesterday, all of a sudden an extra meter showed up and I had to delete and reconfigure the integration.
These are my devices:
[image]

Remko76, 2022-08-07:
[image]

Remko76, 2022-08-07:
And happened again today. These are the devices when it created the second meter

[image]

WillCodeForCats, 2022-08-07:
What is the real model label on the meter?

Remko76, 2022-08-07:
I need to check. I am not at home currently, so will not be able to get that information until next week.

WillCodeForCats, 2022-08-07:
Is there only one meter on the RS485 bus ore more than one meter?

Remko76, 2022-08-07:
Only one meter. Nothing else connected to the modbus.

Remko76, 2022-08-08:
Ok, I was able to find which meter is connected. It is the SE-MTR-3Y-400V-A

So my problem is exactly the same as from the OP. Initially it detects the meter with a wrong identification and after some time (unknown how much time; it can be good for weeks or the issue shows up every day like this weekend) it "finds" the correct identification and we end up with the 2 meters in the integration.

WillCodeForCats, 2022-08-08:
The integration only reports the data the inverter tells it. The question is why is the inverter reporting one model number then changing it later.

You could try looking at the inverter configuration and make sure it only has one meter in configuration and it's the correct one. The meter setting screens for a SetApp inverter are here:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration#meter-settings-on-an-inverter

WillCodeForCats, 2022-08-08:
Also, it is possible for a hidden meter to be inside the inverter plus an external meter for import/export or consumption monitoring. The internal meter is usually set up as revenue grade production for higher accuracy than the inverter's self reporting.

Remko76, 2022-08-08:
I used Setapp in the past to check all the configuration options and it only has one meter configured (the right one). So maybe it is something with this "hidden meter"....the question is...what do I do with this now? Do I wait till the second meter shows up again and use that in my configuration and ignore the "unavailable" one...? 
I used to use the original integration and I never had this issue at all...

WillCodeForCats, 2022-08-08:
If there is a revenue grade meter inside the inverter (that's what SE-RGMTR means: "revenue grade" meter), you would probably configure it as a "production" meter in setapp. You would need to take the covers off the inverter and anything else related to look inside to see if it came with a built in meter.

The other integration just polls registers and ignores device information. It doesn't try to detect anything either. Since the model randomly changing will conflict with the auto-detection feature, I would have to consider what options I could come up with to disable auto-detect and ignore device information.

Remko76, 2022-08-08:
I will check what is inside when I get back home, but as I said... will take another week at least. I will see what happens if it detects the second meter again and if I can just use that in my configuration and ignore the other one. I will update here if I find out more.

Remko76, 2022-08-08:
One other interesting fact....after I deleted and reinstalled the integration it now shows my meter as the correct one right away. I do not have that  SE-RGMTR now. Maybe....just maybe it detected the right meter as the first one now. I'll keep an eye on it.

WillCodeForCats, 2022-08-08:
It just seems strange to me that the inverter would randomly change what device information it reports. I would think it has to get that data from somewhere.

Remko76, 2022-08-08:
Yes, I also do not really understand it and I would like to get to the bottom of it. Since I am not the only one running into this issue, there must be something going on.

skl283, 2022-08-08:
Next Time the issue will occur, i'll try to add logs.
Also in my case - only 1 Meter ist installed, but sometimes the Addon gets a 2nd Meter.

FranciscoBigWings, 2022-08-10:
Hello,

I have same issue as describe, this is happening to me not since the last update but since the previous one. At I didnt update the inverters manually, if they were updated the update was OTA but I dont really know because I dont have tracking of the firmware versions over time.

This is issue, as mentioned before, happens randomly and it solves by just but deleting the integration and adding it again. I have two inverters and one battery. Within the detected devices, the integration doesnt show the model of my meter, only shows WattNode in the Manufacturer column

skl283, 2022-08-10:
in my case it was after a reboot of an Update
[image]

skl283, 2022-08-10:
after delete and reconfigure it is ok:
[image]

WillCodeForCats, 2022-08-10:
Device detection is only done once when the integration is loaded, either at startup or manual reload.

So far everyone with the problem has an SEnK-RWS model inverter (I have meters but a different model inverter).

Inverter firmware updates would be shown with a change in the `solaredge_i1_version` sensor.

skl283, 2022-08-10:
In my case i have the latest FW installed.

WillCodeForCats, 2022-08-11:
I've thought about this for a bit and I think the best option to keep auto detection and still work if the meter identification reported by the inverter changes for unknown reasons is to not rely on the meter serial number and model number. These are currently used for the entity registry Unique ID in Home Assistant. When the Unique ID changes because the model number changed, the issue of seeing a "new" meter and new entities will happen.

Since this is not being reported as a problem with the inverter's device information (otherwise you would see the same thing with a new/different inverter being detected) I think it's still safe to rely on the inverter's device information as a base for the unique id.

WillCodeForCats, 2022-08-11:
Just as another point of data the meter model change has never happened to me, but I opened my meter box and mine is labeled as a SE-RGMTR-3Y-277V-A but it's reported as SE-RGMTR-3Y-208V-A in Home Assistant.

skl283, 2022-08-11:
Could be a Solution to detect the meter Inlay by adding the Integration not by every bootup? Or what do you think could be to fix this issue?

WillCodeForCats, 2022-08-12:
The fix will be to rely on different values that are stable for the unique id.

WillCodeForCats, 2022-08-17:
@FranciscoBigWings What model inverter do you have (from sensor.solaredge_i1_device)?

FranciscoBigWings, 2022-08-17:
i1 is SE5000H-RW000BNV4
i2 is SE5000H-RWS00BNO4

Francisco F. Mayoralas

WillCodeForCats, 2022-08-23:
PR #83 is my solution to the issue of duplicate meters.

FranciscoBigWings, 2022-08-23:
> PR #83 is my solution to the issue of duplicate meters.

Great. Thank you a lot for your dedication. I will update within the next days and test it in time. If it happens again I will update you

WillCodeForCats, 2022-08-23:
PR #83 is based on PR #81 so it also contains a bunch of updates to modbus error handling.

WillCodeForCats, 2022-08-24:
I have [Release v2.2.0-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.0-pre.1) for this issue. Please read the release notes since this is a breaking change.

Remko76, 2022-08-24:
Thanks for your effort. Just installed it and everything seems to work fine so far. I will update if I see any problems.

closed 2022-08-26 by WillCodeForCats