WillCodeForCats/solaredge-modbus-multi #80: ModbusTcpClient(redacted:1502): Connection unexpectedly closed

bakboord4, 2022-08-21:
Basically there two, connected, error messages (copies from Log)

1.
2022-08-20 23:08:40.007 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Error fetching SolarEdge Coordinator data: Failed to update devices: Modbus Error: [Connection] ModbusTcpClient(.......:1502): Connection unexpectedly closed 0.000131 seconds into read of 8 bytes without response from unit before it closed connection

2.
2022-08-21 05:54:43.009 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Error fetching SolarEdge Coordinator data: Failed to update devices: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
2022-08-21 05:55:46.007 ERROR (SyncWorker_5) [pymodbus.client.sync] Connection to (......., 1502) failed: timed out
2022-08-21 05:56:49.006 ERROR (SyncWorker_0) [pymodbus.client.sync] Connection to (......., 1502) failed: timed out
....... continuing...

Solaredge thru modbus is unreachable. No data is coming into HA. Solaredge entities are: not available.
Getting control back by either by resetting or reinstall of the integration Solaredge.
Reconfigured the WiFI connection with my LAN brought no solution.
Every time within a few hours after a restart, both errors appear.

Solaredge Support updated my Solaredge and stated that the transformer is operating ok.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2022.8.6 and .5
 - solaredge-modbus-multi Version:  2.1.2 also 2.1.1. The integration is standard configured.

Please, do you have  a solution?

WillCodeForCats, 2022-08-21:
First try turning on the "Keep Modbus Connection Open" option.

bakboord4, 2022-08-21:
Thanks for the reply.
Turned Keep .... open on.
Will report tomorrow after 24 hrs.

WillCodeForCats, 2022-08-21:
If that doesn't help, I've also created a pre-release for the latest pymodbus dev version: [v2.1.2-pymodbusdev.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.1.2-pymodbusdev.1)

If you are using HACS you will need to turn on "show beta versions" for the integration in HACS (you can do this by choosing "redownload")

bakboord4, 2022-08-21:
@WillCodeForCats. Will keep your suggestion in mind.

A Status update.
Applied Keep ..connection open. 
In the last hour Solaredge status is alternating per 300 sec (call freq.) between status: 4 (production) and status: not available and per 10 minutes an error: 
Error fetching SolarEdge Coordinator data: Failed to update devices: Modbus Error: [Connection] ModbusTcpClient(192.168.2.119:1502): Connection unexpectedly closed 0.000475 seconds into read of 8 bytes without response from unit before it closed connection

WillCodeForCats, 2022-08-21:
We can randomly try things but if the inverter is dropping the connection during the read there's probably no actual fix I can come up with.

I don't use wifi so if there's a network problem related to wifi anywhere in the communication path I probably won't see the issue personally.

WillCodeForCats, 2022-08-21:
Did you try the pymodbusdev.1 pre release version?

bakboord4, 2022-08-21:
@WillCodeForCats.
Not yet. Will do that tomorrow. 
I am also contemplating how to connect Solaredge with a cable instead of WiFi.

WillCodeForCats, 2022-08-21:
I've created [Release v2.1.3-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.1.3-pre.1) which adds disconnect calls on modbus read errors. The idea is that if there is an error it will make sure the connection is closed and re-opened on the next attempt.

bakboord4, 2022-08-21:
@WillCodeForCats.
Tomorrow is this evening. Installed V2.1.2-pymodbusdev.1 pre.
20 min Solaredge status: in production, after that, alternation of status started again: 5 min in production and 5 min not available. No improvement.
Just saw your 2.1.3-pre.1. Will try it.

WillCodeForCats, 2022-08-21:
Just to be clear, read errors can't be prevented by the integration. I can only change how the integration handles errors.

bakboord4, 2022-08-21:
@WillCodeForCats 
Applied  2.1.3-pre.1. It looks good so far. No alternation of statuses seen.
Let see what the performance will be the coming 24 hrs.
Support for your focus on error handling.

bakboord4, 2022-08-22:
@WillCodeForCats 
In status "production", last night started at 20:39 and continued in sleeping mode at 21:14 56 error messages were recorded until 5:49:

-----
Deze fout is ontstaan door een aangepaste integratie.

Logger: custom_components.solaredge_modbus_multi
Source: custom_components/solaredge_modbus_multi/hub.py:347 
Integration: SolarEdge Modbus Multi Device (documentation, issues) 
First occurred: 21 augustus 2022 20:39:31 (56 occurrences) 
Last logged: 05:49:31

Unexpected error fetching SolarEdge Coordinator data: Modbus Error: [Connection] ModbusTcpClient(...:1502): Connection unexpectedly closed 0.000135 seconds into read of 8 bytes without response from unit before it closed connection
Unexpected error fetching SolarEdge Coordinator data: Modbus Error: [Connection] ModbusTcpClient(..:1502): Connection unexpectedly closed 0.000132 seconds into read of 8 bytes without response from unit before it closed connection
Unexpected error fetching SolarEdge Coordinator data: Modbus Error: [Connection] ModbusTcpClient(..:1502): Connection unexpectedly closed 0.000136 seconds into read of 8 bytes without response from unit before it closed connection
Unexpected error fetching SolarEdge Coordinator data: Modbus Error: [Connection] ModbusTcpClient(...:1502): Connection unexpectedly closed 0.000138 seconds into read of 8 bytes without response from unit before it closed connection
Unexpected error fetching SolarEdge Coordinator data: Modbus Error: [Connection] ModbusTcpClient(1....:1502): Connection unexpectedly closed 0.000164 seconds into read of 8 bytes without response from unit before it closed connection
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 205, in _async_refresh
    self.data = await self._async_update_data()
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 121, in _async_update_data
    return await self._hub.async_refresh_modbus_data()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 291, in async_refresh_modbus_data
    await self._hass.async_add_executor_job(inverter.read_modbus_data)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 447, in read_modbus_data
    inverter_data = self.hub.read_holding_registers(
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 347, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 109, in execute
    return self.transaction.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 174, in execute
    response, last_exception = self._transact(
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 294, in _transact
    result = self._recv(response_length, full)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/transaction.py", line 326, in _recv
    read_min = self.client.framer.recvPacket(min_size)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/framer/__init__.py", line 49, in recvPacket
    return self.client.recv(size)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 89, in recv
    return self._recv(size)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 297, in _recv
    return self._handle_abrupt_socket_close(
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 342, in _handle_abrupt_socket_close
    raise ConnectionException(msg)
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] ModbusTcpClient(.....:1502): Connection unexpectedly closed 0.000101 seconds into read of 8 bytes without response from unit before it closed connection
------

At 5.49 sleeping mode changed into a non recoverable "not available" generating per 5 min. error messages:

------
2022-08-22 05:54:34.007 ERROR (SyncWorker_4) [pymodbus.client.sync] Connection to (.., 1502) failed: timed out
2022-08-22 05:59:37.009 ERROR (SyncWorker_2) [pymodbus.client.sync] Connection to (.., 1502) failed: timed out
2022-08-22 06:04:40.007 ERROR (SyncWorker_2) [pymodbus.client.sync] Connection to (.... 1502) failed: timed out
.........
------------

At 8:30 I made a total hardware restart and in HA, Solaredge status is showing "production"......

WillCodeForCats, 2022-08-22:
Added connection exception in [Release v2.1.3-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.1.3-pre.2)

bakboord4, 2022-08-22:
@WillCodeForCats 
Implemented -pre.2

bakboord4, 2022-08-23:
@WillCodeForCats.

Update freq. set to 120 sec.
Results -pre.2. 

Happy with the results of work; it works; thanks for far. However,
With status 2 (sleeping) for the second night a non recoverable error meaning in the morning status not changed from 2 to 4. A restart of HA was necessary.

In sleeping mode, is an automatic recovery possible after experiencing an error just like in production mode? 
It will prevent a restart every morning and I would be grateful!

error message:
----------------
Logger: pymodbus.client.sync
Source: /usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py:218 
First occurred: 01:04:49 (209 occurrences) 
Last logged: 08:11:13

Connection to (192.168.2.119, 1502) failed: timed out
---------------------

WillCodeForCats, 2022-08-23:
Instead of restarting HA can you try reloading only the integration?

[image]

bakboord4, 2022-08-23:
@WillCodeForCats.
1. Reload. Will do the next time. Thanks!
2. -pre.3 successful updated thru standard update procedure. Tomorrow night I will report.

WillCodeForCats, 2022-08-24:
[Release v2.1.3-pre.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.1.3-pre.4)

Discovered a potential shutdown or reload bug during my testing.

bakboord4, 2022-08-24:
@WillCodeForCats.
Thanks for informing.
With release V2.13-pre3.
This afternoon a non-recoverable error. Status production changed into 'unknown" but no return into 'production'. 
Reload and restart had no effect. A Host restart did the trick.

error message:
------
2022-08-24 13:03:23.010 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Error fetching SolarEdge Coordinator data: Update failed: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)

2022-08-24 13:05:26.017 ERROR (SyncWorker_3) [pymodbus.client.sync] Connection to (....., 1502) failed: timed out
2022-08-24 13:07:29.005 ERROR (SyncWorker_3) [pymodbus.client.sync] Connection to (......, 1502) failed: timed out
etc. etc. etc. ....
------
May this error message be helpful with discovery of a potential shutdown.

Release 2.2.0-pre.1 installed and continued monitoring.

bakboord4, 2022-08-25:
@WillCodeForCats
Yes I am using WiFI. I contemplating to implement a cable backbone. But it is not a spare afternoon job. Some planning and cable pulling have to be done.
Appreciate your efforts for which I am very thankful. Hopefully some help is offered.

bakboord4, 2022-09-12:
@bakboord4 
Some extensive testing done and conclusion is that this issue, for me, can be closed.
Changed in My Fritz modem/router from Autochannel to fixed. Remains a more than 2 min. WiFi not available (Mesh reorganization?) per night which causes about 30 min not available status of Solaredge but Release 2.2.1 recovers that after 30 min. or so. A reboot automation will be fired for the situation that it takes more than 30 min. Perhaps not the best solution. This integration needs sometimes some help.
A cable connection for Solaredge in my home is not easy to be done.

Seth, many thanks for all the effort you put in to create solutions.

WillCodeForCats, 2022-09-29:
Thanks for your help testing.

Once the pymodbus library has a 3.0.0 release I will make a pre-release for that.

bakboord4, 2022-09-30:
Perfect.

WillCodeForCats, 2022-10-12:
pymodbus has released v3.0.0

I have updated the pymodbus-dev branch to support the latest release version. You can download the branch to try it manually: https://github.com/WillCodeForCats/solaredge-modbus-multi/archive/refs/heads/pymodbus-dev.zip

bakboord4, 2022-10-12:
@WillCodeForCats. Thanks, will implement it and report back

bakboord4, 2022-10-18:
@WillCodeForCats. Tested. pymodbus functioned uninterrupted or recovered automatic from WiFi issues.

WillCodeForCats, 2022-10-23:
That's great to hear it's recovering properly with the latest pymodbus. I will plan to merge PR #132 and make a pre-release with the dependency change to `pymodbus==3.0.0`.

closed 2022-10-24 by WillCodeForCats