WillCodeForCats/solaredge-modbus-multi #82: Status codes wrong in v2.1.2?

pejotigrek, 2022-08-21:
**Describe the bug**
I've noticed that after updating to v2.1.2 some status codes are changed - normally at night my inverter went to "sleep" mode with status code "2" ["Sleeping (auto-shutdown) – Night mode"]. now night status is "5" ["Production (curtailed)"].

when operating at day it is ok - status "4" ["Inverter is ON and producing power"]

**Expected behavior**
I suppose that night mode should by status "4" as "sleeping", because inverter is indeed off, or am I missing something?

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2022.8.6
 - solaredge-modbus-multi Version: 2.1.2

**Additional context**
Add any other context about the problem here.

WillCodeForCats, 2022-08-21:
There have not been any changes to status codes since v1.0

pejotigrek, 2022-08-22:
> There have not been any changes to status codes since v1.0

got it. so I guess it's something inverter-related then?
thanks,!

closed 2022-08-22 by pejotigrek