WillCodeForCats/solaredge-modbus-multi #85: Energy Meter with Modbus Connection not recognized

hoeselr, 2022-08-24:
Setup

SE82.8K Inverter on Modbus address 2
Solaredge Energy Meter connected to Inverter on Modbus address 1 but not recognized by [solaredge-modbus-multi] as sensor.

[image]

WillCodeForCats, 2022-08-24:
Attach the list of attributes for entities `sensor.solaredge_i1_device` and `sensor.solaredge_i2_device` and the list of devices for the integration.

WillCodeForCats, 2022-08-24:
Thanks, I will work on this.

hoeselr, 2022-08-24:
@WillCodeForCats 
thank you very much!

WillCodeForCats, 2022-08-27:
Please try Pre-[Release v2.2.0-mmppt.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.0-mmppt.1)

hoeselr, 2022-08-27:
@WillCodeForCats 

now it works like a charm - SE-MTR-3Y-400V-A successfully discovered on Modbus address 1  - thank you!

[image]

WillCodeForCats, 2022-08-28:
[Pre-Release v2.2.0-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.0-pre.2)

This pre-release also adds a new mmppt events sensor.

hoeselr, 2022-08-28:
@WillCodeForCats 

after the update the integration is not able to communicate with inverter, we also tried fresh install : following Message in logfile

Logger: pymodbus.client.sync
Source: /usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py:218 
First occurred: 18:26:00 (1 occurrences) 
Last logged: 18:26:00

Connection to (10.31.1.240, 1502) failed: [Errno 111] Connection refused

WillCodeForCats, 2022-08-28:
Unless there are any exception errors, "connection refused" is a response from the inverter.

WillCodeForCats, 2022-08-28:
SolarEdge inverters can only handle one connection at a time, so if it thinks something else is connected it will refuse additional connections. Usually the inverter times out after 2 minutes.

hoeselr, 2022-08-28:
@WillCodeForCats 

we waited 5-10 minutes with no success - then reverted to v2.2.0-pre.1 and now the integration connects to Inverter again - but no meters are shown in this Version....

[image]

WillCodeForCats, 2022-08-28:
Not sure on that one. I'm running pre.2 on my live and test systems and it's connecting fine.

pre.1 does not support the Multi MPPT inverters (which is what the SE82.8K is)

I can try removing the new MMPPT Events sensor.

hoeselr, 2022-08-28:
@WillCodeForCats ok - thank you then i will try to install the new version again

WillCodeForCats, 2022-08-28:
[Release v2.2.0-pre.3](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.0-pre.3) reverts the commits to add the new sensor and device attributes for Multi MPPT inverters.

hoeselr, 2022-08-28:
@WillCodeForCats - now connection is up again an Meter is recognized - 
one question about the Meter : now the Meter reports a negative value to Grid (no PV Production)
shouldn't this value be in positive right now ?

[image]

thank you very much

WillCodeForCats, 2022-08-28:
On an Export+Import meter negative is import from grid, positive is export to grid.

hoeselr, 2022-08-28:
@WillCodeForCats ok - thank‘s for the Info!

hoeselr, 2022-08-28:
@WillCodeForCats if there is something that i can try on new version for MMPPT Events sensor - please let me know.

WillCodeForCats, 2022-08-28:
I'll start with this for the v2.2.0 release and work on adding more sensors for Multi MPPT on a minor release.

hoeselr, 2022-08-28:
@WillCodeForCats  Ok great!

closed 2022-08-28 by WillCodeForCats