WillCodeForCats/solaredge-modbus-multi #88: Using wrong version information, causing HACS to stop working

Remko76, 2022-08-28:
Since a couple of days, HACS does not start anymore. I opened a issue there and got this feedback:
"WillCodeForCats/solaredge-modbus-multi is using an invalid version format."

The issue for HACS can be found here: https://github.com/hacs/integration/issues/2837

skl283, 2022-08-28:
Good catch, i also wanted to create this issue - hier is my log

`Logger: homeassistant.components.update
Source: components/update/__init__.py:355
Integration: Aktualisieren ([documentation](https://www.home-assistant.io/integrations/update), [issues](https://github.com/home-assistant/home-assistant/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+update%22))
First occurred: 09:38:55 (2 occurrences)
Last logged: 09:38:55

Error adding entities for domain update with platform hacs
Error while setting up hacs platform for update
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 428, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 673, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 776, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 532, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 570, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 538, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/update/__init__.py", line 355, in state
    newer = AwesomeVersion(latest_version) > installed_version
  File "/usr/local/lib/python3.10/site-packages/awesomeversion/awesomeversion.py", line 170, in __gt__
    return self._compare_versions(self, compareto)
  File "/usr/local/lib/python3.10/site-packages/awesomeversion/awesomeversion.py", line 198, in _compare_versions
    result = handler(AwesomeVersion(version_a), AwesomeVersion(version_b))
  File "/usr/local/lib/python3.10/site-packages/awesomeversion/comparehandlers/modifier.py", line 28, in compare_handler_semver_modifier
    SEMVER_MODIFIER_MAP[version_a.modifier_type]
KeyError: 'mmppt'`

WillCodeForCats, 2022-08-28:
I've deleted the release.

WillCodeForCats, 2022-08-28:
I've made a [Pre-Release v2.2.0-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.0-pre.2) to replace that release.

I'm not sure where I went wrong with the semantic versioning format (https://semver.org) but it's been deleted anyway.

closed 2022-09-17 by WillCodeForCats