WillCodeForCats/solaredge-modbus-multi #91: Config entry 'SolarEdge' for solaredge_modbus_multi integration not ready yet: 'NoneType' object is not subscriptable

dhover, 2022-09-05:
**Describe the bug**
After update from 2.1.3 to 2.2.0 performed according to the documentation I got an error saying:

```
2022-09-05 20:22:56.936 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Unexpected error fetching SolarEdge Coordinator data: 'NoneType' object is not subscriptable
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 205, in _async_refresh
    self.data = await self._async_update_data()
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 121, in _async_update_data
    return await self._hub.async_refresh_modbus_data()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 283, in async_refresh_modbus_data
    await self._async_init_solaredge()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 118, in _async_init_solaredge
    await self._hass.async_add_executor_job(new_inverter.init_device)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 486, in init_device
    self.decoded_mmppt["mmppt_DID"] == SUNSPEC_NOT_IMPL_UINT16
TypeError: 'NoneType' object is not subscriptable
2022-09-05 20:22:56.938 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SolarEdge' for solaredge_modbus_multi integration not ready yet: 'NoneType' object is not subscriptable; Retrying in background
2022-09-05 20:23:01.942 ERROR (SyncWorker_6) [pymodbus.client.sync] Connection to (192.168.178.16, 1502) failed: [Errno 111] Connection refused
```

**Expected behavior**
Normal initialization of SolarEdge device

**Debug Logs**
```
2022-09-05 20:55:02.112 DEBUG (SyncWorker_2) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.178.21', 53269)
2022-09-05 20:55:02.117 DEBUG (SyncWorker_1) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.178.21', 40793)
2022-09-05 20:55:02.128 DEBUG (SyncWorker_0) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 20:55:02.128 DEBUG (SyncWorker_12) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 20:55:02.146 DEBUG (SyncWorker_12) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_ID 0x53756e53
2022-09-05 20:55:02.146 DEBUG (SyncWorker_12) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_DID 0x1
2022-09-05 20:55:02.146 DEBUG (SyncWorker_12) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_Length 0x41
2022-09-05 20:55:02.146 DEBUG (SyncWorker_12) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 20:55:02.175 DEBUG (SyncWorker_12) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Manufacturer SolarEdge
2022-09-05 20:55:02.175 DEBUG (SyncWorker_12) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Model SE5000H-RW000BNN4
2022-09-05 20:55:02.175 DEBUG (SyncWorker_12) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Option
2022-09-05 20:55:02.175 DEBUG (SyncWorker_12) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Version 0004.0012.0028
2022-09-05 20:55:02.175 DEBUG (SyncWorker_12) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SerialNumber 74020113
2022-09-05 20:55:02.175 DEBUG (SyncWorker_12) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Device_address 0x1
2022-09-05 20:55:02.176 DEBUG (SyncWorker_12) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 20:55:02.321 DEBUG (SyncWorker_12) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: ReadHoldingRegistersResponse (65)
2022-09-05 20:55:02.349 DEBUG (SyncWorker_0) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_ID 0x53756e53
2022-09-05 20:55:02.349 DEBUG (SyncWorker_0) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_DID 0x1
2022-09-05 20:55:02.349 DEBUG (SyncWorker_0) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_Length 0x41
2022-09-05 20:55:02.349 DEBUG (SyncWorker_0) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 20:55:02.360 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Unexpected error fetching SolarEdge Coordinator data: 'NoneType' object is not subscriptable
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 205, in _async_refresh
self.data = await self._async_update_data()
File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 121, in _async_update_data
return await self._hub.async_refresh_modbus_data()
File "/config/custom_components/solaredge_modbus_multi/hub.py", line 283, in async_refresh_modbus_data
await self._async_init_solaredge()
File "/config/custom_components/solaredge_modbus_multi/hub.py", line 118, in _async_init_solaredge
await self._hass.async_add_executor_job(new_inverter.init_device)
File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/solaredge_modbus_multi/hub.py", line 486, in init_device
self.decoded_mmppt["mmppt_DID"] == SUNSPEC_NOT_IMPL_UINT16
TypeError: 'NoneType' object is not subscriptable
2022-09-05 20:55:02.362 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.254 seconds (success: False)
2022-09-05 20:55:02.362 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SolarEdge' for solaredge_modbus_multi integration not ready yet: 'NoneType' object is not subscriptable; Retrying in background
```

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2022.8.7
 - solaredge-modbus-multi Version: 2.2.0

dhover, 2022-09-05:
I have two inverters. The other one a Fronius Galvo is working fine.
I've been using this integration since february. I've installed all the updates without any problems. Till now!
What went wrong?

WillCodeForCats, 2022-09-05:
PR #92 will fix this.

WillCodeForCats, 2022-09-05:
Try pre-release Release v2.2.1-pre.1

dhover, 2022-09-05:
Tried 2.2.1-pre.1

```
2022-09-05 22:38:05.847 DEBUG (SyncWorker_14) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.178.21', 55445)
2022-09-05 22:38:05.848 DEBUG (SyncWorker_9) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:38:05.877 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_ID 0x53756e53
2022-09-05 22:38:05.877 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_DID 0x1
2022-09-05 22:38:05.877 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_Length 0x41
2022-09-05 22:38:05.877 DEBUG (SyncWorker_9) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:38:05.898 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Manufacturer SolarEdge
2022-09-05 22:38:05.898 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Model SE5000H-RW000BNN4
2022-09-05 22:38:05.899 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Option
2022-09-05 22:38:05.899 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Version 0004.0012.0028
2022-09-05 22:38:05.899 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SerialNumber 74020113
2022-09-05 22:38:05.899 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Device_address 0x1
2022-09-05 22:38:05.899 DEBUG (SyncWorker_9) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:38:05.926 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: ReadHoldingRegistersResponse (65)
2022-09-05 22:38:05.926 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Unexpected error fetching SolarEdge Coordinator data: 'NoneType' object is not subscriptable
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 205, in _async_refresh
self.data = await self._async_update_data()
File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 121, in _async_update_data
return await self._hub.async_refresh_modbus_data()
File "/config/custom_components/solaredge_modbus_multi/hub.py", line 283, in async_refresh_modbus_data
await self._async_init_solaredge()
File "/config/custom_components/solaredge_modbus_multi/hub.py", line 118, in _async_init_solaredge
await self._hass.async_add_executor_job(new_inverter.init_device)
File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/solaredge_modbus_multi/hub.py", line 486, in init_device
self.decoded_mmppt["mmppt_DID"] == SUNSPEC_NOT_IMPL_UINT16
TypeError: 'NoneType' object is not subscriptable
2022-09-05 22:38:05.928 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.083 seconds (success: False)
2022-09-05 22:38:05.928 WARNING (MainThread) [homeassistant.config_entries] Config entry 'SolarEdge' for solaredge_modbus_multi integration not ready yet: 'NoneType' object is not subscriptable; Retrying in background
2022-09-05 22:38:10.930 ERROR (SyncWorker_0) [pymodbus.client.sync] Connection to (192.168.178.16, 1502) failed: [Errno 111] Connection refused
2022-09-05 22:38:10.932 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.003 seconds (success: False)
2022-09-05 22:38:20.933 DEBUG (SyncWorker_9) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.178.21', 41899)
2022-09-05 22:38:20.934 DEBUG (SyncWorker_2) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:38:20.961 DEBUG (SyncWorker_2) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_ID 0x53756e53
2022-09-05 22:38:20.961 DEBUG (SyncWorker_2) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_DID 0x1
2022-09-05 22:38:20.961 DEBUG (SyncWorker_2) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_Length 0x41
2022-09-05 22:38:20.961 DEBUG (SyncWorker_2) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:38:20.983 DEBUG (SyncWorker_2) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Manufacturer SolarEdge
2022-09-05 22:38:20.983 DEBUG (SyncWorker_2) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Model SE5000H-RW000BNN4
2022-09-05 22:38:20.983 DEBUG (SyncWorker_2) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Option
2022-09-05 22:38:20.983 DEBUG (SyncWorker_2) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Version 0004.0012.0028
2022-09-05 22:38:20.983 DEBUG (SyncWorker_2) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SerialNumber 74020113
2022-09-05 22:38:20.983 DEBUG (SyncWorker_2) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Device_address 0x1
2022-09-05 22:38:20.983 DEBUG (SyncWorker_2) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:38:21.002 DEBUG (SyncWorker_2) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: ReadHoldingRegistersResponse (65)
2022-09-05 22:38:21.002 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Unexpected error fetching SolarEdge Coordinator data: 'NoneType' object is not subscriptable
Traceback (most recent call last):
[... cut ...]
await self._hass.async_add_executor_job(new_inverter.init_device)
File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
result = self.fn(*self.args, **self.kwargs)
File "/config/custom_components/solaredge_modbus_multi/hub.py", line 486, in init_device
self.decoded_mmppt["mmppt_DID"] == SUNSPEC_NOT_IMPL_UINT16
TypeError: 'NoneType' object is not subscriptable
2022-09-05 22:38:21.003 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.071 seconds (success: False)
```

dhover, 2022-09-05:
One remark: is it possible to add the name of the inverter or something unique to the debug log. Now I can't see which line is from which inverter.
In this last test I've only installed the SolarEdge inverter. Still the same error.

dhover, 2022-09-05:
Sorry something went wrong. It is working now!!!!

dhover, 2022-09-05:
```
2022-09-05 22:51:44.170 DEBUG (SyncWorker_12) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.178.21', 53267)
2022-09-05 22:51:44.175 DEBUG (SyncWorker_1) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:51:44.206 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_ID 0x53756e53
2022-09-05 22:51:44.206 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_DID 0x1
2022-09-05 22:51:44.206 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_Length 0x41
2022-09-05 22:51:44.207 DEBUG (SyncWorker_1) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:51:44.241 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Manufacturer SolarEdge
2022-09-05 22:51:44.241 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Model SE5000H-RW000BNN4
2022-09-05 22:51:44.241 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Option
2022-09-05 22:51:44.241 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Version 0004.0012.0028
2022-09-05 22:51:44.241 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SerialNumber 74020113
2022-09-05 22:51:44.241 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Device_address 0x1
2022-09-05 22:51:44.241 DEBUG (SyncWorker_1) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:51:44.264 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: ReadHoldingRegistersResponse (65)
2022-09-05 22:51:44.264 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1 is NOT Multiple MPPT
2022-09-05 22:51:44.315 DEBUG (SyncWorker_10) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:51:44.330 DEBUG (SyncWorker_10) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_SunSpec_DID 0xffff
2022-09-05 22:51:44.330 DEBUG (SyncWorker_10) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_SunSpec_Length 0x0
2022-09-05 22:51:44.387 DEBUG (SyncWorker_0) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:51:44.400 DEBUG (SyncWorker_0) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 2: Exception Response(131, 3, IllegalAddress)
2022-09-05 22:51:44.483 DEBUG (SyncWorker_1) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:51:44.501 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 3: Exception Response(131, 3, IllegalAddress)
2022-09-05 22:51:44.507 DEBUG (SyncWorker_9) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:51:44.522 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_DID 0x65
2022-09-05 22:51:44.522 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_Length 0x32
2022-09-05 22:51:44.522 DEBUG (SyncWorker_9) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-09-05 22:51:44.544 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_DID 0x65
2022-09-05 22:51:44.544 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Current 0x0
2022-09-05 22:51:44.544 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Current_A 0x0
2022-09-05 22:51:44.544 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Current_B 0xffff
2022-09-05 22:51:44.544 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Current_C 0xffff
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Current_SF -0x2
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_AB 0x90e
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_BC 0xffff
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_CA 0xffff
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_AN 0xffff
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_BN 0xffff
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_CN 0xffff
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_SF -0x1
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Power 0x0
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Power_SF 0x0
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Frequency 0xc36e
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Frequency_SF -0x3
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_VA 0x0
2022-09-05 22:51:44.545 DEBUG (SyncWorker_9) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_VA_SF 0x0
[... cut ...]
2022-09-05 22:52:24.524 DEBUG (SyncWorker_10) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_Temp_SF -0x8000
2022-09-05 22:52:24.524 DEBUG (SyncWorker_10) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_Status 0x7
2022-09-05 22:52:24.524 DEBUG (SyncWorker_10) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_Status_Vendor 0x7
2022-09-05 22:52:24.525 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 2.219 seconds (success: True)
````

dhover, 2022-09-05:
> One remark: is it possible to add the name of the inverter or something unique to the debug log. Now I can't see which line is from which inverter.

Is the whole setup of an inverter done completely before the next inverter is initialized?
If so than there is no need for extra info in the logfiles.

closed 2022-09-06 by WillCodeForCats

WillCodeForCats, 2022-09-06:
The logs work for me for what I need to see to work on it.