WillCodeForCats/solaredge-modbus-multi #101: [cosmetic bug] warning about polling

spali, 2022-09-20:
Before reporting a bug, make sure your configuration is correct and your LAN is working.
https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

**Describe the bug**
getting following warning every restart:
```
2022-09-20 09:32:18.237 WARNING (MainThread) [custom_components.solaredge_modbus_multi] Polling frequency < 10, requiring keep modbus open.
```

**Expected behavior**
would expect to suppress this warning when keep modbus open is checked.

**Screenshots**
[image]

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2022.9.5
 - solaredge-modbus-multi Version: 2.2.2

**Additional context**
same applied for version 2.2.1.

WillCodeForCats, 2022-09-20:
The warning is generated in the data update class that does not have access to the options data (only the interval is passed to it), so there probably isn't a way to suppress it.

spali, 2022-09-20:
just checked the source, looks like it's in this integration:
https://github.com/WillCodeForCats/solaredge-modbus-multi/blob/e6daed4e148836b30ba01e239996e7ef36868967/custom_components/solaredge_modbus_multi/__init__.py#L114-L116

So it looks like it force sets the connection to stay open anyway in this case. But maybe the `if` could be extended to also check the setting:
```python
        if scan_interval < 10 and not self._hub.keep_modbus_open:
            _LOGGER.warning("Polling frequency < 10, requiring keep modbus open.")
            self._hub.keep_modbus_open = True
```
assuming `self._hub.keep_modbus_open` has already the value set from the user config at this time.

WillCodeForCats, 2022-09-20:
Yeah that should probably do it. PR #102

closed 2022-09-20 by WillCodeForCats