WillCodeForCats/solaredge-modbus-multi #113: Use battery data in Energy dashboard

rploeg, 2022-09-30:
Hi,

I receive import and export data from my LG battery via this integration, but I can't select that yet in the Energy dashboard:

[image]

Would it be possible to use those energy import and export sensor from a battery in the energy dashboard?

WillCodeForCats, 2022-09-30:
It works on my test system:

[image]

rploeg, 2022-09-30:
Strange, I don't see it, only the ones for the solaredge API integration:

[image]

And see the two fields also in the statistics of HA:

[image]

rploeg, 2022-09-30:
Reboot of HA did the trick

closed 2022-09-30 by rploeg