WillCodeForCats/solaredge-modbus-multi #118: Battery data - unknown

rploeg, 2022-10-03:
Hi,

I am using your integration now for a few days and it is working great. The only thing what sometimes goes wrong is the energy import and export of the Battery:
[image]
When the battery is loading it gives both import and export as unknown. When there is a bigger load it looks like that it is working. For example the battery is loading at the moment, but I see an unknown. History shows that the sensor is working:
[image]

The sensor looks updated 8 minutes ago.

WillCodeForCats, 2022-10-03:
Unknown means the data returned failed a validation check. The two conditions are if it returned max value (18446744073709552000) or if it goes backwards (it should always go up).

I can add an option to ignore data validation if you think that's what you need, but you probably won't like the results.

WillCodeForCats, 2022-10-03:
You can also turn on debug logging to see what the actual value being returned is.

rploeg, 2022-10-03:
how can I enable debug logging?

WillCodeForCats, 2022-10-03:
Use the bug report template.

rploeg, 2022-10-04:
Sorry, didn't saw that. I have enabled the logger, but can't find it in the location that you mention in the bug report. 
[image]

rploeg, 2022-10-04:
thanks. Have enabled the debugging. I think I have to wait till tonight before I see the unknown. Because due to the reboot the battery in and out export are working again.

rploeg, 2022-10-04:
@WillCodeForCats  here the output, have now an unknown:

2022-10-04 21:03:40.501 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Temp_Average 20.700000762939453
2022-10-04 21:03:40.501 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Temp_Max 0.0
2022-10-04 21:03:40.501 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_DC_Voltage 410.0
2022-10-04 21:03:40.501 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_DC_Current -1.399999976158142
2022-10-04 21:03:40.501 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_DC_Power -574.0
2022-10-04 21:03:40.501 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Export_Energy_WH 0x2dc
2022-10-04 21:03:40.502 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Import_Energy_WH 0x0
2022-10-04 21:03:40.502 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Energy_Max 9800.0
2022-10-04 21:03:40.502 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Energy_Available 8518.0
2022-10-04 21:03:40.502 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_SOH 91.5
2022-10-04 21:03:40.502 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_SOE 21.98872947692871
2022-10-04 21:03:40.502 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Status 0x4
2022-10-04 21:03:40.502 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Status_Vendor 0x3
2022-10-04 21:03:40.502 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log1 0x0
2022-10-04 21:03:40.502 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log2 0x0
2022-10-04 21:03:40.502 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log3 0x0
2022-10-04 21:03:40.502 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log4 0x0
2022-10-04 21:03:40.503 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log5 0x0
2022-10-04 21:03:40.503 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log6 0x0
2022-10-04 21:03:40.503 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log7 0x0
2022-10-04 21:03:40.503 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log8 0x0
2022-10-04 21:03:40.503 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log_Vendor1 0x0
2022-10-04 21:03:40.503 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log_Vendor2 0x0
2022-10-04 21:03:40.503 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log_Vendor3 0x0
2022-10-04 21:03:40.503 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log_Vendor4 0x0
2022-10-04 21:03:40.503 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log_Vendor5 0x0
2022-10-04 21:03:40.503 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log_Vendor6 0x0
2022-10-04 21:03:40.503 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log_Vendor7 0x0
2022-10-04 21:03:40.504 DEBUG (SyncWorker_5) [custom_components.solaredge_modbus_multi.hub] Inverter 2 batt 1: B_Event_Log_Vendor8 0x0

WillCodeForCats, 2022-10-04:
The debug value `B_Import_Energy_WH 0x0` means that the inverter is reporting 0 for that data point, which will be displayed as "unknown" in home assistant because accumulator data points (`B_Import_Energy_WH` is a lifetime watt-hour meter) need to be non-zero and always increasing. They should not be 0 or go backwards. Then cay be 0 when the equipment is new, but any usage would make them start to count up.

If you look for other values of `B_Import_Energy_WH` in the debug logs and it changes from a higher value backwards to 0x0, that is probably an issue on the inverter or battery side not reporting that data point correctly.

rploeg, 2022-10-05:
Thank you. This morning (also when loading the battery) the import/ export still was unknown. I had manually reloaded the integration and directly the sensors became available.

rploeg, 2022-10-06:
I also don't understand. But when the battery is charging I would expect that unknown will go away with the numbers. But that doesn't happen (also not today). So when I do reload of the integration this morning, the numbers came back. Would it be possible that the sensors are hanging in HA because it's so long a unknown?

WillCodeForCats, 2022-10-06:
The debug shows 0x0 for that sensor, and from what I know I don't see how reloading could change that.

I don't have a battery so I can't actually observe the problem, and I've never seen it with the other watt-hour type sensors.

WillCodeForCats, 2022-10-06:
What I can do is change the code to allow those two values to reset to zero while the integration is running if that's what's happening.

I'll work on that and make a pre-release that you can try.

WillCodeForCats, 2022-10-24:
Allowing battery energy to reset to zero has been added to [Release v2.2.6-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.6-pre.2)

A note about this issue has been added to the wiki: https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues#battery-energy

closed 2022-10-24 by WillCodeForCats

djfanatix, 2022-11-18:
Sorry to open a closed issue, but I still have the issue that the battery energy goes to unknown.
Only a reload of the integration makes it working again.
(version v2.2.7-pre.10)

WillCodeForCats, 2022-11-18:
Please read the known issues and enable the option that lets the battery reset to zero.

djfanatix, 2022-11-18:
Yes I have enabled this option, but the data still goes to unknown at times

WillCodeForCats, 2022-11-18:
Commit 0b381a953831ecdc569eaf4e3a3a333efdfdddb4 should fix that, but it's not in a pre-release yet.

WillCodeForCats, 2022-11-19:
> Yes I have enabled this option, but the data still goes to unknown at times

@djfanatix Please try [Release v2.2.7-pre.11](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.7-pre.11)

djfanatix, 2022-11-20:
Looks good for the moment. Many thanks
I made an utility meter for the battery meters and at the moment they are exactly the same as the values that I retrieve from the SolarEdge API. 

BTW according Solaredge api documentation:
`As LG battery does not provide lifetime charge/discharge data, the monitoring system aggregates the delta charge/discharge values. In cases where telemetries containing delta energy values are lost or not sent, the calculated lifetime energy values will be incomplete. Values provided are not revenue grade.`

WillCodeForCats, 2022-11-20:
Thanks for the info on the LG battery, I'll add it to the wiki.