WillCodeForCats/solaredge-modbus-multi #123: Battery DC Voltage and Current values are massive

Djelibeybi, 2022-10-07:
**Describe the bug**

The values being reporting for DC voltage and current are way out of proportion and hopefully not valid.

Entity ID: `sensor.solaredge_b1_dc_current`
State: `-3.40282346638529e+38`

Entity ID: `sensor.solaredge_b1_dc_voltage`
State: `-3.40282346638529e+38`

**Expected behavior**

Sane/reasonable values. :) 

**Screenshots**
[image]
[image]

**Logs**
Copy any log warnings or errors in Home Assistant from "custom_components.solaredge_modbus" and "pymodbus.client.sync".

No warnings or errors. 

**Debug Logs**

```log
2022-10-07 17:15:19.097 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_DID 0x65
2022-10-07 17:15:19.097 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_Length 0x32
2022-10-07 17:15:19.118 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_DID 0x65
2022-10-07 17:15:19.118 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Current 0x1c4
2022-10-07 17:15:19.118 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Current_A 0x1c4
2022-10-07 17:15:19.119 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Current_B 0xffff
2022-10-07 17:15:19.119 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Current_C 0xffff
2022-10-07 17:15:19.119 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Current_SF -0x2
2022-10-07 17:15:19.119 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_AB 0x97c
2022-10-07 17:15:19.119 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_BC 0xffff
2022-10-07 17:15:19.119 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_CA 0xffff
2022-10-07 17:15:19.120 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_AN 0x97c
2022-10-07 17:15:19.120 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_BN 0xffff
2022-10-07 17:15:19.120 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_CN 0xffff
2022-10-07 17:15:19.120 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Voltage_SF -0x1
2022-10-07 17:15:19.120 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Power 0x2411
2022-10-07 17:15:19.121 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Power_SF -0x1
2022-10-07 17:15:19.121 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Frequency 0xc315
2022-10-07 17:15:19.121 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Frequency_SF -0x3
2022-10-07 17:15:19.121 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_VA 0x2ae6
2022-10-07 17:15:19.121 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_VA_SF -0x1
2022-10-07 17:15:19.121 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_var -0x173a
2022-10-07 17:15:19.122 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_var_SF -0x1
2022-10-07 17:15:19.122 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_PF -0x20d7
2022-10-07 17:15:19.122 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_PF_SF -0x2
2022-10-07 17:15:19.122 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Energy_WH 0x7e1c
2022-10-07 17:15:19.122 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: AC_Energy_WH_SF 0x0
2022-10-07 17:15:19.122 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_DC_Current 0x62d2
2022-10-07 17:15:19.123 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_DC_Current_SF -0x4
2022-10-07 17:15:19.123 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_DC_Voltage 0xe79
2022-10-07 17:15:19.123 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_DC_Voltage_SF -0x1
2022-10-07 17:15:19.123 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_DC_Power 0x249d
2022-10-07 17:15:19.124 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_DC_Power_SF -0x1
2022-10-07 17:15:19.124 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_Temp_Cab -0x8000
2022-10-07 17:15:19.124 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_Temp_Sink 0xee7
2022-10-07 17:15:19.125 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_Temp_Trns -0x8000
2022-10-07 17:15:19.125 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_Temp_Other -0x8000
2022-10-07 17:15:19.125 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_Temp_SF -0x2
2022-10-07 17:15:19.125 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_Status 0x4
2022-10-07 17:15:19.125 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_Status_Vendor 0x0
2022-10-07 17:15:19.146 DEBUG (SyncWorker_0) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_SunSpec_DID 0xcb
2022-10-07 17:15:19.146 DEBUG (SyncWorker_0) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_SunSpec_Length 0x69
2022-10-07 17:15:19.171 DEBUG (SyncWorker_0) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_SunSpec_DID 0xcb
2022-10-07 17:15:19.171 DEBUG (SyncWorker_0) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: AC_Current 0x2f
[... 85 lines cut ...]
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log2 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log3 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log4 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log5 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log6 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log7 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log8 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor1 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor2 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor3 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor4 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor5 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor6 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor7 0x0
2022-10-07 17:15:19.209 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor8 0x0
2022-10-07 17:15:19.211 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.210 seconds (success: True)
```

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2022.10.1
 - solaredge-modbus-multi Version: 2.2.3

**Additional context**

The inverter and battery were only installed yesterday.

WillCodeForCats, 2022-10-07:
I need you to run the debug logging again with release [v2.2.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.4)

Those values are invalid and the sensor should already check that then display them as "unknown", but the logs are showing a number that isn't helpful for debugging so I updated logging in release v2.2.4 to show hex for those data points.

WillCodeForCats, 2022-10-07:
PR #125 will limit display of battery and voltage to ranges based on reading a bunch of spec sheets and picking values that look like they will cover all of the different battery options. Any values outside of these ranges will display as "unknown".

WillCodeForCats, 2022-10-07:
Please try Release v2.2.5-pre.1 for the display value fix as well.

Djelibeybi, 2022-10-07:
I grabbed new logs using 2.2.4 before I read the rest of your responses. I included `pymodbus` as well this time:

```log
2022-10-08 03:40:38.925 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=1, start_device_id=1, detect_meters=True, detect_batteries=True, single_device_entity=True, keep_modbus_open=False, 
2022-10-08 03:40:38.929 DEBUG (SyncWorker_12) [pymodbus.client.sync] Connection to Modbus server established. Socket ('192.168.254.200', 53389)
2022-10-08 03:40:38.938 DEBUG (SyncWorker_10) [pymodbus.transaction] Current transaction state - IDLE
2022-10-08 03:40:38.939 DEBUG (SyncWorker_10) [pymodbus.transaction] Running transaction 1
2022-10-08 03:40:38.939 DEBUG (SyncWorker_10) [pymodbus.transaction] SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x40 0x0 0x4
2022-10-08 03:40:38.939 DEBUG (SyncWorker_10) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-10-08 03:40:38.941 DEBUG (SyncWorker_10) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-10-08 03:40:38.967 DEBUG (SyncWorker_10) [pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-10-08 03:40:38.967 DEBUG (SyncWorker_10) [pymodbus.transaction] RECV: 0x0 0x1 0x0 0x0 0x0 0xb 0x1 0x3 0x8 0x53 0x75 0x6e 0x53 0x0 0x1 0x0 0x41
2022-10-08 03:40:38.967 DEBUG (SyncWorker_10) [pymodbus.framer.socket_framer] Processing: 0x0 0x1 0x0 0x0 0x0 0xb 0x1 0x3 0x8 0x53 0x75 0x6e 0x53 0x0 0x1 0x0 0x41
2022-10-08 03:40:38.968 DEBUG (SyncWorker_10) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-10-08 03:40:38.968 DEBUG (SyncWorker_10) [pymodbus.transaction] Adding transaction 1
2022-10-08 03:40:38.969 DEBUG (SyncWorker_10) [pymodbus.transaction] Getting transaction 1
2022-10-08 03:40:38.969 DEBUG (SyncWorker_10) [pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-10-08 03:40:38.969 DEBUG (SyncWorker_10) [pymodbus.payload] [21365, 28243, 1, 65]
2022-10-08 03:40:38.970 DEBUG (SyncWorker_10) [pymodbus.payload] [b'Su', b'nS']
2022-10-08 03:40:38.970 DEBUG (SyncWorker_10) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_ID 0x53756e53
2022-10-08 03:40:38.970 DEBUG (SyncWorker_10) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_DID 0x1
2022-10-08 03:40:38.970 DEBUG (SyncWorker_10) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_Length 0x41
2022-10-08 03:40:38.970 DEBUG (SyncWorker_10) [pymodbus.transaction] Current transaction state - TRANSACTION_COMPLETE
2022-10-08 03:40:38.971 DEBUG (SyncWorker_10) [pymodbus.transaction] Running transaction 2
2022-10-08 03:40:38.971 DEBUG (SyncWorker_10) [pymodbus.transaction] SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x3 0x9c 0x44 0x0 0x41
2022-10-08 03:40:38.971 DEBUG (SyncWorker_10) [pymodbus.client.sync] New Transaction state 'SENDING'
2022-10-08 03:40:38.972 DEBUG (SyncWorker_10) [pymodbus.transaction] Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2022-10-08 03:40:38.987 DEBUG (SyncWorker_10) [pymodbus.transaction] Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2022-10-08 03:40:38.988 DEBUG (SyncWorker_10) [pymodbus.transaction] RECV: 0x0 0x2 0x0 0x0 0x0 0x85 0x1 0x3 0x82 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x36 0x30 0x30 0x30 0x48 0x2d 0x41 0x55 0x53 0x4e 0x42 0x42 0x58 0x31 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0x2e 0x30 0x30 0x31 0x36 0x2e 0x30 0x30 0x32 0x33 0x0 0x0 0x37 0x34 0x30 0x37 0x41 0x35 0x39 0x41 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1
2022-10-08 03:40:38.988 DEBUG (SyncWorker_10) [pymodbus.framer.socket_framer] Processing: 0x0 0x2 0x0 0x0 0x0 0x85 0x1 0x3 0x82 0x53 0x6f 0x6c 0x61 0x72 0x45 0x64 0x67 0x65 0x20 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x53 0x45 0x36 0x30 0x30 0x30 0x48 0x2d 0x41 0x55 0x53 0x4e 0x42 0x42 0x58 0x31 0x34 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x30 0x30 0x30 0x34 0x2e 0x30 0x30 0x31 0x36 0x2e 0x30 0x30 0x32 0x33 0x0 0x0 0x37 0x34 0x30 0x37 0x41 0x35 0x39 0x41 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x1
2022-10-08 03:40:38.989 DEBUG (SyncWorker_10) [pymodbus.factory] Factory Response[ReadHoldingRegistersResponse: 3]
2022-10-08 03:40:38.989 DEBUG (SyncWorker_10) [pymodbus.transaction] Adding transaction 2
2022-10-08 03:40:38.990 DEBUG (SyncWorker_10) [pymodbus.transaction] Getting transaction 2
2022-10-08 03:40:38.990 DEBUG (SyncWorker_10) [pymodbus.transaction] Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2022-10-08 03:40:38.990 DEBUG (SyncWorker_10) [pymodbus.payload] [21359, 27745, 29253, 25703, 25888, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 21317, 13872, 12336, 18477, 16725, 21326, 16962, 22577, 13312, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 12336, 12340, 11824, 12337, 13870, 12336, 12851, 0, 14132, 12343, 16693, 14657, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 1]
[... cut ...]
2022-10-08 03:40:39.613 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor6 0x0
2022-10-08 03:40:39.613 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor7 0x0
2022-10-08 03:40:39.613 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1 batt 1: B_Event_Log_Vendor8 0x0
2022-10-08 03:40:39.614 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.688 seconds (success: True)
```

WillCodeForCats, 2022-10-07:
I was expecting to see 0xff7fffff as the hex value and this confirms is, but I'm not sure why the existing checks didn't catch that. In any case the validation change in 2.2.5-pre.1 is more strict with the allowed values. I read a bunch of data sheets and picked some values that follow the specs.

Unfortunately there is no useful value to display since the inverter is effectively returning "offscale low" (the maximum negative IEEE 754 floating point number possible) which means those data points for voltage and current are not available. This could be related to the inverter, battery, and communication combination or a firmware issue with the inverter or battery. But I'm just guessing since I don't have access to any battery systems.

Djelibeybi, 2022-10-07:
Thanks. This confirms my suspicion that the battery isn't operating correctly as the official app shows no charging or discharging happening for the past 24 hours. I have opened a support ticket with my solar installer, so I'll escalate that.

WillCodeForCats, 2022-10-07:
Yeah, as far as I can recall, all of the battery data samples I've seen always had working voltage and current values.

Djelibeybi, 2022-10-08:
The inverter lost comms with the battery completely, so I did a full power cycle of the entire system and it's now back up and running with proper values for those attributes. Thanks for the quick response, though.

closed 2022-10-08 by Djelibeybi

WillCodeForCats, 2022-10-08:
Displaying that data still shouldn't have happened, so updated battery validation is now in [Release v2.2.5-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.5-pre.2)