WillCodeForCats/solaredge-modbus-multi #127: New KeyError for B_RatedEnergy with 2.2.5-pre2

Djelibeybi, 2022-10-09:
**Describe the bug**

After upgrading to 2.2.5-pre2, the following `KeyError` appears in my logs:

```python
2022-10-10 07:36:35.065 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up solaredge_modbus_multi platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 289, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 428, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 691, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 777, in add_to_platform_finish
    self.async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 533, in async_write_ha_state
    self._async_write_ha_state()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 571, in _async_write_ha_state
    state = self._stringify_state(available)
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 539, in _stringify_state
    if (state := self.state) is None:
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 404, in state
    value = self.native_value
  File "/config/custom_components/solaredge_modbus_multi/sensor.py", line 1760, in native_value
    > self._platform.decoded_model["B_RatedEnergy"]
KeyError: 'B_RatedEnergy'
```

This is most likely happening because my battery got fully drained overnight and is currently showing 0% in the app for available energy. 

I can grab logs if necessary, but I suspect it's just a case of checking for the existence of the key in `decoded_model` before creating the sensor.

WillCodeForCats, 2022-10-09:
Thanks, I made a mistake: B_RatedEnergy is not in decoded_model, it's in decoded_common

WillCodeForCats, 2022-10-09:
Fixed in PR #128 and will make a new pre-release.

closed 2022-10-09 by WillCodeForCats