WillCodeForCats/solaredge-modbus-multi #131: Secondary Inverter not visible

mkrasselt1, 2022-10-11:
First of all, thanks for creating this alternative implementation :)
We have an exsiting Solaredge Setup consiting of one SE17K-RW000BNN4 as as Slave and another SE10K-RWS48BNN4 with LG Resu 13 Battery and a Grid Meter.
All showing up on the monitoring website just fine.
Also Modbus TCP over Ethenret is working stable and reliable.
Previously using another binsentsu/home-assistant-solaredge-modbus integration with a refreshrate of 1s
But: Yours and theirs both show only the Data from the Secondary Inverter with meter and Battery.
The First/Bigger one is completly invisible :O

I checked both inverters and the first one is configured as Solaredge Secondary Device on modbus and Visible and acessible for the second one. 

Reported production data is also only showing the second inverter.
Adding the Plugin using Parameter "Amount of Inverters: 2" Leads to the Integration failing with "Inverter with ID:2 not found" -
but i noticed the Meter beeing configured on ID:2 for the Secondary Modbus Connection of the Second Inverter.

**Expected behavior**
I was expection to see both Inverters to be detected or at least see their data combined into one "virtual" inverter

**Logs**
Their are no error messages as the Integrations failed to start using parameter "Amount of Inverters: 2" and working fine with one

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2022.10.3
 - solaredge-modbus-multi Version: v.2.2.4

I will try to set the first Inverter to ID 2 or route another modbus cable to it, but i am not expection much improvement

mkrasselt1, 2022-10-11:
Okay - that was easy - so for everyone as a documentation - the inverters have to be set up first, and then the meter - even for the adresses .
So the battery and cloud connected inverter has to be modbus id 1, the secondary inverter id 2, the grid meter modbus id 3 - even though the are on different bus networks.
Wiring this bug report helped me finding the error, and i am keeping it for documentation, but it can be considerd: solved :)

closed 2022-10-11 by mkrasselt1

WillCodeForCats, 2022-10-11:
I have seen several configs with the first inverter as ID 1, the meter as address 2 (connected to inverter ID 1) and then the second inverter is address 3 because the installer doesn't realize the addresses can overlap since the inverter chain and accessories (like meters) are on different RS485 ports.

I use IDs 1 and 2 for my inverters on RS485-1 and and ID 2 for my export+import meter on RS485-2. The screenshots in the wiki are from my system: https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration