WillCodeForCats/solaredge-modbus-multi #136: Power control functions

WillCodeForCats, 2022-10-13:


bertybuttface, 2022-10-29:
Is this ready for pre-release?

WillCodeForCats, 2022-10-30:
The most helpful thing right now is making sure pymodbus 3.0.0 is ok in Release v2.2.6-pre.3  because I'm not going to do work on control functions for older pymodbis versions.

bertybuttface, 2022-10-30:
> The most helpful thing right now is making sure pymodbus 3.0.0 is ok in Release v2.2.6-pre.3 because I'm not going to do work on control functions for older pymodbus versions.

Been running it since you released it, not seen any issues yet. I am also using pymodbus 3.0.0 as part of solaredge-meter-proxy and it is working fine there too. 

Will keep my eyes out for any issues. Thanks for your hard work on this.

Cereal2nd, 2022-10-30:
Same here have been running it since you released, no problems seen

WillCodeForCats, 2022-11-02:
Accidentally closed.

WillCodeForCats, 2022-11-02:
StorEdge power control functions are ready to get a pre-release, which will come immediately after the 2.2.6 release.

Based on PR #133 by @mpredfearn I've added commits 315e83fd4cfbf6625f5ec17746c75b94622c0ba4, 3ed64a57c5a17db01e04e7db044091e67e779279, 61a4c315c8da8d47932383cecf6a52a29b579b62, and 0f59f161bda547d377c1bd1a2bbbfc4df513f928 to get it to what I feel is pre-release ready.

I messed things up a bit by accidentally merging #133 before another pre-release so I reverted it in #153, but then #154 will revert that revert which will happen as soon as 2.2.6 is released.

bertybuttface, 2022-11-02:
Excellent work, I sent you some $$$ and would encourage others to do the same. Ready to test when you are ready, let me know if there is anything you want to specifically focus on.

WillCodeForCats, 2022-11-03:
[Release v2.2.7-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.7-pre.1) is now available.

Cereal2nd, 2022-11-03:
thanks, i have been playing with this and i see the following:

2022-11-03 21:10:12.418 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.418 seconds (success: True)
2022-11-03 21:10:27.323 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.number] set SE5000H-RW000BNN4_74098B35_storedge_backup_reserved to 0.0
2022-11-03 21:10:27.357 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 30
2022-11-03 21:10:27.361 ERROR (SyncWorker_3) [pymodbus.client.tcp] Connection to (192.168.188.4, 1502) failed: [Errno 111] Connection refused
2022-11-03 21:10:27.366 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Error fetching SolarEdge Coordinator data: Could not open Modbus/TCP connection to 192.168.188.4
2022-11-03 21:10:27.369 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 0.012 seconds (success: False)

so on every change the connection to the  modbus is lost

WillCodeForCats, 2022-11-03:
It does immediately disconnect at the end of the write (that's supposed to happen) and the next step is to request a data update which requests a reconnect, but the inverter rejected the connection. It works on my modbus simulator but it's possible the disconnect-then-reconnect is happening too fast for a real inverter.

I can try removing the disconnect call.

WillCodeForCats, 2022-11-03:
Changed in [Release v2.2.7-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.7-pre.2)

Cereal2nd, 2022-11-04:
@WillCodeForCats pre.2 is working perfectly.
I'm not 100% sure what all the options do, any idea where i can find some documentation for this?
I'm looking for a way to charge my battery overnight for a certain time of certain load

WillCodeForCats, 2022-11-04:
I can make a wiki page that describes the different settings with the docs I have.

I can't give you a direct answer on which ones to set since I don't have batteries on my system, but hopefully someone that does have a battery can say which combination to use and we can add that to the wiki too.

WillCodeForCats, 2022-11-05:
I've created wiki page: https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Power-Control-Options

WillCodeForCats, 2022-11-06:
I'm thinking about changing the sensor names because I don't like how they currently appear on the dashboard in cards where you just see a pile of things called "storedge" with the most useful part of the entity name cut off by default.

Current default:

[image]

Possible new default:

[image]

Or keep StorEdge naming in selectors only:

[image]

Cereal2nd, 2022-11-06:
i like the second screenshot the most, as the storeEdge keyword is not always correct, you can also have a setisi installed with and external battery (my installation)

also the twiki page is a good start,
but whats the difference between the storedge control mode, default mode and remote mode?

WillCodeForCats, 2022-11-06:
The way I understand it is that the "remote" settings temporarily override the default mode for the length of the timeout, then it goes back to operating in the default mode.

Cereal2nd, 2022-11-06:
yeah indeed, i like the temporary naming

bertybuttface, 2022-11-06:
> The way I understand it is that the "remote" settings temporarily override the default mode for the length of the timeout, then it goes back to operating in the default mode.
> 
> Edit: rereading docs again and maybe "remote mode" could be called "temporary mode", "remote charge limit" as "charge limit", "remote discharge limit" as "discharge limit", then "remote command timeout" could be "temporary mode timeout". If the limit values are applied to either "default" or "remote" mode then having remote in the name makes it look like they only activate during remote mode.

I'm not sure that it is temporary mode, where are you reading that?

I've had my battery in remote command mode for weeks. I thought it needed to be in order to accept remote Storedge Modbus commands.

WillCodeForCats, 2022-11-06:
From a UI perspective I find all of the "remote" references confusing. The docs are written from the perspective of the inverter, but if you're working inside the Home Assistant UI then HA is the "remote". The original naming is taken straight from the docs. For me a user should not have to refer to docs to understand the basic meaning of a setting, and if I'm not able to keep track of what each mode does and how the others are affected while developing/testing without having to keep looking up which is which then the presentation is bad.

I would change "Remote Control" to "Manual Control", drop every reference to "remote" in all settings, and differentiate the default command mode and the command mode affected by the timeout setting by calling it a temporary mode since it will expire based on the timeout value then revert operation to the default mode.

Although I realize people may be coming from another integration and what it called things, I don't want to be constrained by design decisions in other integrations that confuse me.

These are the groupings that make sense to me based on how the docs explain how things work:
[image]

WillCodeForCats, 2022-11-06:
Basically if I'm a user then I should be able to look at that for the first time and get a general idea what the settings might do.

bertybuttface, 2022-11-06:
Is it a good idea to deviate from the manual and any other implementation? I'd argue not but it is your code.

bertybuttface, 2022-11-06:
I also don't get why it would be called temporary mode when it can be turned on indefinitely. How can something be both temporary and permanent at the same time?

Remote mode was obvious to me. I have a TV remote, it lets me control the TV remotely. I have an Inverter remote mode, turning it on lets me control the inverter remotely. I don't really get where the confusion is coming from?

WillCodeForCats, 2022-11-06:
> Is it a good idea to deviate from the manual and any other implementation? I'd argue not but it is your code.

I mean, I don't want to copy someone else's work just because that's what everyone is already used to blindly just because they exist. If it was good enough I wouldn't have spent all this time learning how to develop for HA and writing a new integration for my system. [I posted my backstory on the HA forums.](https://community.home-assistant.io/t/solaredge-modbus-configuration-for-single-inverter-and-battery/464084/17)

There's a technical note that someone leaked that wasn't available publicly from SolarEdge, so it was written with a technical intent and not as a manual for end users. The integration should be end user experience first.

> I also don't get why it would be called temporary mode when it can be turned on indefinitely. How can something be both temporary and permanent at the same time?

The docs define that register as "the operating mode during the defined time frame" and "when expired, it reverts to the default mode". I considered the definition of the word "temporary" as "lasting for only a limited period of time" so that's where I approached it from. 

What I don't know though is if the timer starts counting when the timeout register is written, the timeout command mode is written, or both. In any case it clearly says limited time up to a max of 24 hours (86400 seconds).

> Remote mode was obvious to me. I have a TV remote, it lets me control the TV remotely. I have an Inverter with remote mode, turning it on lets me control the inverter remotely. I don't really get where the confusion is coming from?

It just isn't for me. I was going to just let it go until someone else started asking questions, but then if people are asking basic questions then that's probably a sign to reconsider the presentation. The names were already pretty long where I was having to look at screens like this and try and remember what *position* a setting was in for testing:

[image]

There's literally three things called "Storege Rem..." and while the middle one is obviously different since it's seconds (it's the timeout for the register called "Storedge Remote Mode" accoding to the docs, a name which gives no indication it's related to the timeout) the other two are literally indistinguishable as presented by default in HA with the original names.

I should be able to *read* it and know what it is, not have to remember its relative position or open new windows to figure out which is which.

And then let's circle back "remote mode" where after I read the docs I found out *aren't actually related as directly as the names imply*. If "Storedge Control Mode -> Remote Control" then I would think that is directly related to the setting called "Storedge Remote Mode". But then I read the docs and it's totally not. "Storedge Remote Mode" is directly related to "Storedge Remote Command Timeout", whereas when in "Remote Control" mode it's actually "Storedge Default Mode" that is the effective setting unless it's counting a timeout. The naming is all over the place. But it's not anyone's fault because that's how the docs are written in how the names it uses don't always line up with how the docs then describe how they are related when considering someone who may never actually read those docs.

> You would not even be able to guess at what temporary mode does if you didn't know. You would however at least be able to take an educated guess as to what remote mode might do.

There's no indication in the original which setting is tied to the timeout setting. I know now that "Storedge Remote Command Timeout" and "Storedge Remote Mode" are the ones that go together, but I didn't know that the first time I loaded it on my dev system since they're not named in a way that suggests that to me.

Remember that I have not been exposed to how the other integration did these things so I'm coming at all of this like a new user experiencing them for the first time through PR #133. So between my "first time user" confusion, continuing to second guess them looking at code, questions about them were made by another user here, and this conversation even exists is a clear sign that the naming is unclear and potentially confusing.

bertybuttface, 2022-11-06:
> I mean, I don't want to copy someone else's work just because that's what everyone is already used to blindly just because they exist. If it was good enough I wouldn't have spent all this time [...]

I hear you but sticking to the manual is a fairly accepted strategy. I could see deviating from the manual if it is absolutely unavoidable but this feels like a solution in search of a problem. We would be copying the manual not the other integration. 

> There's a technical note that someone leaked that wasn't available publicly from SolarEdge, so it was written with a technical intent and not as a manual for end users. The integration should be end user experience first.

Yes but it used specific terms, terms that are now in wide use within the community.

> The docs define that register as "the operating mode during the defined time frame" and "when expired, it reverts to the default mode". I considered the definition of the word "temporary" as "lasting for only a limited period of time" so that's where I approached it from.

The timeout can be set (and I believe defaults) to zero making the mode indefinite. This is why I think temporary mode is a poor name. I have been running in Remote Command Mode for months. Would it make sense for me to make the same comment calling it Temporary Mode?

> What I don't know though is if the timer starts counting when the timeout register is written, the timeout command mode is written, or both. In any case it clearly says limited time up to a max of 24 hours (86400 seconds).

Remote command mode is not limited to 24 hours if you set timeout to 0 (again I think this is the default)

> > Remote mode was obvious to me. I have a TV remote, it lets me control the TV remotely. I have an Inverter with remote mode, turning it on lets me control the inverter remotely. I don't really get where the confusion is coming from?

> It just isn't for me. I was going to just let it go until someone else started asking questions, but then if people are asking basic questions then that's probably a sign to reconsider the [...]

If we are going to deviate from the manual can we at the very least have a translation document showing how to translate from SolarEdge manual and all other tooling to our own custom naming? I really feel like this change is bad but I doubt I will be able to convince you. Can you let some others weigh in too before making these changes?

> There's literally three things called "Storege Rem..." and while the middle one is obviously different since it's seconds (it's the timeout for the register called "Storedge Remote Mode" accoding to [...]

I'm not sure solving UI issues in Home Assistant by butchering the names to fit is the best way forward.

> I should be able to _read_ it and know what it is, not have to remember its relative position or open new windows to figure out which is which.

I agree in theory, where I don't agree is that "Temporary Mode" is any clearer than "Remote Command Mode". Especially for something that is in fact not temporary at all.
 
> And then let's circle back "remote mode" where after I read the docs I found out _aren't actually related as directly as the names imply_. If "Storedge Control Mode -> Remote Control" then I would [...]

Sure but sticking to established norms is less confusing than making our own standards which others do not follow.

Consistency with SolarEdge Apps, SolarEdge Manuals and existing SolarEdge knowledge (users, fitters) is hugely important. Deviating from that adds complexity, both for you and end users. 

> There's no indication in the original which setting is tied to the timeout setting. I know now that "Storedge Remote Command Timeout" and "Storedge Remote Mode" are the ones that go together, but I [...]

Sure, and I couldn't write C the first time I tried. Not everything is simple enough to understand at first glance. Hiding the complexity by making up our own names will just move the complexity not eliminate it.

> Remember that I have not been exposed to how the other integration did these things so I'm coming at all of this like a new user experiencing them for the first time through PR #133. So between my [...]

You also have not been exposed to the official apps, the solaredge batteries etc. I get that you kinda want to go your own way with this but don't forget this is an integration with SolarEdge Inverters. We are playing in their sandpit. If we can't rename the setting in all the documentation, the official apps and in the installers heads we shouldn't rename it at all.

WillCodeForCats, 2022-11-06:
If there is an actual user manual for the commands, *not the technical note*, it should be submitted to the docs directory. Otherwise any argument about following the manual won't sway me.

WillCodeForCats, 2022-11-06:
Submit screenshots from the apps (English please) so I can see what they show and exactly what words they use.

bertybuttface, 2022-11-06:
> If there is an actual user manual for the commands, _not the technical note_, it should be submitted to the docs directory. Otherwise any argument about following the manual won't sway me.

Would you please stop skipping over the fact that temporary mode isn't temporary.

How does it make any sense to have a temporary mode that can be on indefinitely?

WillCodeForCats, 2022-11-06:
> Would you please stop skipping over the fact that temporary mode isn't temporary.

I find it somewhat offensive that you're accusing me of "skipping over" facts. I quoted the tech note word for word.

bertybuttface, 2022-11-06:
I have a battery here, the setting has been set to remote command mode since I got it installed several months ago. I have asked several times why temporary mode makes sense. I'm sorry if that offends you, it certainly isn't my intention.

Can you see why I think a setting called "temporary" that isn't temporary is poorly named?

WillCodeForCats, 2022-11-06:
The tech note is already been proven to be inaccurate in one case, further inaccuracies are not a surprise, so they should be documented on the wiki as they are brought up:

https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues

Telling me I'm ignoring facts and wanting to go my own way are offensive, after all the time and effort I've put into this. I don't have batteries, I can easily just say forget the battery crap if everyone is so happy with the other integration.

bertybuttface, 2022-11-06:
Ok so can we discuss the fact that temporary mode can be on indefinitely. Does that not strike you as confusing?

Please don't suggest I've not appreciated your hard work on this project when I'm the first person to sponsor you and you are the first person I have ever sponsored to write code (being a programmer myself).

bertybuttface, 2022-11-06:
That still doesn't mean we can name a setting that can be indefinitely set as temporary. You defined the word temporary above and indefinite does not match that definition.

WillCodeForCats, 2022-11-06:
Temporary mode was what I thought of first and is not some final word. I work very loosely and was presenting possible alternatives to @Cereal2nd who asked the quesiton.

I'm trying to find a manual for the solaredge app to see what it says but haven't found anything that describes battery commands yet.

bertybuttface, 2022-11-06:
I'm also looking for something concrete, I'm not at home right now but I will be tomorrow, I'll get some screenshots out the app (and the web portal).

I'm only one data point but I'm against changing names from SolarEdge defaults. If we absolutely must do it we need to think long and hard about the names to make sure they make sense. I can't help but feel this thinking time could be better spent elsewhere.

bertybuttface, 2022-11-06:
Out of interest I don't understand why matching a technical note is not of equal importance to matching a manual? Ultimately it is highly likely to be developers (or at least technically minded users) installing this tool and turning on the advanced settings which are gated behind a disclaimer.

WillCodeForCats, 2022-11-06:
I see it as a guideline because it's not published on the SolarEdge website, already has two documented errors (the timeout behavior and the behavior of the kWh counters), how well it works varies between users, and at least one person has reported trying to contact SolarEdge support for issues related to tech note was ignored. I can't remember where all the comments were but there was someone running a fancy smart panel system that needed to integrate over modbus and the devs of that system had to work magic with SE behind the scenes first. There's also two users that I know if both with the new SolarEdge branded battery banks, one reports data and the other doesn't, even though they have the same batteries.

About the only thing that is accurate in it are the register addresses. But then even the procedure to be able to change these settings is incorrect, it says certain registers need to be written before others will be accepted but in another issue someone reported that it works even though the register the tech note says is required was set to disabled in their system. So those parts of the tech note are inaccurate too.

Compared to the sunspec modbus tech note which doesn't really contain any surprises and its behavior is consistent to real life, which is published on the SolarEdge website. The power control tech note was either withdrawn from download for some reason or never intended to be public in the first place, and the reason it's available looks more like a leak than something SolarEdge indented to be out there. By comparison the power control tech note comes across as something that would be given to developers that have some kind of dev agreement with SolarEdge (like that smart panel system which I can't remember the name).

One example in naming is where the tech note says "Storage AC Charge Policy" (original PR used "Storedge AC Charge Limit" note storage vs. storedge) but in SetApp it's called "AC Charge Limit" on my inverter (I can view and set some storage settings and all of the export limit settings even though I can't access them over modbus). So from that I would assume there are more differences - I only checked a few screens as some of them are very different to the point where I need to do more research, but my takeaway is that the tech note should not be the final word on naming since deployed equipment doesn't use those names.

WillCodeForCats, 2022-11-07:
So as I dig deeper into this I find that tech note naming was only loosely followed, which only establishes a precedent for that other integration, but not necessarily one that should be followed here based on issues already identified.

Examples:

Original named "Storedge Remote Mode" is named "Remote Control Command Mode" in the tech note.

Original named "Storedge Control Mode" is named "Storage Charge/Discharge Default Mode" in the tech note.

Original named "Storedge Remote Mode" is named "Remote Control Command Mode" in the tech note.

The original transposes "storage" with "storedge" in every name. Easy to miss but nevertheless not tech note accurate.

The tech note itself fails to follow a naming convention for similar setings: "Remote Control Charge Limit" vs. "Remote Control Command Discharge Limit". Original further deviates with "Storedge Remote Charge Limit" and "Storedge Remote Discharge Limit"

I don't want this to sound like I'm nitpicking the contributions of PR #133 which are greatly appreciated and critical to getting those features implemented faster than I was able to do on my own, but if the argument against my name changes are about being tech note accurate with the original, well, it never was tech note accurate.

As far as "temporary": naming it alternate or secondary with alternate or secondary timeout and add a known issue note that alternative or secondary mode may be treated as default by the inverter if timeout is set to zero. But prefixing "storedge" and "remote" multiple names adds long repetitive prefixing to entity names that can chop the name into ellipses in the HA UI (which I dislike seeing). However, since names in HA can be changed by the user, the user can simply rename all the entities to their preferred names and a wiki note with how to rename and a renaming map from other names would be a better solution IMO for people coming over from another integration.

bertybuttface, 2022-11-07:
> So as I dig deeper into this I find that tech note naming was only loosely followed, which only establishes a precedent for that other integration, but not necessarily one that should be followed here based on issues already identified.
> 
> Examples:
> 
> Original named "Storedge Remote Mode" is named "Remote Control Command Mode" in the tech note.
> 
> Original named "Storedge Control Mode" is named "Storage Charge/Discharge Default Mode" in the tech note.
> 
> Original named "Storedge Backup Reserved" is named "Storage Backup Reserved Setting" in the tech note.
> 
> The original transposes "storage" with "storedge" in every name. Easy to miss but nevertheless not tech note accurate.

Sounds like something the marketing department would do just before release. (s/Storage/StorEdge).

> The tech note itself fails to follow a naming convention for similar setings: "Remote Control Charge Limit" vs. "Remote Control Command Discharge Limit". Original further deviates with "Storedge Remote Charge Limit" and "Storedge Remote Discharge Limit"

Looks like they are trying to keep the names under a certain length in the tech note (less than 32 chars).

> I don't want this to sound like I'm nitpicking the contributions of PR #133 which are greatly appreciated and critical to getting those features implemented faster than I was able to do on my own, [...]

Accuracy isn't nitpicking, you are absolutely correct to point out inconsistencies.

> As far as "temporary": naming it alternate or secondary with alternate or secondary timeout and add a known issue note that alternative or secondary mode may be treated as default by the inverter if [...]

I agree the prefixing can get a bit long. I'm not sure how "alternative" is shorter than "remote" however.

> However, since names in HA can be changed by the user, the user can simply rename all the entities to their preferred names and a wiki note with how to rename and a renaming map from other names would be a better solution IMO for people coming over from another integration.

Sure, it might be worth having a compatibility shim that lets people transition from the old integration. The only thing is that's not going to be fun for you when you start getting bug reports with various different names in use.

> Edit: I also think that using "StorEdge" should be avoided because there's new inverter models that support batteries that no longer use the StorEdge branding.

StorEdge does seem to be the old branding.

WillCodeForCats, 2022-11-07:
These are probably issues where the community can step in if there's a need.

I already have different naming for all inverter registers, so anyone coming over already has to deal with updating entity names and I'm not aware of any major complaintst. [The name change wasn't a problem for Remko's energy dashboard.](https://community.home-assistant.io/t/solaredge-modbus-configuration-for-single-inverter-and-battery/464084) Someone who has used both could contribute a wiki page with an old name -> new name table for that.

I really don't like "remote" in the names becasue it simply doesn't make sense to me mentally and probably never will since I won't be using it regularly to become accustomed to what they actually do vs. what the name implies. But if battery users are stuck on it then it can stay. The only major drawback is lack of support from me about something I can't keep straight in my head, but others can answer those questions when they come up.

DougMunford, 2022-11-08:
I have 3 inverters, 2 of which have SE energy banks connected to them via energy net. All firmwares at latest from SetApp. For me setting the “Storedge Remote Mode” to anything different than the “Storedge Default Mode” observes the  “Remote Command Timeout” perfectly, returning to the default after this timeout expires.

bertybuttface, 2022-11-08:
> These are probably issues where the community can step in if there's a need.

I'm happy to maintain this once you've decided on names.

> But if battery users are stuck on it then it can stay.

I seem to be in the minority here as others want it changed. I also seem to have a weird misconfiguration as it looks like Storedge Remote Mode should in fact be temporary. I have also been messing with Modbus on my inverter so it is entirely possible I have screwed something up.

WillCodeForCats, 2022-11-08:
What batteries/inverters do you have? It may just be difference in equipment where people with newer equipment don't experience the same behavior as people with older equipment.

@DougMunford just reported that his Energy Bank batteries are not affected by the "reset to zero" issue that were reported by other users with LG batteries, so that behavior got turned into an option in the latest pre-release.

bertybuttface, 2022-11-08:
I have a SE energy bank here but I don't have a SolarEdge meter and I've been using my own, both with https://github.com/nmakel/solaredge_meterproxy and with a custom package I've been developing for an ESP32. It could be that in the process of developing this I've gotten my inverter into a weird state.

DougMunford, 2022-11-08:
So, when I first switched the control mode to remote the import/export limit values were set at 5000, the default mode was not set and the timeout was set at 3600 seconds 1 hour). The inverters and batteries had only been used in MSC (out of the box) or installer account power profile ToU mode applied before now.

Batteries:
[image]

Inverters:
[image]

WillCodeForCats, 2022-11-08:
Separate question, but do you know what Status = 10 is? It's not defined in any of the specs I have.

DougMunford, 2022-11-08:
I ran them dry. This is what the app reports right now while they are at 10. 

[image]

EddAshforth, 2022-11-08:
I get 10 when my battery is ran to empty too. I have a SE branded batterySent from my iPhone

WillCodeForCats, 2022-11-08:
Thanks, will add Power Saving mode in the next release.

DougMunford, 2022-11-08:
I see status 6 in my logs too, I'll set an alert so I can find out what the app shows next time it happens.

EddAshforth, 2022-11-08:
6 I believe is a full battery whilst on MSC (charging from PV), where as 1
is full when charging from remote control PV and AC.

EddAshforth, 2022-11-08:
This was my b1 status today:

[image: image.png]

Charging battery from 2am, it was full just after 3am. Actually, my
automation then puts the system into a disabled state (meaning i supply the
house from the grid), which is probably 1.
3 is charging. Then 6 is as mentioned, full during an MSC charge.

DougMunford, 2022-11-08:
Oh, the logs are interesting, This is where my 0 figures were reported:

[image]

DougMunford, 2022-11-08:
Sorry, adding the other battery makes it clearer:

[image]

When in a status of 0 the values are also reported as 0.

DougMunford, 2022-11-09:
Interesting, when I had some other figure than 3600 set as the remote timeout it was reset to 3600 on one of these availability and init events. This kit is very weird.

[image]

DougMunford, 2022-11-09:
On the naming thing, yes it needs to be clearer to the user. I find it interesting that the UI for an installer in their monitoring app also changes names of modes to be programmed to make things clearer. 

[Documentation](https://www.solaredge.com/sites/default/files/storedge_charge_discharge_profile_programming.pdf)

DougMunford, 2022-11-09:
Status 6 doesn’t reflect in the app. I’ve had one battery charging while the other was not while in status 6, the app just shows charging. While in status 6 the battery has DC current and DC power are reported as 0. I wonder if this is some sort of BMS pause?

DougMunford, 2022-11-09:
Observations:

Setting the “Storedge Remote Mode” on a follower inverter does not stick, whatever is on the leader overrides it. Setting it on the leader also pushes the same setting to the follower, there is no need to action both, they act in unity. This is NOT the case for the other settings. Both of these inverters have SE Energy bank batteries attached via Energy Net comms.

WillCodeForCats, 2022-11-09:
Actually I don't think there's a way to always determine the leader inverter... it will probably almost always be the inverter with the network connection and the first address, but it doesn't have to be (I have network on my follower inverter so I can have parallel testing going and it will respond to modbus too).

DougMunford, 2022-11-09:
Agreed, until I renumbered mine to better fit the requirements of this integration the leader was number 3, a product of adding to the system over time.  All of my inverters have wired Ethernet although that isn’t really necessary as they are linked via RS485. Setting mode config on the follower does no harm, it just instantly changes back. Perhaps just a note in the wiki?

[9 more replies]

closed 2022-11-30 by WillCodeForCats