WillCodeForCats/solaredge-modbus-multi #137: Export Control

mpredfearn, 2022-10-13:
The export control block of registers allows control of how much power is exported from the system to the grid. This may be useful, for example, to prevent export if the export rate is negative.

Once the select / number controls are merged, it will be relatively easy to add support for this feature which should be the last one to bring this integration up to feature parity with https://github.com/binsentsu/home-assistant-solaredge-modbus (this was merged there in https://github.com/binsentsu/home-assistant-solaredge-modbus/pull/42)

WillCodeForCats, 2022-10-13:
I'm not 100% on this one because it can be used to violate a utility interconnect agreement, such as utilities that require zero export as part of being allowed to grid tie. And it's not always a money thing, at least in Hawaii I know they do this on segments that don't have appropriately sized transformers instead of forcing transformer upgrades like some utilities do.

It should probably be locked behind some kind of warning/disclaimer users have to agree to at minimum.

mpredfearn, 2022-10-13:
Totally agree - this feature must always be used in compliance with the connected utility rules. The way that https://github.com/binsentsu/home-assistant-solaredge-modbus (sort of) managed this was by not implementing the unlock sequence documented on page 15 of https://github.com/WillCodeForCats/solaredge-modbus-multi/blob/main/doc/Power-Control-Open-Protocol-for-SolarEdge-Inverters.pdf - i.e. to make use of the export control feature, the user has to manually write to the registers to enable this behaviour.
Although, for a system which is already configured with export limitation I presume that this functionality is already unlocked and so a user may be already to manipulate the Export Control Site Limit register? I don't know since I don't have such a system. Further input from other people required.

purcell-lab, 2022-10-21:
Zero export capability is an important feature for users who are exposed to wholesale market prices, which means the feed in price can go negative for a variable number of hours during the day. As the timing of the negative feed in is variable being able to switch this capability via HomeAssistant becomes very powerful.

I have 2x three phase 7kW inverters and Powerwall2 and am happy to assist with testing. I find myself switching between binsentsu:solaredge-modbus for the export control and this solaredge-modbus-multi for the multi inverter support ;-(

Last week in South Australia the negative feed in price was -$1.10/ kWh!

[image]

WillCodeForCats, 2022-10-21:
I understand the purpose of it, but I want to implement it in a way that does not expose them by default and requires acknowledgement that altering these settings could be in violation of a utility agreement or cause a large bill if used improperly.

purcell-lab, 2022-10-21:
Thanks for 2.2.6-pre.1, my logs are reporting I_AdvPwrCtrlEn for both my inverters

```
$ grep I_AdvP home-assistant.log
2022-10-21 15:45:54.706 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_AdvPwrCtrlEn 0x0
2022-10-21 15:45:55.834 DEBUG (SyncWorker_0) [custom_components.solaredge_modbus_multi.hub] Inverter 2: I_AdvPwrCtrlEn 0x0
2022-10-21 15:45:56.298 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1: I_AdvPwrCtrlEn 0x0
2022-10-21 15:45:57.326 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 2: I_AdvPwrCtrlEn 0x0
```

ThirstyDursty, 2022-10-21:
> Totally agree - this feature must always be used in compliance with the connected utility rules. The way that https://github.com/binsentsu/home-assistant-solaredge-modbus (sort of) managed this was [...]

Hypothetically how would one write these registers? :)

WillCodeForCats, 2022-10-21:
> Hypothetically how would one write these registers? :) speaking as someone from South Australia that is being charged an arm and leg because I can turn it down

I have been trying out different ideas in the UI, but nothing that can be released yet.

ThirstyDursty, 2022-10-21:
> > Hypothetically how would one write these registers? :) speaking as someone from South Australia that is being charged an arm and leg because I can turn it down
> 
> I have been trying out different ideas in the UI, but nothing that can be released yet.

Are they available settings in the commissioning process part of SetApp? In the SetApp they don’t show up in the settings. How would someone normally (technician) modify these values…they are after all in the documentation for use by installers and technicians

purcell-lab, 2022-10-21:
It looks so simple on the video:
https://youtu.be/Tt0tNX_1k-E

ThirstyDursty, 2022-10-21:
> It looks so simple on the video:
> https://youtu.be/Tt0tNX_1k-E

I can do that. Easy. Just put it in AP mode, connect to is wifi and then go to http://172.16.0.1/

Problem is getting HA to do it. As I have to be next to the unit to do it

ThirstyDursty, 2022-10-21:
> It looks so simple on the video:
> https://youtu.be/Tt0tNX_1k-E
Thought I was onto it…but won’t hold these settings either. Change and then 2min later it is back to default 
[image]

WillCodeForCats, 2022-10-23:
This is what I came up with for a config dialog:

[image]

It would be accessed using a two step process through integration options, then an "advanced power control" checkbox, which if checked brings the user to this screen. Thoughts?

purcell-lab, 2022-10-23:
> It would be accessed using a two step process through integration options, then an "advanced power control" checkbox, which if checked brings the user to this screen. Thoughts

You are providing more context than the SetApp dialogue which I think is a good thing, but still allowing users to access the advanced settings.

ThirstyDursty, 2022-10-23:
> This is what I came up with for a config dialog:
> 
> [image]
> 
> It would be accessed using a two step process through integration options, then an "advanced power control" checkbox, which if checked brings the user to this screen. Thoughts?

Looks perfect…merg!!! 😂 the sun is rising

WillCodeForCats, 2022-10-23:
It's in branch `config-flow` but it doesn't actually do anything yet except act as a UI test.

WillCodeForCats, 2022-10-25:
I couldn't get the selectors to work properly in HA for some reason (they only worked from disabled->enabled but not the reverse) so the final version that's in PR #147 is checkboxes for each "enable" confirmation:

[image]

purcell-lab, 2022-10-25:
It might be good for compatibility with binsentsu to have similar controls:

[image]

Export control limit mode
_select.solaredge_export_control_limit_mode_

[image]

Export Control Mode
_select.solaredge_export_control_mode

[image]

Export Control Site limit
_number.solaredge_export_control_site_limit_

[image]

WillCodeForCats, 2022-10-26:
issue closed in error

Kahn, 2022-10-30:
> Zero export capability is an important feature for users who are exposed to wholesale market prices, which means the feed in price can go negative for a variable number of hours during the day. As [...]

We're seeing similar things here in NSW and after Amber broke their retailer level integration with Solaredge grid services API, it would be great to be plugging this with HA. Flipping circuit breakers off today to doge the negative FIT felt wrong in many ways. I just want zero export while still self consuming.

WillCodeForCats, 2022-10-30:
I need people to verify pymodbus 3.0.0 in Release v2.2.6-pre.3 runs ok since I'm not going to do any new or substantial updates to anything based on older pymodbus.

purcell-lab, 2022-10-30:
> I need people to verify pymodbus 3.0.0 in Release v2.2.6-pre.3 runs ok since I'm not going to do any new or substantial updates to anything based on older pymodbus.

2.2.6-pre.3 running solid for me. 2x three phase 7 kW inverters. Anything in particular you would like me to check?

Some minor transient issues, but probably more to do with my home network.

```
This error originated from a custom integration.

Logger: custom_components.solaredge_modbus_multi
Source: helpers/update_coordinator.py:151
Integration: SolarEdge Modbus Multi Device (documentation, issues)
First occurred: 9:22:18 AM (4 occurrences)
Last logged: 4:21:38 PM

Error fetching SolarEdge Coordinator data: Could not open Modbus/TCP connection to 192.168.86.100
Error fetching SolarEdge Coordinator data: Update failed: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
```

Kahn, 2022-10-30:
So far so good here too after upgrading this afternoon. Two 1P inverters SE6000 and SE5000 follower with meter.

WillCodeForCats, 2022-10-30:
As long as it recovers from connection or update errors automatically without requiring a reload or restart to recover from, it should be fine. I'm looking for major errors where you have to restart home assistant or uncaught exceptions.

purcell-lab, 2022-10-30:
I got this one today with 2.2.6-pre.3.

Unsure if additional error handling is required.

```
This error originated from a custom integration.

Logger: custom_components.solaredge_modbus_multi
Source: custom_components/solaredge_modbus_multi/hub.py:684
Integration: SolarEdge Modbus Multi Device (documentation, issues)
First occurred: 6:39:24 AM (1 occurrences)
Last logged: 6:39:24 AM

Unexpected error fetching SolarEdge Coordinator data: 'ModbusIOException' object has no attribute 'exception_code'
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/update_coordinator.py", line 205, in _async_refresh
    self.data = await self._async_update_data()
  File "/config/custom_components/solaredge_modbus_multi/__init__.py", line 178, in _async_update_data
    return await self._hub.async_refresh_modbus_data()
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 329, in async_refresh_modbus_data
    await self._hass.async_add_executor_job(inverter.read_modbus_data)
  File "/usr/local/lib/python3.10/concurrent/futures/thread.py", line 58, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 684, in read_modbus_data
    if inverter_data.exception_code == ModbusExceptions.IllegalAddress:
AttributeError: 'ModbusIOException' object has no attribute 'exception_code'
```

WillCodeForCats, 2022-10-30:
Yes that's the kinda of failures I'm looking for.

WillCodeForCats, 2022-10-30:
Fix will be in PR #148

purcell-lab, 2022-10-31:
Another odd one here:

```
Logger: homeassistant.helpers.entity
Source: helpers/entity.py:545
First occurred: 4:37:46 AM (2 occurrences)
Last logged: 7:31:35 AM

Updating state for sensor.solaredge_m1_imported_a_kwh (<class 'custom_components.solaredge_modbus_multi.sensor.ACEnergy'>) took 0.658 seconds. Please report it to the custom integration author.
Updating state for sensor.washing_machine_today_s_consumption (<class 'homeassistant.components.tplink.sensor.SmartPlugSensor'>) took 0.672 seconds. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue+label%3A%22integration%3A+tplink%22
```

WillCodeForCats, 2022-11-01:
[Release v2.2.6-pre.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.6-pre.4) is now available.

Assuming this corrects the problems in pre.4 I will make a 2.2.6 release then add storedge control.

purcell-lab, 2022-11-01:
Thanks I have downloaded pre.4 and can now see the Advanced Power Controls in the config flow.

Can you provide some further detail as to what options this chooses?

Direct Export/ indirect Export?

Also would it be possible for a select entity to enable dynamic control and setting the actual limitations in W? Per my comments at https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/137#issuecomment-1290004193

[image]

WillCodeForCats, 2022-11-01:
They don't do anything at all yet.

WillCodeForCats, 2022-11-01:
Once I have a final 2.2.6 release then I will release 2.2.7-pre.1 which will include PR #133, those control functions will be covered under the storedge control option.

You can set the options now and the related function will appear in a future release, but they will not enable any new functionality in 2.2.6.

WillCodeForCats, 2022-11-03:
[Release v2.2.7-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.7-pre.1) is now available.

WillCodeForCats, 2022-11-07:
Almost done with export limit controls, need another day or two to finish up for the next pre-release.

WillCodeForCats, 2022-11-08:
PR #179 adds export limit control.

[image]

The external production and negative site limit settings are disabled by default since it seems unlikely to me that anyone in a residential setting will need those, but they are available if needed.

WillCodeForCats, 2022-11-10:
[Release v2.2.7-pre.6](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.7-pre.6) contains the limit controls PR

WillCodeForCats, 2022-11-10:
[Release v2.2.7-pre.7](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.7-pre.7) adds export controls. Most of it says limit controls though because that's the naming SetApp uses, and I am preferring to use names that user facing things show.

purcell-lab, 2022-11-10:
Excellent, just loading up now and will be able to do some testing later today.

purcell-lab, 2022-11-10:
> Most of it says limit controls though because that's the naming SetApp uses, and I am preferring to use names that user facing things show.

In SetApp the Limit Control Mode options are:
- Export Control
- Production Control
- Minimum Input Control
- Disabled

[image]

I see you have implemented a different set of modes:
- Disabled
- Export/ Import Meter
- Consumption Meter
- Production Limitation

[image]

Are these modes the same?

WillCodeForCats, 2022-11-10:
The selection modes in SetApp do not directly match what is available for control registers.

The control registers have two different mode options where SetApp shows a single "export" mode: "use a consumption meter" and "use a E+I meter". The difference should be which meter it will read to calculate the site limit (I presume this only matters if someone has both meter types otherwise the "best" type is an E+I meter so I would presume SetApp hides that logic and picks E+I first and consumption second).

Minimum input should be the same as negative export mode in registers, but as a non-exclusive flag that can coexist with either export meter mode and production control. In SetApp it appears as an exclusive option.

So In SetApp you have 3 distinct modes presented but in modbus there are more possible combinations.

WillCodeForCats, 2022-11-10:
It's also not clear for negative site limit setting if it signals that the site limit number should be treated as a negative internally or actually expects it to be set to a negative, because the description for the site limit value specifies positive numbers only. It can be changed to allow negative input but it's not clear what the proper behavior should be from the docs.

Unfortunately I can't test al the possible combinations personally since while my inverters do have all of those settings in SetApp, none of them expose the data over modbus.

WillCodeForCats, 2022-11-10:
After I made the release I thought the two export modes seemed unclear and changed the names in PR #181

WillCodeForCats, 2022-11-10:
Would these names be better?

* "Export Control (Export/Import Meter)"
* "Export Control (Consumption Meter)"
* "Production Control"

purcell-lab, 2022-11-11:
Core functionality seems to be working well.

You can see below I have set 0.0 W export and my two inverters are showing 86% & 80% active power limit and my actual export has dropped to a minor -27 W.

[image]

purcell-lab, 2022-11-11:
I have received the occosinal error message when changing modes, the device goes unavailable for a couple of seconds and then comes back online.

```
Logger: homeassistant.components.websocket_api.http.connection
Source: custom_components/solaredge_modbus_multi/hub.py:478
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 7:34:42 AM (3 occurrences)
Last logged: 7:39:29 AM

[281472546293696] Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 200, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 1744, in async_call
    task.result()
  File "/usr/src/homeassistant/homeassistant/core.py", line 1781, in _execute_service
    await cast(Callable[[ServiceCall], Awaitable[None]], handler.job.target)(
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 208, in handle_service
    await service.entity_service_call(
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 678, in entity_service_call
    future.result()  # pop exception if have
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 943, in async_request_call
    await coro
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 715, in _handle_entity_call
    await result
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 109, in async_set_value
    await entity.async_set_native_value(native_value)
  File "/config/custom_components/solaredge_modbus_multi/number.py", line 352, in async_set_native_value
    await self._platform.write_registers(
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 1076, in write_registers
    await self.hub.write_registers(self.inverter_unit_id, address, payload)
  File "/config/custom_components/solaredge_modbus_multi/hub.py", line 478, in write_registers
    raise ModbusWriteError(result)
custom_components.solaredge_modbus_multi.hub.ModbusWriteError: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
```

purcell-lab, 2022-11-11:
There seems to be an odd default site limit of -3.40282346638529e+38 W

[image]

[image]

WillCodeForCats, 2022-11-11:
Will work on cleaning it up a bit.

WillCodeForCats, 2022-11-11:
> Probably better names, but I'm unsure how the end user would choose between meters as they are unlikely to know which one they have installed.

It's on each meter device page:

[image]

So for now they'd just have to look, maybe in the future I can do something else.

WillCodeForCats, 2022-11-11:
Already handled in PR #181

WillCodeForCats, 2022-11-11:
Already handled in commit a80737213e89bb54c826c18a827f613ce324604a

WillCodeForCats, 2022-11-11:
Various fixes will be in PR #188

WillCodeForCats, 2022-11-14:
[Release v2.2.7-pre.8](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.7-pre.8) is now available. It contains a large amount of changes to naming and validation for power control functions.

WillCodeForCats, 2022-11-15:
[Release v2.2.7-pre.9](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.7-pre.9) fixes an issue on inverters that don't support power control functions over modbus.

purcell-lab, 2022-11-16:
Thanks installed 2.2.7-pre.9

The display is better and we no longer have that crazy value.

I am still seeing the device go unavailable briefly when changing modes and an error in the logs, but not much details.

I am trying to reduce export to zero, but it looks like it is now reducing total production to zero, pre.7 was limiting export correctly :-(

[image]

[image]

```
This error originated from a custom integration.

Logger: custom_components.solaredge_modbus_multi
Source: helpers/update_coordinator.py:182
Integration: SolarEdge Modbus Multi Device (documentation, issues)
First occurred: 5:26:29 PM (1 occurrences)
Last logged: 5:26:29 PM

Error fetching SolarEdge Coordinator data: Update failed: Meter read error: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
```
```
2022-11-16 16:26:26.730 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 6.727 seconds (success: True)
2022-11-16 16:26:29.730 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response rece
ived, expected at least 8 bytes (0 received)
2022-11-16 16:26:29.732 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Error fetching SolarEdge Coordinator data: Update failed: Meter read error: Modbus Error: [Input/Output
] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
2022-11-16 16:26:29.733 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 8.642 seconds (success: False)
```

WillCodeForCats, 2022-11-16:
There weren't any functional changes, only naming and validation cleanup. As far as I understand export site limit that's how its supposed to work: if any production exceeds consumption that would cause export in excess of the site limit value, production is throttled back to meet the site limit value. The limit control is described as "dynamically adjusts the PV power production in order to ensure that exported power does not exceed a preconfigured limit".

If an inverter is unresponsive (or a read cycle fails for any reason) entities go into unavailable state until the next successful polling cycle to prevent stale or invalid data from being displayed. This behavior was added in PR #38 (suggested in issue #23). In pre.9 updates were made to conform with existing behavior whereas the earlier pre-releases were incomplete.

If the inverter goes unresponsive after sending it a command I could add a config option to make the integration "sleep" after sending a command, then you can just put some value of 3-5 seconds for the sleep or whatever works for you. I can't actually do anything to fix the inverter going unresponsive.

WillCodeForCats, 2022-11-16:
Delay after sending a command option will be in PR #197 with a default of 3 seconds, config range is 0 to 10 seconds.

WillCodeForCats, 2022-11-16:
[Release v2.2.7-pre.10](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.7-pre.10) includes the new delay option. I fast tracked this one and did very little testing on it, hopefully no bugs or regressions.

purcell-lab, 2022-11-16:
> There weren't any functional changes, only naming and validation cleanup. As far as I understand export site limit that's how its supposed to work: if any production exceeds consumption that would [...]

Thanks, the issue for me was it didn't appear to be limiting export to zero, it appeared to be limiting production to zero.

I will do some further testing today with pre.10 once the sun comes up.

WillCodeForCats, 2022-11-16:
Export limitation is a variable production limitation. The inverter(s) will dynamically adjust between 0% to 100% to seek the target site limit whenever consumption is less than production and there's nothing else to consume excess PV power such as charging a battery or other communicating device like the Smart EV Charger.

WillCodeForCats, 2022-11-16:
If it was in production limitation mode with a 0 site limit, it would have hard dropped to 0 and stayed there. The graphs show them dynamically adjusting and the integration can't do that.

purcell-lab, 2022-11-16:
With pre.10 the device seems to now go unavailable for a longer period when I change modes/ values.

This doesn't seem to impact functionality, but it just appears in the logs and lovelace.

```
2022-11-17 06:05:11.731 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 5.730 seconds (success: True)
2022-11-17 06:05:14.734 DEBUG (SyncWorker_8) [custom_components.solaredge_modbus_multi.hub] Inverter 1: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, ex
pected at least 8 bytes (0 received)
2022-11-17 06:05:14.737 ERROR (MainThread) [custom_components.solaredge_modbus_multi] Error fetching SolarEdge Coordinator data: Update failed: No response from inverter ID 1
2022-11-17 06:05:14.739 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 4.662 seconds (success: False)
2
```

[15 more replies]

closed 2022-11-30 by WillCodeForCats