WillCodeForCats/solaredge-modbus-multi #141: Can't add multiple inverters

stepir, 2022-10-18:
**Describe the bug**
My setup is based on 1 master (with one battery), 2 slave (each with a battery), 1 production (slave) inverter with AC coupling.
I can setup integration with the inverters individually but not all together. 

**Expected behavior**

**Screenshots**
I can add 1 and 2 together:
[image]

I can add 3 and 4 alone but not in chain with 1 and 2. I cannot add also 3 and 4 together (2 inverters in integration config flow, starting ID 3)

**Logs**
```
Logger: homeassistant.config_entries
Source: config_entries.py:1049 
First occurred: 14:15:27 (6 occurrences) 
Last logged: 15:26:09

Config entry 'SolarEdge' for solaredge_modbus_multi integration not ready yet: Inverter device ID 3 not found.; Retrying in background
Config entry 'SolarEdge' for solaredge_modbus_multi integration not ready yet; Retrying in background
```

**Debug Logs**
```
2022-10-18 13:33:07.213 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] solaredge_modbus_multi configuration: number_of_inverters=4, start_device_id=1, detect_meters=True, detect_batteries=False, single_device_entity=True, keep_modbus_open=False,
2022-10-18 13:33:07.238 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_ID 0x53756e53
2022-10-18 13:33:07.238 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_DID 0x1
2022-10-18 13:33:07.238 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SunSpec_Length 0x41
2022-10-18 13:33:07.293 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Manufacturer SolarEdge
2022-10-18 13:33:07.293 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Model SE5000H-RW000BNN4
2022-10-18 13:33:07.293 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Option
2022-10-18 13:33:07.293 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Version 0004.0016.0023
2022-10-18 13:33:07.293 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_SerialNumber 7309592F
2022-10-18 13:33:07.293 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1: C_Device_address 0x1
2022-10-18 13:33:07.340 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: mmppt_DID 0x1
2022-10-18 13:33:07.340 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: mmppt_Length 0x41
2022-10-18 13:33:07.340 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: mmppt_DCA_SF 0x5761
2022-10-18 13:33:07.340 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: mmppt_DCV_SF 0x7474
2022-10-18 13:33:07.340 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: mmppt_DCW_SF 0x4e6f
2022-10-18 13:33:07.340 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: mmppt_DCWH_SF 0x6465
2022-10-18 13:33:07.340 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: mmppt_Events 0x0
2022-10-18 13:33:07.340 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: mmppt_Units 0x0
2022-10-18 13:33:07.340 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1 MMPPT: mmppt_TmsPer 0x0
2022-10-18 13:33:07.340 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 1 is NOT Multiple MPPT
2022-10-18 13:33:07.416 DEBUG (SyncWorker_6) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_SunSpec_DID 0x1
2022-10-18 13:33:07.416 DEBUG (SyncWorker_6) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_SunSpec_Length 0x41
2022-10-18 13:33:07.519 DEBUG (SyncWorker_6) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_Manufacturer WattNode
2022-10-18 13:33:07.519 DEBUG (SyncWorker_6) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_Model WND-3Y-400-MB
2022-10-18 13:33:07.519 DEBUG (SyncWorker_6) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_Option Export+Import
2022-10-18 13:33:07.519 DEBUG (SyncWorker_6) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_Version 25
2022-10-18 13:33:07.519 DEBUG (SyncWorker_6) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_SerialNumber 4124348
2022-10-18 13:33:07.519 DEBUG (SyncWorker_6) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 1: C_Device_address 0x2
2022-10-18 13:33:07.520 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Found meter 1 on inverter ID 1
2022-10-18 13:33:07.701 DEBUG (SyncWorker_2) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 2: C_SunSpec_DID 0x2bd
2022-10-18 13:33:07.701 DEBUG (SyncWorker_2) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 2: C_SunSpec_Length 0x99
2022-10-18 13:33:07.769 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 3: C_SunSpec_DID 0xffff
2022-10-18 13:33:07.769 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 1 meter 3: C_SunSpec_Length 0x0
2022-10-18 13:33:08.000 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_ID 0x53756e53
2022-10-18 13:33:08.000 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_DID 0x1
2022-10-18 13:33:08.000 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SunSpec_Length 0x41
2022-10-18 13:33:08.201 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Manufacturer SolarEdge
2022-10-18 13:33:08.201 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Model SE5000H-RWSACBNN4
2022-10-18 13:33:08.201 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Option
2022-10-18 13:33:08.201 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Version 0004.0016.0023
2022-10-18 13:33:08.201 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_SerialNumber 740277B5
2022-10-18 13:33:08.201 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 2: C_Device_address 0x2
2022-10-18 13:33:08.601 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 2 MMPPT: mmppt_DID 0x2bd
2022-10-18 13:33:08.601 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 2 MMPPT: mmppt_Length 0x99
2022-10-18 13:33:08.601 DEBUG (SyncWorker_7) [custom_components.solaredge_modbus_multi.hub] Inverter 2 MMPPT: mmppt_DCA_SF 0x0
[... 114 lines cut ...]
2022-10-18 13:33:21.363 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 4: C_SunSpec_ID 0x53756e53
2022-10-18 13:33:21.363 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 4: C_SunSpec_DID 0x1
2022-10-18 13:33:21.363 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 4: C_SunSpec_Length 0x41
2022-10-18 13:33:21.645 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 4: C_Manufacturer SolarEdge
2022-10-18 13:33:21.645 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 4: C_Model SE9K
2022-10-18 13:33:21.645 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 4: C_Option
2022-10-18 13:33:21.645 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 4: C_Version 0003.2724
2022-10-18 13:33:21.645 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 4: C_SerialNumber 7E13D3FD
2022-10-18 13:33:21.645 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 4: C_Device_address 0x1
2022-10-18 13:33:21.876 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 4 MMPPT: ReadHoldingRegistersResponse (65)
2022-10-18 13:33:21.876 DEBUG (SyncWorker_4) [custom_components.solaredge_modbus_multi.hub] Inverter 4 is NOT Multiple MPPT
2022-10-18 13:33:22.218 DEBUG (MainThread) [custom_components.solaredge_modbus_multi] Finished fetching SolarEdge Coordinator data in 5.001 seconds (success: False)
2022-10-18 13:33:22.305 DEBUG (SyncWorker_1) [custom_components.solaredge_modbus_multi.hub] Inverter 4 meter 1: C_SunSpec_DID 0xffff
[... cut ...] hub] Inverter 4 meter 1: C_SunSpec_Length 0x0
```
Debug logging will generate a large amount of data: it is recommended to configure it, collect the data needed, then remove debug logging during normal operation.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2022.10.3
 - solaredge-modbus-multi Version: v2.2.4

**Additional context**
The integration does not allow setup of multiple inverters 'manually' (it might have been a workaround?):

[image]

WillCodeForCats, 2022-10-18:
An IP address can only be used once.

stepir, 2022-10-18:
> An IP address can only be used once.

Thanks got it - just mentioned it would have been a workaround design to my issue.  The issue is that I can add my inverters individually but not with inverters = 4, starting id =1.

Do you have ideas on what I could further investigate? I thought I could workaround by putting 3 modbus proxies behind my main proxy....

WillCodeForCats, 2022-10-18:
Add `pymodbus: debug` to the logs.

And also use [Release v2.2.5-pre.5](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.5-pre.5) since there's a PR that removed the "not found" error during setup.

WillCodeForCats, 2022-10-18:
One thing I'm seeing is the `Finished fetching SolarEdge Coordinator data in 5.001 seconds` is a problem because 5 seconds is the timeout, after which it raises a timeout error and stops. I could raise that if the problem is the inverters are slow to respond but it shouldn't be that slow. The specs say timeout is <100 ms per inverter.

I can do a quick pre-release with a higher timeout and see if that helps or you can just change the line yourself. In file `__init__.py` change line 168 where it says `async with async_timeout.timeout(5):` change the 5 to 30 so it reads `async with async_timeout.timeout(30):`

WillCodeForCats, 2022-10-18:
Do you want to change it yourself or is it easier for you if I make a pre-release version with it? If changing the timeout fixes the issue I will make that a permanent change.

stepir, 2022-10-18:
thanks looking into it now - will change my local code to test and report back

closed 2022-10-18 by WillCodeForCats