WillCodeForCats/solaredge-modbus-multi #161: Error fetching SolarEdge Coordinator data - downgrade to Warning?

purcell-lab, 2022-11-03:
**Describe the bug**

When the inverter IP address is unavailable solaredge-modbus-multi reports an ERROR.

[image]

**Expected behavior**

If the network device is unavailable it should be reported as a WARNING as it is likely a transient network issue.
cf, how other integrations handle this condition.

The solaredge-modbus-multi integration does retry in the background, so no functionality is lost, just the initial unable to connect errors should be logged as a WARNING.

[image]

**Logs**
Copy any log warnings or errors in Home Assistant from "custom_components.solaredge_modbus" and "pymodbus.client.sync".

```
This error originated from a custom integration.

Logger: custom_components.solaredge_modbus_multi
Source: helpers/update_coordinator.py:151
Integration: SolarEdge Modbus Multi Device (documentation, issues)
First occurred: 9:03:09 AM (1 occurrences)
Last logged: 9:03:09 AM

Error fetching SolarEdge Coordinator data: Could not open Modbus/TCP connection to 192.168.86.100

Logger: pymodbus.client.tcp
Source: /usr/local/lib/python3.10/site-packages/pymodbus/client/tcp.py:187
First occurred: 9:03:09 AM (1 occurrences)
Last logged: 9:03:09 AM

Connection to (192.168.86.100, 1502) failed: timed out
```

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2022.11
 - solaredge-modbus-multi Version: v2.2.7-pre2

WillCodeForCats, 2022-11-03:
Neither of those are being logged from this integration.

WillCodeForCats, 2022-11-03:
If you note the source lines you can look at where they're actually coming from:

* https://github.com/riptideio/pymodbus/blob/dev/pymodbus/client/tcp.py
* https://github.com/home-assistant/core/blob/dev/homeassistant/helpers/update_coordinator.py

Although the logs messages are being triggered by using components from pymodbus and homeassistant core, I can't change their log levels from within the integration.

closed 2022-11-03 by WillCodeForCats

purcell-lab, 2022-11-04:
The bug report is still valid, I guess I need to raise against those upstream components?

WillCodeForCats, 2022-11-04:
Yes you would have to open issues with the other repos referenced, so it's not a valid issue with respect to this component since the file/code referenced in the `Source:` lines are not mine.