WillCodeForCats/solaredge-modbus-multi #162: Modbus Integration broken after Upgrade to 2.2.6

alsFC, 2022-11-04:
After Upgrading to v 2.2.6 with the new pymodbus version my HA internal Modbus integration is broken. Checking config says: 'Component error: modbus - No module named 'pymodbus.client.sync' and won't let me reboot. Also my manually configured Modbus sensors are unavailable.

I'm on HA 2011.1. Rolling back to 2011.0 before the 2.2.6 upgrade didn't help.

alsFC, 2022-11-04:
After removing SolarEdge Modbus Multi Device and reinstalling v2.2.5 internal Modbus integration is working!

EddAshforth, 2022-11-04:
I had the same issue and rolled back. I also have the older single inverter integration installed and in use, so am wondering if they’re clashing?

2.2.6 also not working for me..

WillCodeForCats, 2022-11-04:
We are using pymodbus 3.0.0 as of release 2.2.6 which is probably not compatible with other things that depend on old versions of pymodbus.

WillCodeForCats, 2022-11-04:
I will try testing on my dev system with also setting up the `modbus` core integration.

WillCodeForCats, 2022-11-04:
I've confirmed I can see the issue when using the core modbus integration together.

All versions of Home Assistant still use a version of pymodbus older than 3.0.0 so changing any HA version won't help.

pymodbus 3.0.0 contains changes that are not backwards compatible with code that relies on old versions.

If you are using the core modbus integration I expect that Home Assistant will eventually update to a current release version of pymodbus, but they have not done so yet. Unfortunately this means you can't use any version of solaredge-modbus-multi newer than v2.2.5 together with the core modbus integration until Home Assistant gets updated.

If you are NOT using the modbus core integration, I recommend completely removing all other custom integrations that rely on older versions of pymodbus to use solaredge-modbus-multi v2.2.6 and newer. We updated to use pymodbus 3.0.0 quickly because it resolved issues several users were having, so we can't downgrade since it would break issues that were closed as fixed.

As far as me fixing it: I'm not going to open a PR with Home Assistant core to fix this because I tried making some simple contributions in the past (start small and work up to bigger things) but they were rejected as unwanted. Shortly after that though someone else submitted a PR for the exact same thing I did which was accepted without question, so what I took away from that experience was that I'm not welcome there.

dolenec, 2022-11-04:
So this mean that we must wait until HA is updated to pymodbus 3.0.0?

stepir, 2022-11-04:
> I've confirmed I can see the issue when using the core modbus integration together.
>
> All versions of Home Assistant still use a version of pymodbus older than 3.0.0 so changing any HA version won't [...]

Sorry to hear such @WillCodeForCats as I think this is the best SE integration available and you did a great job with it. As per the issue I will keep 2.2.5 until a new solution is found

stepir, 2022-11-04:
ok so rolling back will not work for me it seems - this happen while trying to setup 2.2.5 and also tried with 2.2.4

[image]

`2022-11-04 18:26:06.493 ERROR (MainThread) [homeassistant.config_entries] Error occurred loading configuration flow for integration solaredge_modbus_multi: cannot import name 'ModbusTcpClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/__init__.py)`

WillCodeForCats, 2022-11-04:
> So this mean that we must wait until HA is updated to pymodbus 3.0.0?

Yes, you'll have to wait for Home Assistant to catch up if you want to use the core modbus integration as released together with solaredge-modbus-multi v2.2.6+.

Or if you feel like trying an override to the core modbus integration with a custom one, I went through the core integration and changed a few things to get it to load. I haven't tested it beyond reading a couple registers from a modbus power meter I have.

Download this file: [modbus.zip](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/9940772/modbus.zip)

Unzip the folder and copy it into your `config/custom_components` folder and restart Home Assistant. You may have to comment out any modbus sections in `configuration.yaml` to get it to pass the config reboot test.

Source code: https://github.com/WillCodeForCats/home-assistant-core/tree/pymodbus-3-0-0
Changes: https://github.com/home-assistant/core/compare/dev...WillCodeForCats:home-assistant-core:pymodbus-3-0-0

stepir, 2022-11-04:
I removed first 2.2.6 and my modbus config started back working - do you mean comment out just for the time of the setup of 2.2.5? because I have other stuff integrated via modbus that need to keep

WillCodeForCats, 2022-11-04:
I would just try commenting it out for the purposes of getting it to forget that it had pymodbus 3.0.0 loaded, I didn't mean permanently.

WillCodeForCats, 2022-11-04:
The maintainer for the modbus integration in Home Assistant said it isn't getting an update anytime soon, they are waiting for a 3.0.1 release and they are going to rewrite some code first. No idea how long that will be.

So unfortunately the bottom line for anyone using Home Assistant modbus core along with solaredge-modbus-multi, you are stuck at release v2.2.5 until Home Assistant gets updated.

Ref: https://github.com/riptideio/pymodbus/issues/873#issuecomment-1303943371

alsFC, 2022-11-04:
> I removed first 2.2.6 and my modbus config started back working - do you mean comment out just for the time of the setup of 2.2.5? because I have other stuff integrated via modbus that need to keep

I removed the integration first, then deleted the HACS component. After reboot enabled Modbus again and as it was working I reinstalled the HACS component 2.2.5 and set up the integration.

stepir, 2022-11-04:
same steps followed but get the invalid handler error

WillCodeForCats, 2022-11-05:
Added a compatibility warning to the 2.2.6 release about other modbus integrations.

New pymodbus 3.0.1 was just released so maybe it won't be too long before Home Assistant catches up.

roykrikke, 2022-11-05:
> Added a compatibility warning to the 2.2.6 release about other modbus integrations.
> 
> New pymodbus 3.0.1 was just released so maybe it won't be too long before Home Assistant catches up.

After some reading and searching for a solution I came across a new release: [pymodbus 3.0.2](https://pymodbus.readthedocs.io/en/latest/changelog.html#version-3-0-2)

pymodbus 3.0.1 and pymodbus 3.0.2 were released shortly after each other.

WillCodeForCats, 2022-11-05:
We're already using it.

roykrikke, 2022-11-05:
Faster than light, how wonderful! Keep up the good work! From my side, thank you very much for the effort!

WillCodeForCats, 2022-11-05:
[Release v2.2.7-pre.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.7-pre.4) is pinned to pymodbus 3.0.2

janiversen, 2022-11-06:
Upgrading the HA modbus integration is a bit more complicated than anticipated but it is on my short list.

@WillCodeForCats I am sorry to hear about your experience with HA, some of the reviewers can be quite “tough” but I do not remember I have seen anything from you for the modbus integration, I normally react very fast to PR for that integration with a positive result.

ThirstyDursty, 2022-11-15:
I'm having this issue...the only modes item i have is my SolarEdge inverter.

I've been using the other Solaredge plugin today....

what do I need to do? delete the other plugin? how do I get rid of the HA Modbus core integration? I don't believe this is something I've installed or can see

janiversen, 2022-11-15:
You have configured modbus in the ha configuration file.

ThirstyDursty, 2022-11-15:
Checked my config file. No mention of modbus. Sent from my iPhone

janiversen, 2022-11-15:
As you might have noted my response was deleted, because you need to do a number of things. The real problem is that this integration uses a tight coupling to HA and not a loose coupling (separate process) that would allow the use of different versions of third party libraries,

ThirstyDursty, 2022-11-15:
Sorry didn’t notice till now…just replied to the email. 

Guess I just have to wait till HA updates the library

janiversen, 2022-11-15:
It is not as simple as just to update the Library. I am the current maintainer of both the modbus integration and pymodbus and I have a number of issues (mainly long term stability and error recovery) that needs solving before I add it to HA.

WillCodeForCats, 2022-11-15:
Developing for HA is still somewhat new for me so I'm not aware if there is a different method of loading the integration that would let it run with a newer version of pymodbus than HA core uses without breaking the modbus integration in core.

ThirstyDursty, 2022-11-17:
I would very much like use this plugin. Can anyone detail what I need to do to get it to work?

WillCodeForCats, 2022-11-17:
> I would very much like use this plugin. Can anyone detail what I need to do to get it to work?

In your logs the conflicting components should log "Unable to import component: No module named 'pymodbus.client.sync'".

If you are using any of the other solaredge modbus custom integrations, you need to delete them (clear from your custom_components folder ore delete in HACS) and reboot before you install v2.2.6 of `solaredge-modbus-multi`.

WillCodeForCats, 2023-01-31:
Home Assistant core has merged pull requests to update pymodbus to v3.1.1 so this issue should be resolved shortly. Once there's a HA release with a current version of pymodbus, I will update the minimum required HA version to match.

* https://github.com/home-assistant/core/pull/85961
* https://github.com/home-assistant/core/pull/86513

WillCodeForCats, 2023-01-31:
Looks like it will be the 2023.2 HA release.

janiversen, 2023-01-31:
Yes, but be aware that pymodbus 3.1.2 is already released and pymodbus 3.1.3 will be released within a week, it solves a requirement problem. The modbus integration will be bumped again in February to 3.1.3

WillCodeForCats, 2023-01-31:
In [Release v2.2.10-pre.1](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.10-pre.1) I've changed requirements to `=>3.0.2` so any newer version should be fine, but I'll update the minimum to be at whatever version HA 2023.2.0 releases with.

Anyone who wants to try the 2023.2 beta channel with any of the v2.2.10 pre-releases it should work when loading `mobus` built-in integration along with `solaredge-modbus-multi`. I've tried it on my dev system with 2023.2.0b8 and it works without conflicts.

closed 2023-02-01 by WillCodeForCats