WillCodeForCats/solaredge-modbus-multi #189: Integration fails to load

pspeirs, 2022-11-13:
**Describe the bug**
Seeing the following errors in the HA logs and the integration fails to load.

I am able to telnet to this port without issues.

**Expected behavior**
Modbus Solaredge integration loads correctly.
Inverter is working, I've had this error a number of times, but most recently updated the integration version maybe a week or so back but didn't have time to verify correct operation.

**Logs**
Logger: homeassistant.setup
Source: setup.py:192
First occurred: 9:02:18 AM (1 occurrences)
Last logged: 9:02:18 AM

Setup failed for custom integration solaredge_modbus_multi: Unable to import component: cannot import name 'ModbusTcpClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/__init__.py)

2022-11-13 09:02:18.321 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration solaredge_modbus_multi: Unable to import component: cannot import name 'ModbusTcpClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/__init__.py)

2022-11-13 10:39:10.189 ERROR (SyncWorker_7) [pymodbus.client.sync] Connection to (10.1.1.125, 502) failed: timed out
2022-11-13 10:47:38.306 ERROR (SyncWorker_1) [pymodbus.client.sync] Connection to (10.1.1.125, 502) failed: timed out
2022-11-13 10:50:46.937 ERROR (SyncWorker_3) [pymodbus.client.sync] Connection to (10.1.1.125, 502) failed: timed out
2022-11-13 10:51:07.959 ERROR (SyncWorker_1) [pymodbus.client.sync] Connection to (10.1.1.125, 502) failed: timed out

**Debug Logs**
Debug logs are not always necessary, but may be required for more complex issues. You can enable debug logs by adding the following to your `configuration.yaml` file and restarting Home Assistant.
```
logger:
    default: warning
    logs:
        custom_components.solaredge_modbus_multi: debug
        #pymodbus: debug #only if needed
```
Debug logging will generate a large amount of data: it is recommended to configure it, collect the data needed, then remove debug logging during normal operation.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version:
 - solaredge-modbus-multi Version:

**Additional context**
Add any other context about the problem here.

WillCodeForCats, 2022-11-13:
> 2022-11-13 09:02:18.321 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration solaredge_modbus_multi: Unable to import component: cannot import name 'ModbusTcpClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/**init**.py)

This means that your Home Assistant installation does not have a compatible version of pymodbus loaded. How are you running Home Assistant?

pspeirs, 2022-11-14:
After a restart yesterday, the Solaredge integration started working but I received this error assumably being the inbuilt modbus?

Setup failed for modbus: Unable to import component: No module named 'pymodbus.client.sync'
November 13, 2022 at 11:11:44 AM – (ERROR) setup.py

This isn't a new integration, it's been previously running for ages.

I'm not sure what you mean by how am I running HA, I'm running HA OS on a Pi4.

WillCodeForCats, 2022-11-14:
Sorry, I read it wrong. Yes the "setup failed" is the core modbus integration which can't be loaded at the same time the integration is loaded. See issue #162 for more info. Basically if you are using the core modbus integration you have to stay on v2.2.5 until Home Assistant gets updated to work with newer versions of pymodbus.

pspeirs, 2022-11-17:
Is there an easy way to rill back the version?

WillCodeForCats, 2022-11-17:
Remove 2.2.6, reboot, and reinstall 2.2.5 is what I suggest.

https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/162#issuecomment-1304011425

closed 2022-11-30 by WillCodeForCats