WillCodeForCats/solaredge-modbus-multi #210: Import exception for switch.py, number.py, and select.py

ehendrix23, 2022-12-07:
Before reporting a bug, make sure your configuration is correct and your LAN is working.
https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Configuration

Do not ask for setup help: setup is not a "bug". Post help requests in one of the community support options.

**Describe the bug**
Import exceptions on start for StorageACChargePolicy, SolarEdgeExternalProduction, and StorageACChargeLimit

**Expected behavior**
Import works as expected.

**Screenshots**
If applicable, add screenshots of the problem and/or configuration.

**Logs**
Copy any log warnings or errors in Home Assistant from "custom_components.solaredge_modbus" and "pymodbus.client.sync".
```
2022-12-07 14:44:40.760 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.solaredge_modbus_multi.select
Traceback (most recent call last):
  File "/Users/Media/homeassistant/lib/python3.9/site-packages/homeassistant/loader.py", line 657, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/Users/Media/homeassistant/lib/python3.9/site-packages/homeassistant/loader.py", line 674, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/Users/Media/hass_config/custom_components/solaredge_modbus_multi/select.py", line 125, in <module>
    class StorageACChargePolicy(SolarEdgeSelectBase):
  File "/Users/Media/hass_config/custom_components/solaredge_modbus_multi/select.py", line 140, in StorageACChargePolicy
    def current_option(self) -> str | None:
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'
2022-12-07 14:44:40.764 ERROR (MainThread) [homeassistant.setup] Unable to prepare setup for platform solaredge_modbus_multi.select: Platform not found (Exception importing custom_components.solaredge_modbus_multi.select).
2022-12-07 14:44:40.775 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.solaredge_modbus_multi.switch
Traceback (most recent call last):
  File "/Users/Media/homeassistant/lib/python3.9/site-packages/homeassistant/loader.py", line 657, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/Users/Media/homeassistant/lib/python3.9/site-packages/homeassistant/loader.py", line 674, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/Users/Media/hass_config/custom_components/solaredge_modbus_multi/switch.py", line 73, in <module>
    class SolarEdgeExternalProduction(SolarEdgeSwitchBase):
  File "/Users/Media/hass_config/custom_components/solaredge_modbus_multi/switch.py", line 93, in SolarEdgeExternalProduction
    def is_on(self) -> bool | None:
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'
2022-12-07 14:44:40.777 ERROR (MainThread) [homeassistant.setup] Unable to prepare setup for platform solaredge_modbus_multi.switch: Platform not found (Exception importing custom_components.solaredge_modbus_multi.switch).
2022-12-07 14:44:41.005 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing platform custom_components.solaredge_modbus_multi.number
Traceback (most recent call last):
  File "/Users/Media/homeassistant/lib/python3.9/site-packages/homeassistant/loader.py", line 657, in get_platform
    cache[full_name] = self._import_platform(platform_name)
  File "/Users/Media/homeassistant/lib/python3.9/site-packages/homeassistant/loader.py", line 674, in _import_platform
    return importlib.import_module(f"{self.pkg_path}.{platform_name}")
  File "/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 850, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/Users/Media/hass_config/custom_components/solaredge_modbus_multi/number.py", line 109, in <module>
    class StorageACChargeLimit(SolarEdgeNumberBase):
  File "/Users/Media/hass_config/custom_components/solaredge_modbus_multi/number.py", line 131, in StorageACChargeLimit
    def native_unit_of_measurement(self) -> str | None:
TypeError: unsupported operand type(s) for |: 'type' and 'NoneType'
2022-12-07 14:44:41.008 ERROR (MainThread) [homeassistant.setup] Unable to prepare setup for platform solaredge_modbus_multi.number: Platform not found (Exception importing custom_components.solaredge_modbus_multi.number).
```
**Debug Logs**
Debug logs are not always necessary, but may be required for more complex issues. You can enable debug logs by adding the following to your `configuration.yaml` file and restarting Home Assistant.
```
logger:
    default: warning
    logs:
        custom_components.solaredge_modbus_multi: debug
        #pymodbus: debug #only if needed
```
Debug logging will generate a large amount of data: it is recommended to configure it, collect the data needed, then remove debug logging during normal operation.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2022.8.7
 - solaredge-modbus-multi Version: v2.2.7

**Additional context**
Add any other context about the problem here.

WillCodeForCats, 2022-12-07:
This integration was developed on systems running Home Assistant Operating System using Python 3.10 looking forward to Python 3.11, whereas it looks like you're using Python 3.9. I believe union typing was introduced in Python 3.10.

Since HA OS uses Python 3.10 I assume you're running HA in some kind of manual environment instead?

WillCodeForCats, 2022-12-08:
I don't know what a backwards compatible method is for older versions of Python, so I would have to look into it. I only have 3.10 in my environment.

WillCodeForCats, 2022-12-26:
Commit [#82193](https://github.com/home-assistant/core/pull/82193) moves Python 3.9 to "depreciated" status so I don't know if I'll spend too much time on this if support for Python 3.9 is about to be removed.

ehendrix23, 2022-12-30:
That works. 😀

closed 2022-12-30 by ehendrix23