WillCodeForCats/solaredge-modbus-multi #214: Plugin Configuration is not possible

dominikamann, 2022-12-21:
**Describe the bug**
Hi,

i installed the integration by HACS (latest Version) and tried to configure the integration. I always running into the following issue.

1.) Select integration
[image]

2.)
[image]

I tried several restarts and different versions. 

Thank you!

**Expected behavior**
Configuration dialog will open.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: Home Assistant 2022.12.7 Supervisor 2022.11.2 Operating System 9.4 Frontend 20221213.1 - latest 
 - solaredge-modbus-multi Version: v2.2.7

dominikamann, 2022-12-21:
If found the following log:
`
 Logger: homeassistant.config_entries
Source: config_entries.py:834
First occurred: 12:00:15 (2 occurrences)
Last logged: 12:02:31
Error occurred loading configuration flow for integration solaredge_modbus_multi: cannot import name 'ModbusTcpClient' from 'pymodbus.client' (/usr/local/lib/python3.10/site-packages/pymodbus/client/__init__.py)`

dominikamann, 2022-12-21:
Duplicate of https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/162

closed 2022-12-21 by dominikamann

WillCodeForCats, 2022-12-21:
Install pymodbus 3.0.1 or newer.