WillCodeForCats/solaredge-modbus-multi #219: Remote Control Charge/Discharge Limit inconsistent

rtlinux, 2022-12-31:
**Describe the bug**

This might be a display issue or it's specific to the 11.4 Hub inverter where the DC charge/discharge limit is in floating point 11.4k (0x46322000) instead of the maximum battery charge power 5000  

**Expected behavior**

Maybe this is a display issue, it's showing inconstant data

**Screenshots**
[image]
[image]

**Logs**

N/A

**Debug Logs**

```
2022-12-31 12:29:34.642 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1: charge_limit 0x46322000
2022-12-31 12:29:34.643 DEBUG (SyncWorker_3) [custom_components.solaredge_modbus_multi.hub] Inverter 1: discharge_limit 0x46322000
```

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2022.12.8
 - solaredge-modbus-multi Version: v2.2.8

Home Assistant 2022.12.8
Supervisor 2022.12.1
Operating System 9.4
Frontend 20221213.1 - latest
 OS Alpine Linux v3.17 (RaspberryPi4)

**Additional context**
https://github.com/WillCodeForCats/solaredge-modbus-multi/blob/e6587867b73049a9f340adb6e2df8fe2902e4687/custom_components/solaredge_modbus_multi/number.py#L305

(feel free to close this if you think it's a SolarEdge firmware issue)

WillCodeForCats, 2023-01-07:
Does it change values depending on what mode it's in?

rtlinux, 2023-01-15:
Based on my testing so far, it doesn't look like it's mode dependent (aside from limit options are Unavailable/not applicable) .

I think it resets back to 11.4 shortly after, regardless what it was set to previously - although I am not 100% sure on this statement, I would need to confirm in logging.
[image]

Also I noticed there is a status 6 for the battery which perhaps is not a documented option yet?

https://github.com/WillCodeForCats/solaredge-modbus-multi/blob/1f495ab8d825b375d31208e9f538774c67119212/custom_components/solaredge_modbus_multi/const.py#L219-L228
[image]

rtlinux, 2023-01-15:
> I found things needed to be set in sequence for them to take.My default mode is always remote. I then set command timeout, delay 500ms, then storage command mode delay 500 then set the limit. After that I can change the limit at any time.Sent from my iPhone

Interesting... It does seem like it made the change but somehow it got rolled back.

I wonder if it's needing an additional "Commit" after setting the holding registry?

WillCodeForCats, 2023-01-16:
> Also I noticed there is a status 6 for the battery which perhaps is not a documented option yet?

Status with value 6 is not defined anywhere that I can find, and not in any of the docs I have.

WillCodeForCats, 2023-01-31:
Status 6 was added in [Release v2.2.10-pre.2](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.10-pre.2)

pwo108, 2023-02-18:
Hello, I have a similar problem.
Since in my region, a part of my bill depends on the quarter hour peak power consumption, I have put in place an automation to preload partially the battery **at slow rate** when the day is not forecasted as sunny:

id: '1675523736717'
alias: Storedge
trigger:
  - platform: time
    at: '07:00:00'
condition:
  - type: is_battery_level
    condition: device
    device_id: 5f2dc6bec0321ccdf1ae62ca8c11bc19
    entity_id: sensor.batterie_storedge_lg_state_of_energy
    domain: sensor
    below: 20
  - condition: numeric_state
    entity_id: sensor.forecasted_solar_energy_today
    below: 7
action:
  - device_id: f7e8f7a5654721590274e7792be53cd0
    domain: select
    entity_id: select.solaredge_i1_storage_command_mode
    type: select_option
    option: Charge from Solar Power and Grid
  - device_id: f7e8f7a5654721590274e7792be53cd0
    domain: select
    entity_id: select.solaredge_i1_storage_control_mode
    type: select_option
    option: Remote Control
  - device_id: f7e8f7a5654721590274e7792be53cd0
    domain: select
    entity_id: select.solaredge_i1_storage_default_mode
    type: select_option
    option: Charge from Solar Power and Grid
  - device_id: f7e8f7a5654721590274e7792be53cd0
    domain: number
    entity_id: number.solaredge_i1_storage_charge_limit
    type: set_value
    value: 1000
mode: single

I have observed that the **storage charge limit keeps the default (5400W)** after the proper execution of this flow (no error message). When I change it manually, it works.

I have added a second corrective automation, with a delay of 5 seconds, to correct the storage charge limit, but it does not work either:
trigger:
  - platform: device
    device_id: f7e8f7a5654721590274e7792be53cd0
    domain: select
    entity_id: select.solaredge_i1_storage_control_mode
    type: current_option_changed
    from: Maximize Self Consumption
    to: Remote Control
    for:
      hours: 0
      minutes: 0
      seconds: 5
condition:
  - condition: numeric_state
    entity_id: number.solaredge_i1_storage_charge_limit
    above: 1000
action:
  - device_id: f7e8f7a5654721590274e7792be53cd0
    domain: number
    entity_id: number.solaredge_i1_storage_charge_limit
    type: set_value
    value: 1000
mode: single

Is something wrong in the way I handle this or is there a bug with the charge limit value (which is key for my purpose of limiting the quarter peak consumption)?

I am running on a Raspberry PI4 and collect the modbus data through an ethernet connection.

Thxs for advice,
Paul

blowk, 2023-02-24:
I also have the same behaviour.
When changing the values to slow charge the battery, it's set back to the 'high' default value.
I haven't found a solution for this yet and I don't know if this is a Solaredge issue that's keeps reset the given input.

WillCodeForCats, 2023-02-24:
Unfortunately I do not have batteries so the only thing I can do is test that it matches what the specs say it's supposed to do with my modbus simulator for any control value. I can set the values and they get sent to the simulator and they hold, so I don't see any integration bugs for setting the storage charge/discharge limits with the simulator. The test values I set are stable through reloads, restarts, and connection interruptions.

WillCodeForCats, 2023-02-24:
Of course, even if it works perfectly in my modbus simulator doesn't mean something could still be missing or not quite right when communicating with a real inverter. I just don't know what that could be since I don't have access to any real inverters with batteries to develop with.

rtlinux, 2023-02-25:
I suspect there might be two issues:
1. the setting resets by an unknown process (either by Battery internal firmware or pushed by SolarEdge from the cloud?)
2. The gauge display in HA is inconsistent sometimes (for example: 11.4 vs 5k vs command overwrites say... 3k)

Here are my observations (measured externally of HA):
```
E00E (F70E) 2 R/W  Remote Control Charge Limit           ==> 11400
E144 (F544) 2 R    Battery 1 Max Charge Continues Power  ==>  5000
```

[image]

[image]

And the following are the events right after the (internal) battery resets I am seeing:

[image]

[image]

[image]

Not sure if it's the best approach or not but based on above findings; perhaps there should be a hard limit of 5K instead of 11.4k (since we know the battery can never go beyond the 5k charge/discharge limit from the battery side)

In any event, thanks for all your hard work!

WillCodeForCats, 2023-02-26:
It already uses the limits determined from the battery for validating the maximum input, but it will display whatever value it's given (even if the value to display is outside of what it will accept based on the limits determined at startup).

rtlinux, 2023-02-28:
No, you're right. I forgot there are customers who have two 5kW batteries that may want to set the limit to 10,000. Thanks for confirming!

Since the behavior aligns with the statements from SolarEdge I believe this is work-as-designed. Feel free to close this issue :)

> 1 Inverter has a max output from the battery(ies) of 5kW. While multiple batteries can be connected, the combined output of the batteries will never exceed 5kW.

>2 Inverter has max output from the batteries of 10kW if 2 or more batteries are connect to the inverter.
NOTE: AC output is limited by the AC rating of the inverter.

Deztak, 2023-04-30:
I have a 3 Battery System and am having this issue, only I can't see the solution.

I'm trying to achieve a battery charge gradient, so on cloudy days I don't over/undercharge the battery while energy prices are low.

Any change to these limits above 5,000W does not get sent to the inverter as it's "outside the valid range", so if I limit my charge rate for the battery, I can't restore the SolarEdge default of 11,400 W ... which is suitable for Batteries with a combined 15,000W power output.

I think that when the Integration is first loaded that it's pulling the max number from the battery but, it should assume that the inverter is setup correctly and pull it from the initial value from the inverter. The field is bound to the inverter after all...

[Error Log.txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/11360616/Error.Log.txt)

WillCodeForCats, 2023-04-30:
Since modbus only supports two batteries the only possible solution then is to remove the max value. There is no way for the integration to know that more than 2 batteries exist, or what the maximum could be.

WillCodeForCats, 2023-04-30:
> I think that when the Integration is first loaded that it's pulling the max number from the battery but, it should assume that the inverter is setup correctly and pull it from the initial value from the inverter.

We can't do that because if HA reloads or restarts when the value is set lower, it will bind the limit to the lower value on startup.

WillCodeForCats, 2023-04-30:
PR #300 will change the upper limit to a fixed value of 50,000.

Deztak, 2023-04-30:
> Since modbus only supports two batteries the only possible solution then is to remove the max value. There is no way for the integration to know that more than 2 batteries exist, or what the maximum could be.

Yeah this is painful. I use some work arounds to correct data based on numbers from the SolarEdge Monitoring API. I read the thread on the missing data for the third battery. I was just providing this information for context.

Deztak, 2023-04-30:
> A reloads or restarts when the value is set lower, it will bind the limit to the lower value on startup.

I guess there would be a good reason this couldn't happen but, thought the niave question couldn't hurt.

WillCodeForCats, 2023-05-02:
Max values for charge and discharge limits changed to constant values (instead of using battery reported values) in [Release v2.3.0-pre.3](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.3.0-pre.3)

closed 2023-05-20 by WillCodeForCats