WillCodeForCats/solaredge-modbus-multi #227: Config-flow kon niet geladen worden: {"message":"Invalid handler specified"}

sbeukers, 2023-01-12:
When I download the integration from HACS all is well. but when i add it to my integrations I get:
"Config-flow kon niet geladen worden: {"message":"Invalid handler specified"}"

[image]

Can you help me with this problem?
If you need anything let me know

WillCodeForCats, 2023-01-12:
Please use the bug report template to submit issues.

closed 2023-01-12 by WillCodeForCats

sbeukers, 2023-01-12:
What did I do wrong? Don't you want this bug fixed? I followed the bug report template.. to my best understanding.
Can you please tell me what you expect more?

WillCodeForCats, 2023-01-12:
I don't see the template. None of the questions it asks were answered. The template exists so I don't have to go back and forth asking questions and waiting for answers for things that should be filled out in the first post. This issue has no logs, no versions, and no information for me to work with.

sbeukers, 2023-01-12:
I wasn't the first I see. 
https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/214
https://github.com/WillCodeForCats/solaredge-modbus-multi/issues/162
thanks

WillCodeForCats, 2023-01-12:
Yes, that problem has a specific log message (which is why the template asks for logs).

Unfortunately there is no fix until HA core catches up with a newer version of pymodbus, or you don't use any other integrations that rely on versions of pymodbus older than 3.0.0. The maintainer is aware of the problem but has not given any timeline for their work to get an updated version into Home Assistant core.

sbeukers, 2023-01-12:
Thanks for your time anyway. The solution was to just install 2.2.5

WillCodeForCats, 2023-01-12:
Yes until Home Assistant gets the update, the last version that will work with old versions of the pymodbus module is release v2.2.5