WillCodeForCats/solaredge-modbus-multi #230: sensor.solaredge_b1_available_energy becomes "Unknown" if capacity exceeds 9700

liquidv, 2023-01-15:
Before reporting a bug, make sure your configuration is correct, your LAN is working, and you have read about the "Known Issues" in the wiki:
https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki

**Describe the bug**
sensor.solaredge_b1_available_energy becomes "Unknown" if capacity exceeds 9700

**Expected behavior**
If capacity exceeds 9700 it should show a value instead of "Unknown"

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: Home Assistant 2023.1.2
 - solaredge-modbus-multi Version: v2.2.9

WillCodeForCats, 2023-01-15:
What's your value of b1 device attribute `batt_rated_energy`?

WillCodeForCats, 2023-01-15:
I will probably also need to see debug logs for `B_Energy_Available` and `B_RatedEnergy` at the time the problem is occurring.

WillCodeForCats, 2023-01-15:
`batt_rated_energy` is not a sensor, it's an attribute of the device sensor.

WillCodeForCats, 2023-01-15:
Should be `sensor.solaredge_b1_device` and it will have a bunch of attributes which are static info about the device.

liquidv, 2023-01-15:
Is this it? Sorry being dumb here...

[image]

liquidv, 2023-01-15:
[image]

WillCodeForCats, 2023-01-15:
Yes, the second one. Maximum energy is correct, but available energy either is not returning a valid value while the battery is in charge mode, or it's reporting something out of range. The sensor for available energy will only show values that are between 0 and the batt rated energy, or 0-9700 in your case.

But I would need to see the debug log values to be sure which condition it's not meeting.

liquidv, 2023-01-15:
I believe I have turned debug on, would you be able to let me know how to retreive those logs? Thanks

liquidv, 2023-01-15:
Using the Solaredge API and various android apps to get battery info (SolarView and SolarMON), battery tips over the 9700 Wh range sometimes when fully charged

[image]

WillCodeForCats, 2023-01-15:
As this integration is currently written it validates the range as 0 to nameplate rating, and rejects values outside of the specs.

liquidv, 2023-01-15:
Ok understood, I had max capacity issues with my battery which Solaredge had fixed and I had been monitoring it closely over the past month and still keeping an eye on it.
It's not an issue during charging as the value is there.

Any future updates to increase the value instead of matching nameplate rating?

WillCodeForCats, 2023-01-15:
The max rated value is a number that we ask the battery to tell us when the integration starts up, it's not set in code.

liquidv, 2023-01-15:
I see, thank you

WillCodeForCats, 2023-01-15:
No values are altered or "normalized".

WillCodeForCats, 2023-01-15:
You need to stop accusing me of saying I'm doing things that I'm not or saying I'm falsifying data, and reread to comprehend what I actually said is happening.

WillCodeForCats, 2023-01-15:
> Ok understood, I had max capacity issues with my battery which Solaredge had fixed and I had been monitoring it closely over the past month and still keeping an eye on it.
> It's not an issue during charging as the value is there.

My question is, does it only then report a value that exceeds its reported maximum rating while in charge mode, or does it still do that in other modes too?

liquidv, 2023-01-15:
> My question is, does it only then report a value that exceeds its reported maximum rating while in charge mode, or does it still do that in other modes too?

It does it in all modes regardless if it is charging or not (charging/discharging/idle).
It stays in an Unknown state until the battery is fully charged again and the BMS decides what value to assign it.
It has now finished charging and now states a value since it's below nameplate value

[image]

WillCodeForCats, 2023-01-16:
My thought was if it's only happening during charging mode then I can add a condition, but if it happens in any mode then that won't work.

WillCodeForCats, 2023-01-16:
So basically there has to be some kind of validation not just displaying any raw value. I wrote this integration because I came from using other integrations that did just show raw data, but it's common enough for the inverters to return bad data that causes errors in long term statistics or pushes graphs out of range for that time period which is something I don't want to deal with as a user. I know it's possible to go into the database and try to remove bad data, but I don't want to do that personally in my HA system so I don't expect anyone else to do that either.

Right now the data validation standard for Maximum and Available energy is to use the rated energy the battery tells us is what its limits are. These limits can be different for different makes and models. If there's an inaccuracy factor in the fixed limits vs. the dynamic data, say the BMS reports within 2% +/-, I could factor that in. But I don't know if that's a thing so I'm just speculating.

In your case the raw value is slightly above so probably fine to display, but there has to be a line somewhere. If I let it display anything then it's guaranteed I'll just get bug reports about values too high or too low messing up displays and we're back to having to validate again.

rtlinux, 2023-01-16:
> Using the Solaredge API and various android apps to get battery info (SolarView and SolarMON), battery tips over the 9700 Wh range sometimes when fully charged

Typically the available energy drops to 9.5kWh, when installed with hardwired communication cable (RS485) . I am guessing you are probably using EnergyNet (wireless)?

I also noticed your SMCU version is 1.0.62 as opposed to the latest 1.0.66 - do you happen to know if your battery is an early version with the built-in fire extinguisher? The max energy higher than rated might have something to do with temperature coefficient. Perhaps 19.2C is the ideal temperature for this battery :)

liquidv, 2023-01-16:
Not sure if you are on the solaredge facebook enthusiasts group but I had
endless troubles with my battery max capacity issue where it would only
give me approximately 7-8kwh battery opposed to 9.7kwh. It took SE 5 weeks
to diagnose the issue and fix it, prompting an issue with the voltage
during charge or something along those lines. Now I get the full 9.7kwh,
and I am also on hardwired ethernet opposed to wireless from the start of
installation.
Not sure if I have a built-in fire extinguisher, but my battery was fitted
in November 2022 so it's relatively new.
Temperature has little effect on the charge capacity after they fixed the
issue. I am still getting 9.7kwh even when the temp drops to 2c outside
where my battery resides in the unheated/uninsulated garage.

[image: image.png]

liquidv, 2023-01-16:
Thank you for the explanation, it seemed like it could have been an easy
fix at the start but as you stated it's dependent on nameplate rating so
there isn't changeable.
I guess not everyone is too concerned about their max capacity rating.
happy to leave things as it stands. Thank you

WillCodeForCats, 2023-01-16:
Do you have a record of the highest number over spec that you've seen reported?

liquidv, 2023-01-16:
> Do you have a record of the highest number over spec that you've seen reported?

Mine is 9754, however my brother in laws who live 2 doors down the road drew the lucky straw and has a SE battery which seems to top out at 10,225! (see pic)

[image]

WillCodeForCats, 2023-01-16:
Do you have any that show over rated capacity not in charging state?

liquidv, 2023-01-16:
> 

[image]

liquidv, 2023-01-16:
> Do you have any that show over rated capacity not in charging state?

[image]

WillCodeForCats, 2023-01-24:
@liquidv PR #234 will add a static value for overage which will be good enough for a pre-release, but I want to do a user settable value for a full release. Once that is merged look for a pre-release you can try.

liquidv, 2023-01-24:
Thank you, will sure to give it a try when ready