WillCodeForCats/solaredge-modbus-multi #239: Unable to prepare setup for platform solaredge_modbus_multi.number: Platform not found

Djelibeybi, 2023-02-01:
Before reporting a bug, make sure your configuration is correct, your LAN is working, and you have read about the "Known Issues" in the wiki:
https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki

**Describe the bug**

Latest beta release throws an "Unable to prepare setup for platform solaredge_modbus_multi.number: Platform not found" error:

**Expected behavior**

Home Assistant starts without an error.

**Screenshots**
If applicable, add screenshots of the problem and/or configuration.

**Logs**
Copy any logs in Home Assistant from "custom_components.solaredge_modbus".

```
Logger: homeassistant.setup
Source: setup.py:340
First occurred: 6:32:35 PM (1 occurrences)
Last logged: 6:32:35 PM

Unable to prepare setup for platform solaredge_modbus_multi.number: Platform not found (No module named 'custom_components.solaredge_modbus_multi.number').
```

**Debug Logs**
Debug logs are not always necessary, but may be required for more complex issues. You can enable debug logs by adding the following to your `configuration.yaml` file and restarting Home Assistant.
```
logger:
    default: warning
    logs:
        custom_components.solaredge_modbus_multi: debug
```
Debug logging will generate a large amount of data: it is recommended to configure it, collect the data needed, then remove debug logging during normal operation.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.1.7
 - solaredge-modbus-multi Version: 2.2.10-pre.4

**Additional context**

This happened initially after upgrading my setup that had battery detection and advanced power control enabled but no combination of enabled/disabled options resolves the problem. It can be reproduced with a fresh configuration.

WillCodeForCats, 2023-02-01:
I am looking in to it.

WillCodeForCats, 2023-02-01:
Will be fixed in the next pre-release later today.

WillCodeForCats, 2023-02-01:
[Release v2.2.10-pre.5](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.10-pre.5)