WillCodeForCats/solaredge-modbus-multi #243: Just sponsored you - but there is no way to contact you

maksyms, 2023-02-03:
Just wanted to thank you for this repo - works like a charm! I've edited your wiki a bit to add data on how to configure SolarEdge inverters without SetApp functionality (i.e., the ones with LCD screen).

Feel free to delete this issue, but may I suggest putting some contact information on your profile? 🙂

maksyms, 2023-02-03:
@WillCodeForCats, that is. :)

WillCodeForCats, 2023-02-03:
Thanks!

Yeah I guess I don't have a public email address on my profile. I changed addresses recently after abandoning my previous one due to spam, so I don't really have one at the moment to give out.

maksyms, 2023-02-03:
Well, a twitter profile or anything to say thank you in words as well would do. Again - thank you! 😀

closed 2023-02-06 by WillCodeForCats