WillCodeForCats/solaredge-modbus-multi #252: SolarEdge Inline Meter(s) not detected by integration

Zandor300, 2023-02-07:
Before reporting a bug, make sure your configuration is correct, your LAN is working, and you have read about the "Known Issues" in the wiki:
https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki

**Describe the bug**
I have an SE8K-RWS48BEN4 inverter with SolarEdge Home Battery 3x 4.8kWh. Addionally, I have 2 [SolarEdge Inline Meters](https://www.solaredge.com/en/products/metering-and-sensors/solaredge-home-inline-meter) connected through [Home Network / Energy Net](https://www.solaredge.com/en/products/communication/solaredge-home-network-plug-in). The first is the import/export meter and the second is metering 2 non-SolarEdge inverters using "External Production" meter mode.

The 2 SolarEdge Inline Meters aren't being detected by the integration eventhough detect meters is enabled, while the battery is. See screenshots below for configuration.

[image]

**Expected behavior**
I expect that both the SolarEdge Inline Meters will show up as separate devices like I've seen for other users for the Modbus energy meter.

**Screenshots**
[image]
[image]
[image]
[image]

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.2.2
 - solaredge-modbus-multi Version: 2.2.11
 - Inverter firmware version: 0004.0017.0119
 - Both SolarEdge Inline meter firmware versions: 1.0.20

Zandor300, 2023-02-07:
@WillCodeForCats 
Thanks for your time looking into my issue. I already suspected that Energy Net meters aren't being reported towards Modbus TCP but thank you for your confirmation.

We are a Dutch installer of solar panels ourselves, and this installation is at our business location to test out the SolarEdge battery system. (Energy storage is only just now beginning to look attractive in the Netherlands)

I've contacted our SolarEdge account manager regarding this issue and he will ask around internally. I will update here once I know more.

WillCodeForCats, 2023-02-07:
If you can get SolarEdge to add support for them over wireless then it will just work, I don't need to do anything. Otherwise it's a choice of choosing convenience of wireless or supported function with modbus.

Zandor300, 2023-02-08:
@WillCodeForCats 
I got word back from our SolarEdge account manager;

He said this is a known issue and SolarEdge HQ is already working on a fix for this. I've asked for an ETA of the fix and an indication how high this is on their todo list.

frankadons, 2023-03-18:
> @WillCodeForCats I got word back from our SolarEdge account manager;
> 
> He said this is a known issue and SolarEdge HQ is already working on a fix for this. I've asked for an ETA of the fix and an indication how high this is on their todo list.

Hi had you already a reply from the Solaredge guy's?  :) I'm facing the same problem.. maybe Its due to the old meter? I have this one for 5 years and It seems Solaredge now uses other types with more functions ..

Zandor300, 2023-03-18:
@WillCodeForCats @frankadons 

Yes, we got word back. Sorry for not updating.

It is to be fixed in SW version 4.18.xxx which is expected in 2 months.

WillCodeForCats, 2023-06-16:
I'm going to close this for now since it's a SolarEdge firmware support issue, but I will add a link back here in the [Known Issues](https://github.com/WillCodeForCats/solaredge-modbus-multi/wiki/Known-Issues) section of the wiki.

closed 2023-06-16 by WillCodeForCats

stormshaker, 2023-07-02:
Progress update - I have SW 4.17.221 with two StorEdge batteries connected via EnergyNet. Both batteries are detected by this integration! One minor note - both the batteries are added to HA named B1, e.g. sensor.solaredge_b1_status and sensor.solaredge_b1_status_2 and both friendly names are 'Solaredge B1'.

WillCodeForCats, 2023-07-02:
Upload your diagnostic file.

stormshaker, 2023-07-02:
Screenshot showing both batteries and diagnostics attached.

[image]
[config_entry-solaredge_modbus_multi-fa86fc31fc23085b5021ddf175ec57ee.json.txt](https://github.com/WillCodeForCats/solaredge-modbus-multi/files/11928755/config_entry-solaredge_modbus_multi-fa86fc31fc23085b5021ddf175ec57ee.json.txt)

WillCodeForCats, 2023-07-02:
According to the diagnostic file there's only one battery.

If you've added multiple hubs or a device "disappears" from the inverter then each repeated battery will be "B1" and those entities are suffixed with _2. Removed or disappearing devices can be removed by clicking the delete button within the device. B1 or B2 is also per inverter, not "total".

stormshaker, 2023-07-02:
Yes - I saw in the diagnostic only one battery is listed. I've only added one hub - the Leader. There are 2 Followers. There are 2 batteries, one with its DC cables connected to the Leader and a second with its DC cable connected to one of the Followers. EnergyNet association *seems* to pair both batteries to the Leader, but then detect which Inverter the DC cables connect to and associate each battery to the correct inverter. But it's hard to tell - I guess the first Follower might also be talking to its local battery via EnergyNet. I only needed to run the Association process on the Leader for the entire system.

Understood the B1 is per Inverter - so then I'll rename one of the batteries to B2 as that makes more sense to me. As they display currently makes for lots of duplicated entity names.

Let me know if you need anything tested - happy to help.

WillCodeForCats, 2023-07-02:
The way the integration assigns the internal unique ID for each battery entity is based on the inverter it's attached to and the battery count for that inverter, where each inverter can have a B1 and a B2. If a duplicate serial number is detected during startup then the duplicate is ignored.

Since the diagnostic file only shows one battery attached to inverter ID 2, so if you actually have two batteries attached to one of the other inverters (id 1 or 3) it would seem one of them is not being read for some reason at least for when the current diagnostic was generated. I assume one of those two battery devices was not showing active entities in it, but the presence of the _2 named entities means at some point in the past one of the inter inverters registered it had a B1 too.

If inverter 1 B1 and inverter 2 B1 were both detected, then they both show up as "Solaredge B1" since they are both the first battery on each inverter. The second one detected gets entities with a `_2` suffix by Home Assistant. I probably could have made the default names something like "I1 B1" and "I2 B1" but I don't remember why not. I did try that with meters for the 2.0.0 release but reverted it because at the time the name change caused loss of long term statistics for anyone coming from a 1.x.x release. I could probably revisit it since now the entities should be safely tied to their internal unique IDs, so that's a thought for a future release (but it would still probably create new devices with the new names even if the entities were preserved).

Without energynet, batteries were physically connected to a specific inverter's R485 bus, .

stormshaker, 2023-07-02:
Makes sense. My setup would be 'I1 B1' and 'I2 B1' but all good, I've just renamed them for me. I'm actually thinking about switching from EnergyNet to RS485 anyway, as the serial comms seems rock solid whereas I've had Battery comms issues sometimes with EnergyNet. Keep up the great work :)

d0minander, 2024-09-16:
Hello, has anyone got the inline meter to be detected when connected via wireless?
I have the latest software as of today, 4.21.23) and my meter is still not detected...

Thanks

WillCodeForCats, 2024-09-25:
Ref #679