WillCodeForCats/solaredge-modbus-multi #263: Unable to change multiple options at once/in short time span

b3nn0, 2023-02-18:
**Describe the bug**
Hi,
I'm using a script (pyscript, but same would also apply to builtin scripts I guess) to change how my storage works.
I.e. if energy is very cheap at night, and solar forecast predicts that I won't get enough PV tomorrow, I might buy from the grid.
Or it might decide to buy current usage from the grid, and keep a specific state of charge in the battery for later, when it is more expensive.
I used binsentu's integration before, where the script worked, but this seems much better maintained, so I tried to switch.
However, the script fails to work now.
This seems to be related to the fact that the script might change multiple parameters in a short time span:
```
    select.solaredge_i1_storage_default_mode.select_option(option=modestr)
    select.solaredge_i1_limit_control_mode.select_option(option=exportmode)
    if exportlimit is not None:
        number.solaredge_i1_site_limit.set_value(exportlimit)
```
If I comment out some of the calls, e.g. to only select the storage_default_mode, it works.
However, as soon as multiple changes are done, none of them seem to work at all, nothing happens.
I also tried playing with the command delay parameter, but didn't seem to make a difference.

**Expected behavior**
All changes should be applied/queued.

**Logs**
Nothing is logged by the integration.

**Debug Logs**
It does say this:
```
2023-02-18 10:27:33.591 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.select] set SE10K-RWS48BEN4_7E061DAC_storage_default_mode to Charge from Solar Power
2023-02-18 10:27:33.591 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.select] set SE10K-RWS48BEN4_7E061DAC_limit_control_mode bits 0000000000000000
2023-02-18 10:27:35.035 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Sleeping 3 seconds after write.
2023-02-18 10:27:36.674 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Sleeping 3 seconds after write.
2023-02-18 10:27:38.036 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 30
```
or
```
2023-02-18 10:29:18.643 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.select] set SE10K-RWS48BEN4_7E061DAC_storage_default_mode to Charge from Solar Power and Grid
2023-02-18 10:29:18.643 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.select] set SE10K-RWS48BEN4_7E061DAC_limit_control_mode bits 0000000000000000
2023-02-18 10:29:19.301 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] coordinator timeout is 30
2023-02-18 10:29:22.188 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Sleeping 3 seconds after write.
2023-02-18 10:29:22.188 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Sleeping 3 seconds after write.
```
but nothing seems to be actually done.
If I change the parameters in the HA UI, it works just fine.

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.2.5
 - solaredge-modbus-multi Version: 2.2.11

WillCodeForCats, 2023-02-21:
So the difference with this integration is that every time a write command is sent, a read to refresh the status of the inverter follows every write so that the data set is consistent. Validation and availability of some commands is affected by the values of others. The delay between writing a command and reading for updates can be set in options (default 3 seconds) because during the time the inverter is processing a write, it usually doesn't respond to reads.

I'm not sure about the scripts aspect (I've never used scripts in HA) or how that interacts with the command response delay setting since it does work in the HA frontend. It seems like the script is not waiting for each command to finish execution with the command response delay before moving to the next step or something, so I would suggest adding sleep calls between every write command in the script as already suggested.

b3nn0, 2023-02-22:
> I'm not sure about the scripts aspect (I've never used scripts in HA) or how that interacts with the command response 
delay setting since it does work in the HA frontend. It seems like the script is not waiting for each command to finish execution with the command response delay before moving to the next step or something

Yes, AFAIK these calls do not block. That's not unique to scripts, but also e.g. if you create an automation that does multiple things in sequence.
I assume the same would happen when doing it via the UI, but I can't really click fast enough to reproduce it.

>  so I would suggest adding sleep calls between every write command in the script as already suggested.

sure, that works for me. I'm fine with it - it just doesn't "feel right". I'd have expected the modbus writes to be queued by the integration if necessary.

WillCodeForCats, 2023-02-22:
As far as understand how HA works "queuing" is not a possibility. HA is supposed to be asynchronous where I tell HA to wait for something to complete but it can go do other things while the task is processing and come back later, which is how the frontend UI works. This is using Python asynchronous frameworks, not specific to the integration.

You can't duplicate it in the UI because the UI does wait for each command to complete (that's what the command execution delay option is) basically causing it to self-queue multiple actions based on asynchronous calls to the write functions and the "await" keyword which is built-in to Python. Why that doesn't translate to scripts I don't know since it's executing the same code.

WillCodeForCats, 2023-02-22:
Just quickly looking at scripts in the UI, there is a "wait for a trigger" step that maybe could be used after executing each command to verify the command has been updated in the device state before moving to the next step. To me that sounds similar to what the frontend is already doing when you click on things rapidly and they execute over time.

WillCodeForCats, 2023-02-22:
I assume your Inverter Command Delay is set to something other than zero, if it's not, try 

The Inverter Command Delay option is actually just a call to asyncio.sleep() built into the integration.

b3nn0, 2023-02-22:
I did try both, with and without delay, but it didn't seem to make a difference.
Interestingly, it's not only that the second and third command in my script fails to do anything:
```
    select.solaredge_i1_storage_default_mode.select_option(option=modestr)
    select.solaredge_i1_limit_control_mode.select_option(option=exportmode)
    if exportlimit is not None:
        number.solaredge_i1_site_limit.set_value(exportlimit)
```
In fact, if written like this, none of them do anything. Not even the first one. Once I add a sleep in between, they start to work.

WillCodeForCats, 2023-02-22:
It's interesting that asyncio.sleep() works if you put it in the script but it doesn't work it when it's built into the integration. Scripts must not follow the same rules as the frontend UI when executing commands.

Even if I made a dynamic internal verification that commands were processed replacing the inverter command delay option, it would still depend on "await" and calls to asyncio.sleep() internally (send command, wait to see if state is updated, wait more if not, repeat), which the scripts are already bypassing.

I can explore ideas for a queue but since it already works in the frontend UI, I can't do anything that would break that.

WillCodeForCats, 2023-02-22:
Does it support putting an `await` in front of each of the commands in the script? I'm not familiar with pyscript.

b3nn0, 2023-02-22:
Unfortuantely no, that doesn't seem to change anything.

b3nn0, 2023-02-22:
Oh, actually, I found a different solution to achive what you intended with the await:
I can add an additional parameter `blocking=True`, which will result in a blocking service call. With that, it seems to work almost correctly. Just the last one (setting site limit right after the entity becomes available) doesn't work. Have to sleep a bit longer for that.

See https://hacs-pyscript.readthedocs.io/en/latest/reference.html#calling-services

WillCodeForCats, 2023-02-22:
> When making a service call, either using the service.call function or the service name as the function, you can optionally pass the keyword argument blocking=True if you would like to wait for the service to finish execution before continuing execution in your function. 

Which explains why I haven't seen this issue because that's the only way the frontend UI works.

WillCodeForCats, 2023-02-22:
I made a simple 3 command script on one of my test systems using HA's built in scripting with an inverter command delay of 10 seconds, and my observation is that this is not a problem with HA's own scripting service.

```
alias: Test Script
sequence:
  - service: select.select_option
    data:
      option: Charge from Solar Power
    target:
      entity_id: select.solaredge_i1_storage_default_mode
  - service: select.select_option
    data:
      option: Export Control (Export/Import Meter)
    target:
      entity_id: select.solaredge_i1_limit_control_mode
  - service: number.set_value
    data:
      value: "20"
    target:
      entity_id: number.solaredge_i1_backup_reserve
mode: single
```

It executed as expected:

[image]

And checking debug logs you can see the async calls were awaited as it's written in the integration code:

```
2023-02-22 10:44:40.162 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.select] set simulated-inv_000001_storage_default_mode to Charge from Solar Power
2023-02-22 10:44:40.167 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Sleeping 10 seconds after write.
2023-02-22 10:44:50.163 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.select] set simulated-inv_000001_limit_control_mode bits 0000000000000001
2023-02-22 10:44:50.168 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Sleeping 10 seconds after write.
2023-02-22 10:45:00.664 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.number] set simulated-inv_000001_storage_backup_reserve to 20.0
2023-02-22 10:45:00.669 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.hub] Sleeping 10 seconds after write.
```

So my conclusion would be that this is a problem with pyscript choosing to ship with a default of non-blocking when that's the opposite of how HA core and HA frontend work. Adding an option to disable blocking would be fine if you warn people that this isn't the default behavior for HA and doing it may cause problems, because then you get issues like this where I'm not able to duplicate the problem in HA until its discovered the real problem is a third party integration deciding to do opposite of how HA behaves and making opposite its default.

I don't really think this is my problem to solve when the integration works with HA's built in script service and UI, and nobody has reported this problem with HA automatons. I assume you use a third party scripting service because it can do more stuff that the built in scripting can't do. And sure, it worked with another integration that (if I'm being honest the reason I actually went to the trouble to learn how to make this integration) didn't follow HA's development standards as well as I seek to follow and I'm big on staying with how HA core works, they were both kinda doing stuff out of spec that happened to work by coincidence but then doesn't work with a different integration because pyscript decides it knows better and ignores await calls by default that were there for a reason.

WillCodeForCats, 2023-02-22:
I've added a note about using `blocking=True` for pyscript service calls in the wiki.

b3nn0, 2023-02-22:
Yes, I agree, this is probably quite a niche application, so wiki note is appreciated, but that's enough.
Thank you anyway, and sorry for all the fuss. Great work you are doing here.

WillCodeForCats, 2023-02-22:
Honestly if pyscript's default for service calls was blocking like how HA's scripts run, you wouldn't have noticed.

closed 2023-02-22 by WillCodeForCats