WillCodeForCats/solaredge-modbus-multi #264: Resetting battery lifetime meters causes log-warning spam

b3nn0, 2023-02-18:
**Describe the current behavior**
I have a BYD battery, which is also affected by resetting lifetime import/export meter (I took the liberty to add that to the "Known Issues" wiki page). Therefore I enabled the respective checkbox in the configuration. It's not really an issue, I just configured a utility_meter to add up the changes.
However, this will cause quite a lot of log spamming by this integration:
[image]

**What should be updated or changed?**
This really shouldn't be a log warning, IMO. but if it is, it probably should only be logged once.
Looking at the code:
https://github.com/WillCodeForCats/solaredge-modbus-multi/blob/f7eadd889545ee578307dc1952e8eae37661217e/custom_components/solaredge_modbus_multi/sensor.py#L1981
and
https://github.com/WillCodeForCats/solaredge-modbus-multi/blob/f7eadd889545ee578307dc1952e8eae37661217e/custom_components/solaredge_modbus_multi/sensor.py#L2049

I can see, there is already the intention to at least only have it logged once per cycle - which is still too much IMO, because the sensor sometimes resets multiple times per hour. However, it doesn't seem to work correctly, not sure why, so the log is spammed for each single reading until the new value is greater than the last peak (I think?)

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.2.5
 - solaredge-modbus-multi Version: 2.2.11

WillCodeForCats, 2023-02-20:
I can just change it to a debug level log.

b3nn0, 2023-02-20:
Sounds good to me 👍

WillCodeForCats, 2023-02-20:
This will be in the next pre-release.

closed 2023-02-20 by WillCodeForCats