WillCodeForCats/solaredge-modbus-multi #273: Fault tolerant reset battery energy counter

Largelos, 2023-03-04:
**Describe the feature**
I'm using integration with two SolarEdge Energy Battery connected to SE5000H, but it seems like sometimes energy counters are reset anyway, just like with BYD or LG batteries, so I have to enable allow battery energy counters to reset to zero recently. (screen1)
The problem is, if I allow this option, this cause wrong energy calculation because of sporadic cases when a query temporarily result a lower value, and returns to the previous value after, what you can see on second screen at 4:25AM.
I would like to see some kind of fault tolerance, in which the reset of the value occurs only if the battery actually resets the counter, and does not return incorrect data temporary. I think this can be done by check the last few values ​​before reset, and reset only if multiple values are less than the current maximum, or mb there is some other solution for this problem?

**Screenshots**
[image]
(Screen1)
[image]
(Screen2)
[image]
(Screen3)

**Home Assistant (please complete the following information):**
 - Home Assistant Core Version: 2023.2.3
 - solaredge-modbus-multi Version: 2.2.13

**Additional context**

WillCodeForCats, 2023-03-04:
The default way all kWh energy counters are treated is to check that the value is equal to or higher than the previous value, i.e. always increasing. Turning on that option for batteries skips that check.

According to your graphs there it looks like it reset multiple times, which I don't know how you can reliably detect is  a "reset" or just bad data. Integrations do not store data, the recorder component does all that, so it is not aware of previous values of a sensor.

Largelos, 2023-03-05:
Yes, I understand that data saved in recorder database, but if you can get latest value to check, is it possible to check latest-1 value also? If not, possible to create additional auxiliary service sensor or hided integration option for pending energy counters reset? Instead of do reset at first time when value are less previous, do not update entity but change pending state, and update only if pending already triggered? Or maybe you can come up with more elegant solution for this problem?

WillCodeForCats, 2023-03-05:
The default when Allow Battery Energy to Reset is disabled will never let it go backwards or reset to zero (outside of a HA restart). It is checking that _latest-1_ is equal to or greater than _latest_ and _latest>0_.

Turning on Allow Battery Energy to Reset skips the _latest-1_ check. I could make it delay allowing the counter to go backwards and/or rest by N number of polling cycles, but there's no way to know what good value of N would be to hardcode, which means N would have to be a configuration option to go with Allow Battery Energy to Reset.

Largelos, 2023-03-05:
Yes, this is exactly what I propose, and I think it would be perfect if everyone can set in settings the N number of poll cycles for which integration return undefined value, just like when Allow Battery Energy to Reset is disabled, before return of decreased or zero value.

WillCodeForCats, 2023-06-18:
Added in PR #341

WillCodeForCats, 2023-06-19:
Testing in [Release v2.3.4-pre.3](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.3.4-pre.3)

Largelos, 2023-06-19:
With [Release v2.3.4-pre.3](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.3.4-pre.3), entity remain in unknown state after counter decreased.
I have enabled allow to reset option and set update cycles value to "5". 
In debug log went backwards record repeat every pool.
```
	Line 2711: 2023-06-19 17:06:56.110 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Import Energy went backwards: 26860 < 28995
	Line 3076: 2023-06-19 17:11:26.363 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Import Energy went backwards: 26860 < 28995
	Line 3441: 2023-06-19 17:11:44.419 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Import Energy went backwards: 26860 < 28995
	Line 3806: 2023-06-19 17:12:02.640 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Import Energy went backwards: 26860 < 28995
	Line 4171: 2023-06-19 17:12:20.135 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Import Energy went backwards: 26860 < 28995
	Line 4536: 2023-06-19 17:12:38.510 DEBUG (MainThread) [custom_components.solaredge_modbus_multi.sensor] Battery Import Energy went backwards: 26860 < 28995
```
After change update cycles value to 0 and apply settings, entity become available again:
[image]

WillCodeForCats, 2023-06-25:
@Largelos Fixed in PR #346

WillCodeForCats, 2023-06-25:
[Release v2.3.4-pre.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.3.4-pre.4)

closed 2023-06-25 by WillCodeForCats