WillCodeForCats/solaredge-modbus-multi #282: New Device Classes in HA 2023.04

ryanm101, 2023-04-06:
https://www.home-assistant.io/blog/2023/04/05/release-20234/#other-noteworthy-changes

```
[@emontnemery](https://github.com/emontnemery) added energy storage and volume storage device classes; these will allow differentiating between, for example, consumed energy versus stored energy in a battery.
```
battery: Percentage of battery that is left in %
energy: Energy in Wh, kWh, MWh, MJ, or GJ
energy_storage: Stored energy in Wh, kWh, MWh, MJ, or GJ

I'm thinking some of the devices should possibly use the new classes

ryanm101, 2023-04-06:
```
sensor.solaredge_b1_available_energy:
      device_class: energy_storage
sensor.solaredge_b1_maximum_energy:
      device_class: energy_storage
```

WillCodeForCats, 2023-04-06:
I haven't had a chance to look at release notes yet, but I will.

closed 2023-04-09 by WillCodeForCats

WillCodeForCats, 2023-04-09:
This has been added starting with [v2.2.14-pre.4](https://github.com/WillCodeForCats/solaredge-modbus-multi/releases/tag/v2.2.14-pre.4)